Koenkk/zigbee2mqtt #1077: Hanging when not using channel 11 on CC2530

nanosonde, 2019-02-12:
I cannot use other channels than channel 11.

The log says:
```
 zigbee-shepherd:init Start the ZNP as a coordinator... +0ms
  zigbee-shepherd:request REQ --> ZDO:startupFromApp +4ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:stateChangeInd +1s
```
And then it keeps hanging forever.

zigbee2mqtt version 1.1.1 (commit #a2d378c)
Coordinator firmware version: '20190109'
CC2530 without HW handshake (RTS/CTS)

JLFN, 2019-02-12:
Iam running it on 1.1.1 with no problem
```.yaml
serial:
  port: /opt/zigbee2mqttdev/vusb/zigbee_cc2530
  disable_led: false
advanced:
  pan_id: 6754
  channel: 26
  cache_state: true
  log_level: info
  log_directory: data/log/%TIMESTAMP%
  baudrate: 115200
  rtscts: false
  soft_reset_timeout: 0
```

Koenkk, 2019-02-12:
Can you try with the latest dev branch?

nanosonde, 2019-02-12:
Oh. Embarrassing. I was think in terms of Wifi channels (1-11) and Zigbee channels (11-22, [23-26]).
Saw it here: https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence

nanosonde, 2019-02-12:
I think there should be a check for the valid channel range which bails out if it is out of range.

closed 2019-02-12 by nanosonde

Koenkk, 2019-02-13:
@nanosonde done.