Koenkk/zigbee2mqtt #1079: OSRAM Products connected, but offline

thissiac, 2019-02-12:
I have been looking for this problem for a long time, but have the feeling to be the only one. 

- Raspi 3
- OpenHab 2.4
- Homematic radio module
- USB flash drive flashed with zigbee2mqtt

My homematic's are working correct
My XIAOMI's are working correct

Now I have tried to integrate my OSRAM and Müller products. It doesn't work.
After initial difficulties with peering, everything has been recognized.

zigbee2mqtt:info 2019-2-13 00:56:10 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-02-13.00-56-10'
  zigbee2mqtt:info 2019-2-13 00:56:10 Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
  zigbee2mqtt:info 2019-2-13 00:56:10 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-13 00:56:11 zigbee-shepherd started
  zigbee2mqtt:info 2019-2-13 00:56:11 Coordinator firmware version: '20180815'
  zigbee2mqtt:info 2019-2-13 00:56:11 Currently 8 devices are joined:
  zigbee2mqtt:info 2019-2-13 00:56:11 0x00158d0002059fb6 (0x00158d0002059fb6): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
  zigbee2mqtt:info 2019-2-13 00:56:11 0x7cb03eaa0a021f69 (0x7cb03eaa0a021f69): AB3257001NJ - OSRAM Smart+ plug (Router)
  zigbee2mqtt:info 2019-2-13 00:56:11 0x000d6f000f5e0b96 (0x000d6f000f5e0b96): AC01353010G - OSRAM SMART+ Motion Sensor (EndDevice)
  zigbee2mqtt:info 2019-2-13 00:56:11 0x84182600000f65fa (0x84182600000f65fa): AC03641 - OSRAM LIGHTIFY LED Classic A60 clear (Router)
  zigbee2mqtt:info 2019-2-13 00:56:11 0x00158d000273a160 (0x00158d000273a160): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
  zigbee2mqtt:info 2019-2-13 00:56:11 0x00158d0001dbbacb (0x00158d0001dbbacb): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
  zigbee2mqtt:info 2019-2-13 00:56:11 0x00158d000289ead0 (0x00158d000289ead0): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2019-2-13 00:56:11 0x00158d00031dc5fe (0x00158d00031dc5fe): 404006/404008/404004 - Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, opal white (Router)
  zigbee2mqtt:warn 2019-2-13 00:56:11 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-2-13 00:56:11 Allowing new devices to join.
  zigbee2mqtt:warn 2019-2-13 00:56:11 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-2-13 00:56:11 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-2-13 00:56:11 Connecting to MQTT server at mqtt://192.168.34.69
  zigbee2mqtt:info 2019-2-13 00:56:11 zigbee-shepherd ready
  zigbee2mqtt:info 2019-2-13 00:56:11 Connected to MQTT server
  zigbee2mqtt:info 2019-2-13 00:56:11 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-2-13 00:56:11 MQTT publish: topic 'zigbee2mqtt/0x00158d0002059fb6', payload '{"battery":100,"voltage":3035,"linkquality":0}'
  zigbee2mqtt:info 2019-2-13 00:56:11 MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa0a021f69', payload '{"state":"OFF","linkquality":0}'
  zigbee2mqtt:info 2019-2-13 00:56:11 MQTT publish: topic 'zigbee2mqtt/0x000d6f000f5e0b96', payload '{"occupancy":true,"tamper":true,"battery_low":false}'
  zigbee2mqtt:info 2019-2-13 00:56:11 MQTT publish: topic 'zigbee2mqtt/0x00158d000273a160', payload '{"temperature":24.28,"linkquality":0,"humidity":33.82,"pressure":1028,"battery":80,"voltage":2965}'
  zigbee2mqtt:info 2019-2-13 00:56:11 MQTT publish: topic 'zigbee2mqtt/0x00158d0001dbbacb', payload '{"occupancy":false,"linkquality":73,"battery":100,"voltage":3015}'
  zigbee2mqtt:info 2019-2-13 00:56:12 MQTT publish: topic 'zigbee2mqtt/0x00158d000289ead0', payload '{"contact":true,"linkquality":10,"battery":100,"voltage":3055}'
  zigbee2mqtt:error 2019-2-13 00:56:12 Failed to configure 0x000d6f000f5e0b96 (0x000d6f000f5e0b96) ('Error: request unsuccess: 112')
  zigbee2mqtt:info 2019-2-13 00:56:13 Succesfully configured 0x7cb03eaa0a021f69 (0x7cb03eaa0a021f69)
  zigbee2mqtt:info 2019-2-13 00:56:14 MQTT publish: topic 'zigbee2mqtt/0x000d6f000f5e0b96', payload '{"occupancy":true,"tamper":true,"battery_low":false}'
  zigbee2mqtt:info 2019-2-13 00:56:15 MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa0a021f69', payload '{"state":"OFF","linkquality":0}'
  zigbee2mqtt:info 2019-2-13 00:56:19 MQTT publish: topic 'zigbee2mqtt/0x00158d0001dbbacb', payload '{"occupancy":false,"linkquality":73,"battery":100,"voltage":3015}'

But I can't reach a device and integrate it into OH. They are recognized, but they are OFFLINE. All these devices are supported. What can I do more?

[image]

Koenkk, 2019-02-13:
Besides devices being marked as `offline` incorrectly, does anything else not work? It is known that this isn't reliable yet. 

Can you e.g. control the plug by sending to `zigbee2mqtt/0x7cb03eaa0a021f69/set` `ON`?

lolorc, 2019-02-13:
same here, all my osram plugs are marked offline, but they do work as controllable plug and router.

thissiac, 2019-02-14:
Crazy!!!!

Why have I spent so much time looking for a mistake that doesn't exist?

A round beer for everyone. I will pay!

Koenkk, 2019-02-14:
:beer:

closed 2019-02-14 by Koenkk

Nephiel, 2019-07-14:
I also have a Smart+ Plug that reports offline but otherwise works just fine. I suggest to add a note to the [Supported Devices page](https://www.zigbee2mqtt.io/devices/AB3257001NJ) with a link to this issue if it can't be fixed.

thissiac, 2019-07-14:
Same here. But instead of adding a note I suggest solving the problem. 

graphviz is the only tool I can identify the connection.  if I can not count on it, how should I find mistakes or improve the product in the future?