Koenkk/zigbee2mqtt #1084: Socat in Docker

axiom00, 2019-02-14:
Hi Koenkk,

Would it be possible to add Socat to the zigbee2mqtt container? I tried passing the virtual USB port into the container but zigbee2mqtt wasn't able to access it.

Followed the instructions from this how-to: [https://www.zigbee2mqtt.io/how_tos/how_to_esp8266_with_cc2530.html](url)

Thanks.

peterforeman, 2019-02-14:
You might want to check out my docker image which I'm using for exactly this:

https://hub.docker.com/r/forepe/zigbee2mqtt-socat

axiom00, 2019-02-14:
Thank you! I'll give it a try.

JLFN, 2019-02-14:
@axiom00 Please give me feedback and also what case you choose for it.

JLFN, 2019-02-14:
@peterforeman can you do a pull request to add your image to docs?

Koenkk, 2019-02-14:
I would rather prefer to have this Dockerfile in the zigbee2mqtt repo. In this way it can be updated automatically.

axiom00, 2019-02-15:
Thanks @peterforeman  for sharing your docker image. I was able to get it up and running pretty easily. I was a little stuck on where to set the Socat options, but a little digging around, I found that it is meant to be set in the environmental variables. 

@JLFN - My cc2530 is plugged into an old Rpi2 running ser2net which acts as the USB host. Zigbee2mqtt/Socat is running in a docker container on top of Ubuntu 18 in a VM. 

I agree with @Koenkk also. Would prefer to have socat added to the zigbee2mqtt repo for easy maintainability.

peterforeman, 2019-02-17:
Well, adding it should not really be a problem. The real "problem" lies in that docker, by nature, is only meant to run 1 process. So you need to run a sort-of supervisor script which monitors socat as well as zigbee2mqtt. My container does this, but this is very much different from a clean and simple docker container as the current zigbee2mqtt container is, which simply runs 1 process. I'm not really charmed by how it's solved now, although I do not know how to do this any other way.

Do you have any ideas how to implement this more cleanly? I would love to implement it!

And: zigbee2mqtt does not exist when a failure happens, for example:
```
 zigbee2mqtt:error 2019-2-17 10:45:02 Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
```

This prevents a proper restart of the container as well. But this is a separate issue.

Koenkk, 2019-02-18:
Would it be possible to run the socat in a separate container?

closed 2019-04-26 by stale[bot]

KiLLeRRaT, 2019-06-26:
Did this get anywhere?

I'm quite interested in having this run under my HASS.io instead of on my machine natively...

tlc76, 2021-03-15:
Hello, is it possible to build a version for Raspberry Pi architecture - arm32v7 ?
I have a remote zigbee gateway based on ESP8266 + CC2530/CC2591 boards with Esp Easy firmware.
When I run this image (https://hub.docker.com/r/forepe/zigbee2mqtt-socat) on RPi, I get the error:
`standard_init_linux.go:219: exec user process caused: exec format error`

@Koenkk : is it possible to add support in your docker zigbee2mqtt image for external gateways via the `tcp://ip:port` device type ?

Thanks!

peterforeman, 2021-03-15:
> Hello, is it possible to build a version for Raspberry Pi architecture - arm32v7?
> I have a remote zigbee gateway based on ESP8266 + CC2530/CC2591 boards with Esp Easy firmware.
> When I run this image [...]

The standard image from koenkk already has this. Don't use my image anymore :-) Use this in your configuration.yaml:

```
serial:
  port: 'tcp://1.2.3.4:5678'
```

tlc76, 2021-03-15:
Thanks a lot @peterforeman !
I reverted to the official zigbee2mqtt image and everything works fine with the proper configuration, of course :)
I was under the impression that when using the docker image, the serial/port setting is configurable only in docker-compose.yml, as "device".
I also had to change /dev/ttyACM0 to /dev/ttyAMA0 on the physical side.

docker-compose.yml:
```
    devices:
      - "/dev/ttyAMA0:/dev/ttyACM0"
```

configuration.yaml
```
serial:
  port: tcp://10.0.0.96:61775
```

Now everything works perfectly, thanks again!