Koenkk/zigbee2mqtt #1086: CC2531 as router - repairing process

h4nc, 2019-02-14:
I already know that you have to reflash a CC2530 if you have to repair it. It has no buttons to perform a reset.

Does a CC2531 flashes as a router have the ability to be reset, or do we also have to reflash it?

Koenkk, 2019-02-14:
@ptvoinfo has this been implemented?

ptvoinfo, 2019-02-14:
CC2531 can be re-paired pressing S2 for 5 seconds.
СС2530 (the latest version) can be re-paired if you power on/power off it three times (power on, wait 2 seconds, power off, repeat this cycle three times).
http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/

h4nc, 2019-02-14:
Nice to know that, would be great to add that to the zigbee2mqtt docs.

closed 2019-02-14 by h4nc

Koenkk, 2019-02-14:
@h4nc done! @ptvoinfo thanks!

h4nc, 2019-02-14:
Are there more noticable Upgrades in that Firmware.

It should we only update if we want that feature (repairing).?

ptvoinfo, 2019-02-15:
@h4nc One more thing: The previous version allowed to join for deleted devices.

h4nc, 2019-02-15:
You mean without permit_join enabled? So it was a bug.

ptvoinfo, 2019-02-15:
Yes, without permit_join.

h4nc, 2019-02-15:
Ok, I will update.

Should I delete the routers from the coord before I reflash them?

ptvoinfo, 2019-02-15:
no, this is not necessary

h4nc, 2019-02-15:
To add them to the network again after reflashing I would enable permit join and than power them up near the coord.

Is that the right way to do it?

ptvoinfo, 2019-02-15:
yes, exactly.

h4nc, 2019-03-04:
@Koenkk I wondering what would happen if you connect the cable wrong to the CC2531. The connector is not poka yoke, so users could plug in the wrong way.

Would you fry the board with a wrong connected cable or would it simple not flash.
If it would fry the board, we probably should add a WARNING to the How To Flash docs.
The only thing that indicated the right direction now is the picture, users could probably fail.

Koenkk, 2019-03-04:
@h4nc I have no idea, but a warning is always good, could you make a PR?

h4nc, 2019-03-04:
@Koenkk I'm trying for some minutes now, because this would be my first PR, but I don't get it.
It but be nice to know how it works for future requests.

Could you point me in the right direction?
It says I have to compare changes. So i selected dev to left and master to the right also search for this link

zigbee2mqtt.io/getting_started/flashing_the_cc2531.md

But I simple don't get what to do. I think there should open up that text file and than I can change it and later you can confirm or not confirm.

Koenkk, 2019-03-04:
@h4nc you can press the edit/pencil button here: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/getting_started/flashing_the_cc2531.md

h4nc, 2019-03-04:
@Koenkk So this is a pull request? And what about the comparison thing?

Koenkk, 2019-03-04:
I'm not sure what you mean with the comparison thing, but the pull request is what you need.

h4nc, 2019-03-04:
This

[image]

h4nc, 2019-03-04:
ok just postet my first PR, thanks @Koenkk