Koenkk/zigbee2mqtt #1087: Smartthings smart siren Sercomm SZ-SRN12N

JLFN, 2019-02-14:
@Koenkk  Is this product possible to add support?  iam thinking of buying it.
Product Model(VF-SKU) F-SRN-ZB-VF-1 / Sercomm SZ-SRN12N EAN nbr: 856418004990

https://shop.eno.de/en/vh-sercomm-sirene-zigbee-fuer-v-home.html

[image]

Koenkk, 2019-02-14:
Maybe, we don't know before we tried :)

JLFN, 2019-02-14:
@Koenkk i dont have a smartthing gateway is that needed to sniff it? etc?

nldroid, 2019-02-14:
@JLFN, did you try this?

https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

JLFN, 2019-02-14:
@nldroid I dont want to buy a device if i need to get the smartthings bridge also just to sniff it.

nldroid, 2019-02-14:
Ah, didn't read the 'iam thinking of buying it' part ;) 

The problem is that you'll never know until somebody tried it.

mclaudiopt, 2019-03-03:
hello.
i have this camera. someway i can help?
i also do not have vodafone...so i cannot pair it...

mclaudiopt, 2019-04-30:
hello
any solution?

mclaudiopt, 2019-05-03:
i have this sirene so i can try to snif if someone gives me how to do it...i also want to use zigbee2mqtt

i also have the camera wich has WIFI...but dont know the password

JLFN, 2019-05-03:
> i have this sirene so i can try to snif if someone gives me how to do it...i also want to use zigbee2mqtt
> 
> i also have the camera wich has WIFI...but dont know the password

Hey 
You need to sniff it . You got 2 cc2531 or cc2531 one for zigbee2mqtt and one for sniffing?
Guide here
[](https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)
You have the file here that you need to use to flash your cc2531 for sniffing.
[](https://github.com/kirovilya/files/blob/master/2017_09_26_zboss_sniffer_windows_64bit.zip)
[](https://www.youtube.com/watch?v=4vG4CVNAm_A)

mclaudiopt, 2019-05-03:
hi. 
i only have one cc2531.
so i need to have another one, right?
i have a PI3b+ but i can use windows

JLFN, 2019-05-04:
> hi.
> i only have one cc2531.
> so i need to have another one, right?
> i have a PI3b+ but i can use windows

Are you using one cc2531 or can you spare while you sniff? Did you try debug mode to see if you can pair the siren?

mclaudiopt, 2019-05-04:
hi. i only have one stick. i tried to enter in debug mode but nothing happens on the prompt (dash prompt)

JLFN, 2019-05-04:
> hi. i only have one stick. i tried to enter in debug mode but nothing happens on the prompt (dash prompt)
Did you have the siren near and then press the pairing button?

mclaudiopt, 2019-05-04:
yes i did :-(
i will try again and post here my step by step

JLFN, 2019-05-04:
> yes i did :-(
> i will try again and post here my step by step

Do you own a smartthing bridge?

mclaudiopt, 2019-05-04:
i had one...but not any more...

mclaudiopt, 2019-05-05:
anyway to try anything of i must have 2 cc2531?

JLFN, 2019-05-05:
> anyway to try anything of i must have 2 cc2531?

you can sniff to see if its sending signal

mclaudiopt, 2019-05-05:
can you please teach how to do it?
if i try to pair it it identify but cannot pair...

JLFN, 2019-05-05:
> can you please teach how to do it?
> if i try to pair it it identify but cannot pair...

Guide here
[Zigbee2mqtt sniff guide](https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)
You have the file here that you need to use to flash your cc2531 for sniffing.
[Files with sniffer firmware](https://github.com/kirovilya/files/blob/master/2017_09_26_zboss_sniffer_windows_64bit.zip)
[Youtube guide](https://www.youtube.com/watch?v=4vG4CVNAm_A)

JLFN, 2019-05-07:
@Koenkk here is @mclaudiopt troubleshot device.js
https://paste.ubuntu.com/p/8phxCrM4P6/
And his debug after pairing. 
https://paste.ubuntu.com/p/69Qxs5BtJg/
Database file.
https://paste.ubuntu.com/p/xk7WkJmcXV/

Koenkk, 2019-05-08:
@JLFN great, does it work now?

JLFN, 2019-05-08:
@mclaudiopt  do those step. Any suggest @Koenkk 
database
```
{"id":5,"type":"Router","ieeeAddr":"0xd0cf5efffe475d33","nwkAddr":48174,"manufId":4401,"epList":[1],"status":"online","joinTime":1557267320,"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,1280,1282,2821],"outClusterList":[25],"clusters":{}}},"_id":"VYZxZogEMvzcmm0t"}
{"id":5,"type":"Router","ieeeAddr":"0xd0cf5efffe475d33","nwkAddr":48174,"manufId":4401,"manufName":"Sercomm Corp.","powerSource":"Mains (single phase)","modelId":"SZ-SRN12N","epList":[1],"status":"online","joinTime":1557267320,"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,1280,1282,2821],"outClusterList":[25],"clusters":{}}},"_id":"VYZxZogEMvzcmm0t"}
{"$$deleted":true,"_id":"VYZxZogEMvzcmm0t"}
{"id":5,"type":"Router","ieeeAddr":"0xd0cf5efffe475d33","nwkAddr":39567,"manufId":4401,"epList":[1],"status":"online","joinTime":1557267447,"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,1280,1282,2821],"outClusterList":[25],"clusters":{}}},"_id":"BUW4yvHg4J6IauiT"}
{"id":5,"type":"Router","ieeeAddr":"0xd0cf5efffe475d33","nwkAddr":39567,"manufId":4401,"manufName":"Sercomm Corp.","powerSource":"Mains (single phase)","modelId":"SZ-SRN12N","epList":[1],"status":"online","joinTime":1557267447,"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,1280,1282,2821],"outClusterList":[25],"clusters":{}}},"_id":"BUW4yvHg4J6IauiT"}
{"$$deleted":true,"_id":"BUW4yvHg4J6IauiT"}

```
  device.js"alt1"
```.js
    },
    //SZ-SRN12N
    {
        zigbeeModel: ['SZ-SRN12N'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
        model: 'SZ-SRN12N', // Vendor model number, look on the device for a model number
        vendor: 'Sercomm Corp', // Vendor of the device (only used for documentation and startup logging)
        description: 'Smartthings Siren ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
        supports: 'Alarm siren', // Actions this device supports (only used for documentation)
        fromZigbee: [fz.ignore_onoff_change, fz.state],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];

            execute(device, actions, callback);
        },
    },
```
device.js"alt2"
 Same as E1603 IKEA TRADFRI i belive its worth a shut cause its "powerSource":"Mains (single phase)"
```.js
     ///SZ-SRN12N
    {
        zigbeeModel: ['SZ-SRN12N'],
        model: 'SZ-SRN12N',
        vendor: 'Sercomm Corp',
        description: 'SmartThings Siren',
        supports: 'on/off',
        fromZigbee: [fz.ignore_onoff_change, fz.state],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
     },
```
1.First you the device.js "alt1" and rename it to device.js first use file is in place you restart zigbee2mqtt and you pair it again its very important you have  zigbee_shepherd_debug = true and  zigbee_shepherd_devices= Yes
  You take the whole log and describe what log is from pairing and put in on https://paste.ubuntu.com/ and then save post the link
2. next step wait 1 minut make sure the siren is near paste database info to https://paste.ubuntu.com/  and then save post the link
3. restart zigbee2mqtt and wait 1 minute to make sure it has joined again , now long press button to make the Siren sound so it change to first ON and then OFF. paste this new log to https://paste.ubuntu.com/  and then save post the link.
4. You have two device added that i added to you packages/ yesterday one as a switch and one as a sensor that i added yesterday with 0xd0cf5efffe475d33
5. Try the switch " switch.siren " to see if i make sound then you change in home assistant from on to off.

Repeat this steps with device.js"alt2" that you the same settings as E1603 IKEA TRADFRI control outlet (on/off)
```.yaml
switch:
  - platform: "mqtt"
    name: Siren Alarm
    state_topic: "zigbee2mqtt/0xd0cf5efffe475d33>"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_off: "OFF"
    payload_on: "ON"
    value_template: "{{ value_json.state }}"
    command_topic: "zigbee2mqtt/0xd0cf5efffe475d33/set"

sensor:
  - platform: "mqtt"
    name: Siren Linkquality
    state_topic: "zigbee2mqtt/0xd0cf5efffe475d33"
    availability_topic: "zigbee2mqtt/bridge/state"
    unit_of_measurement: "-"
    value_template: "{{ value_json.linkquality }}"
```

JLFN, 2019-05-08:
@Koenkk  it crashes with this code for devices.js can you please if there any typos.

Koenkk, 2019-05-09:
Where can i found the log of the crash?

JLFN, 2019-05-11:
@Koenkk 
We did new try it worked better the receive was receive.
there is the log.
https://paste.ubuntu.com/p/W3t2973VS5/
database.db
```{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001938a2f5","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1557588620,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"WGxH8dUdm9LteOJM"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001938a2f5","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1557588620,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"WGxH8dUdm9LteOJM"}
{"id":2,"type":"Router","ieeeAddr":"0xd0cf5efffe475d33","nwkAddr":8195,"manufId":4401,"epList":[1],"status":"online","joinTime":1557588650,"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,1280,1282,2821],"outClusterList":[25],"clusters":{}}},"_id":"cVkfjnQMKAYTcYYC"}
{"id":2,"type":"Router","ieeeAddr":"0xd0cf5efffe475d33","nwkAddr":8195,"manufId":4401,"manufName":"Sercomm Corp.","powerSource":"Mains (single phase)","modelId":"SZ-SRN12N","epList":[1],"status":"online","joinTime":1557588650,"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,1280,1282,2821],"outClusterList":[25],"clusters":{}}},"_id":"cVkfjnQMKAYTcYYC"}
{"$$deleted":true,"_id":"cVkfjnQMKAYTcYYC"}
```

Koenkk, 2019-05-13:
- What do you mean with 'Not saved?'
- What happens when you switch the device? Can you provide the log of that?

mclaudiopt, 2019-05-18:
here what i have able to to
https://paste.ubuntu.com/p/9KcyM9CNVZ/

mclaudiopt, 2019-05-18:
@Koenkk @JLFN 
here what i have able to to
https://paste.ubuntu.com/p/9KcyM9CNVZ/

mclaudiopt, 2019-05-19:
here is extended log debug
https://paste.ubuntu.com/p/zN3HjnsScw/

mclaudiopt, 2019-06-03:
hello
what else can i do for making the sirene work?

JLFN, 2019-06-04:
@mclaudiopt, @Koenkk  is still waiting for this info.
 What happens when you switch the device with home assistant? Can you provide the debug log of that?
Set Configuration  both on config and your hassio plugin.
homeassistant :false

```
Settings to try with home asssistant
```.yaml
switch:
  - platform: "mqtt"
    name: Siren Alarm
    state_topic: "zigbee2mqtt/0xd0cf5efffe475d33>"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_off: "OFF"
    payload_on: "ON"
    value_template: "{{ value_json.state }}"
    command_topic: "zigbee2mqtt/0xd0cf5efffe475d33/set"

```

mclaudiopt, 2019-06-23:
i give up. if someone want i can sell it

JLFN, 2019-07-05:
> i give up. if someone want i can sell it
I got my Heiman siren working with Zoenkk
try for device.js
```    {
        zigbeeModel: ['Z-SRN12N'],
        model: 'SZ-SRN12N',
        vendor: 'SmartThings Siren',
        description: 'Smart siren',
        supports: 'warning',
        fromZigbee: [ fz.ignore_iaszone_change],
        toZigbee: [tz.warning],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 23}, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
            ];

            execute(device, actions, callback, 1000);
        },
    },````

use this file toZigbee.js
https://raw.githubusercontent.com/Koenkk/zigbee-shepherd-converters/master/converters/toZigbee.js

mclaudiopt, 2019-07-05:
@JLFN thanks for the support
so i should copy that file on the link, rename it devices.js and add the content above, then try to pair. is this correct?

JLFN, 2019-07-06:
Yes i showed you many times.

JLFN, 2019-07-06:
> @JLFN thanks for the support
> so i should copy that file on the link, rename it devices.js and add the content above, then try to pair. is this correct?

No where does it say rename? Are you not reading?

JLFN, 2019-07-06:
@mclaudiopt You need to read and follow instructions if you want support.

mclaudiopt, 2019-07-06:
@JLFN here is the log
```
  zigbee2mqtt:warn 7/6/2019, 8:22:43 PM This can be ignored if the device is working properly
  zigbee2mqtt:info 7/6/2019, 8:22:45 PM New device 'undefined' with address 0xd0cf5efffe475d33 connected!
  zigbee2mqtt:info 7/6/2019, 8:22:45 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xd0cf5efffe475d33","meta":{}}'
  zigbee2mqtt:warn 7/6/2019, 8:22:45 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:22:45 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/6/2019, 8:22:55 PM Message without device!
  zigbee2mqtt:info 7/6/2019, 8:23:20 PM Device incoming...
  zigbee2mqtt:info 7/6/2019, 8:23:20 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 7/6/2019, 8:23:20 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:23:20 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/6/2019, 8:23:20 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:23:20 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 7/6/2019, 8:23:42 PM MQTT publish: topic 'zigbee2mqtt/0x286d970001010cd0', payload '{"battery":"0.00","voltage":2400,"linkquality":68,"temperature":21.14}'
  zigbee2mqtt:warn 7/6/2019, 8:23:50 PM Failed to configure 0x286d970001010cd0 (0x286d970001010cd0) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #3)
  zigbee2mqtt:warn 7/6/2019, 8:23:50 PM This can be ignored if the device is working properly
  zigbee2mqtt:error 7/6/2019, 8:23:52 PM Failed to find device with ieeeAddr: '0xd0cf5efffe475d33'
  zigbee2mqtt:info 7/6/2019, 8:23:52 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"0xd0cf5efffe475d33"}'
  zigbee2mqtt:error 7/6/2019, 8:23:56 PM Failed to find device with ieeeAddr: '0xd0cf5efffe475d33'
  zigbee2mqtt:info 7/6/2019, 8:23:56 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"0xd0cf5efffe475d33"}'
```

mclaudiopt, 2019-07-06:
```
'{"battery":"0.00","voltage":2400,"linkquality":63,"temperature":21.21}'
  zigbee2mqtt:info 7/6/2019, 8:28:31 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00033ecd10', payload '{"temperature":23,"linkquality":18,"humidity":71.85,"battery":91,"voltage":2985}'
  zigbee2mqtt:info 7/6/2019, 8:28:35 PM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 7/6/2019, 8:28:35 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190425,"log_level":"info","permit_join":false}'
  zigbee2mqtt:info 7/6/2019, 8:28:36 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 7/6/2019, 8:28:36 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190425,"log_level":"info","permit_join":true}'
  zigbee2mqtt:warn 7/6/2019, 8:28:39 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:28:39 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/6/2019, 8:29:07 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:29:07 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 7/6/2019, 8:29:08 PM Device incoming...
  zigbee2mqtt:info 7/6/2019, 8:29:08 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 7/6/2019, 8:29:08 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:29:08 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/6/2019, 8:29:08 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:29:08 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 7/6/2019, 8:29:12 PM MQTT publish: topic 'zigbee2mqtt/0x286d970001010cd0', payload '{"battery":"0.00","voltage":2400,"linkquality":70,"temperature":21.22}'
  zigbee2mqtt:warn 7/6/2019, 8:29:17 PM Failed to remove '0xd0cf5efffe475d33', trying force remove...
  zigbee2mqtt:info 7/6/2019, 8:29:17 PM Force removed 0xd0cf5efffe475d33
  zigbee2mqtt:info 7/6/2019, 8:29:17 PM Successfully removed 0xd0cf5efffe475d33
  zigbee2mqtt:info 7/6/2019, 8:29:17 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0xd0cf5efffe475d33"}'
  zigbee2mqtt:info 7/6/2019, 8:29:36 PM New device 'undefined' with address 0xd0cf5efffe475d33 connected!
  zigbee2mqtt:info 7/6/2019, 8:29:36 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xd0cf5efffe475d33","meta":{}}'
  zigbee2mqtt:warn 7/6/2019, 8:29:36 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:29:36 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/6/2019, 8:29:46 PM Message without device!
  zigbee2mqtt:info 7/6/2019, 8:30:00 PM Starting network scan...
  zigbee2mqtt:warn 7/6/2019, 8:30:02 PM Network scan timeout, skipping outstanding lqi scans for '0x000d6ffffebb1f5d'
  zigbee2mqtt:warn 7/6/2019, 8:30:02 PM Network scan timeout, skipping outstanding rtg scans for '0x000d6ffffebb1f5d'
  zigbee2mqtt:info 7/6/2019, 8:30:02 PM MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];
  "0x00124b0018e1a997" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{0x00124b0018e1a997|0x0 |No model information available|online (2019-07-06T20:30:02+01:00)}"];
  "0x00158d0002d45645" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d0002d45645|0xface |Xiaomi MiJia temperature & humidity sensor (WSDCGQ01LM)|online (unknown)}"];
  "0x000d6ffffebb1f5d" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{0x000d6ffffebb1f5d|0x69c5  no lqi no routes|IKEA TRADFRI signal repeater (E1746)|offline (unknown)}"];
  "0x000d6ffffebb1f5d" -> "0x00124b0018e1a997" [style="dotted", weight=0, color="#994444", label="1"]
  "0x00158d00033ecd10" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d00033ecd10|0xb017 |Xiaomi MiJia temperature & humidity sensor (WSDCGQ01LM)|online (00:01:30s ago)}"];
  "0x00158d00033ecd10" -> "0x00124b0018e1a997" [style="dashed", weight=0, color="#994444", label="15"]
  "0x286d970001010cd0" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x286d970001010cd0|0x8c4d |SmartThings Multipurpose sensor (2018 model) (IM6001-MPP01)|offline (00:00:49s ago)}"];
  "0x286d970001010cd0" -> "0x00124b0018e1a997" [style="dashed", weight=0, color="#994444", label="68"]
}'
  zigbee2mqtt:info 7/6/2019, 8:30:10 PM Device incoming...
  zigbee2mqtt:info 7/6/2019, 8:30:10 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 7/6/2019, 8:30:10 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:30:10 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/6/2019, 8:30:10 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/6/2019, 8:30:10 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/6/2019, 8:30:10 PM Failed network lqi scan for device: '0x000d6ffffebb1f5d' with error: 'Error: Timed out after 10000 ms'
  zigbee2mqtt:info 7/6/2019, 8:30:18 PM MQTT publish: topic 'zigbee2mqtt/0x286d970001010cd0', payload '{"battery":"0.00","voltage":2400,"linkquality":76,"temperature":21.26}'
  zigbee2mqtt:warn 7/6/2019, 8:30:20 PM Failed network rtg scan for device: '0x000d6ffffebb1f5d' with error: 'Error: Timed out after 10000 ms'
```

JLFN, 2019-07-07:
The log show that you havent follow the instructions.

JLFN, 2019-07-07:
The log show that you havent follow the instructions devices.js havent been loaded with information that should be added in the file.

mclaudiopt, 2019-07-07:
@JLFN followed all instructions
check log
```
[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-07-07T18:23:51: PM2 log: Launching in no daemon mode
2019-07-07T18:23:51: PM2 log: App [npm:0] starting in -fork mode-
2019-07-07T18:23:51: PM2 log: App [npm:0] online
> zigbee2mqtt@1.4.0 start /app
> node index.js
  zigbee2mqtt:info 7/7/2019, 6:23:53 PM Logging to directory: '/share/zigbee2mqtt/log/2019-07-07.18-23-52'
  zigbee2mqtt:info 7/7/2019, 6:23:53 PM Starting zigbee2mqtt version 1.4.0 (commit #unknown)
  zigbee2mqtt:info 7/7/2019, 6:23:53 PM Starting zigbee-shepherd
  zigbee2mqtt:info 7/7/2019, 6:23:54 PM zigbee-shepherd started
  zigbee2mqtt:info 7/7/2019, 6:23:54 PM Coordinator firmware version: '20190425'
  zigbee2mqtt:info 7/7/2019, 6:23:54 PM Currently 4 devices are joined:
  zigbee2mqtt:info 7/7/2019, 6:23:54 PM Xiaomi 1a (0x00158d0002d45645): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
  zigbee2mqtt:info 7/7/2019, 6:23:54 PM Xiaomi RC (0x00158d00033ecd10): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
  zigbee2mqtt:info 7/7/2019, 6:23:54 PM Sensor Smarthings (0x286d970001010cd0): IM6001-MPP01 - SmartThings Multipurpose sensor (2018 model) (EndDevice)
  zigbee2mqtt:info 7/7/2019, 6:23:54 PM Repetidor IKEA (0x000d6ffffebb1f5d): E1746 - IKEA TRADFRI signal repeater (Router)
  zigbee2mqtt:warn 7/7/2019, 6:23:54 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 7/7/2019, 6:23:54 PM Allowing new devices to join.
  zigbee2mqtt:warn 7/7/2019, 6:23:54 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 7/7/2019, 6:23:54 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 7/7/2019, 6:23:54 PM Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:info 7/7/2019, 6:23:55 PM zigbee-shepherd ready
  zigbee2mqtt:info 7/7/2019, 6:23:55 PM Connected to MQTT server
  zigbee2mqtt:info 7/7/2019, 6:23:55 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 7/7/2019, 6:23:55 PM MQTT publish: topic 'zigbee2mqtt/Xiaomi 1a', payload '{"battery":91,"voltage":2985,"temperature":22.95,"humidity":72.7,"linkquality":2}'
  zigbee2mqtt:info 7/7/2019, 6:23:55 PM MQTT publish: topic 'zigbee2mqtt/Xiaomi RC', payload '{"temperature":22.66,"linkquality":13,"humidity":73.11,"battery":91,"voltage":2985}'
  zigbee2mqtt:info 7/7/2019, 6:23:55 PM MQTT publish: topic 'zigbee2mqtt/Sensor Smarthings', payload '{"battery":"NaN","voltage":null,"linkquality":63,"temperature":20.92,"contact":true,"battery_low":false}'
  zigbee2mqtt:info 7/7/2019, 6:23:55 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190425,"log_level":"info","permit_join":true}'
  zigbee2mqtt:warn 7/7/2019, 6:24:07 PM Failed to configure Sensor Smarthings (0x286d970001010cd0) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
  zigbee2mqtt:warn 7/7/2019, 6:24:07 PM This can be ignored if the device is working properly
  zigbee2mqtt:info 7/7/2019, 6:24:36 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 7/7/2019, 6:24:36 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190425,"log_level":"info","permit_join":true}'
  zigbee2mqtt:warn 7/7/2019, 6:25:21 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:21 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:21 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:21 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:22 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:23 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:23 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:23 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:23 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:23 PM Device with modelID 'SZ-SRN12N' is not supported.
[... cut ...]
  zigbee2mqtt:info 7/7/2019, 6:25:23 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 7/7/2019, 6:25:23 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:23 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:23 PM Device with modelID 'SZ-SRN12N' is not supported.
  zigbee2mqtt:warn 7/7/2019, 6:25:23 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/7/2019, 6:25:34 PM Message without device!
```

mclaudiopt, 2019-07-07:
aditional
```
  zigbee2mqtt:error 7/7/2019, 6:26:57 PM Failed to find device with ieeeAddr: '0xd0cf5efffe475d33'
  zigbee2mqtt:info 7/7/2019, 6:26:57 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"0xd0cf5efffe475d33"}'
  zigbee2mqtt:warn 7/7/2019, 6:26:57 PM Cannot handle MQTT message on 'zigbee2mqtt/0xd0cf5efffe475d33/set' with data '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}'
  zigbee2mqtt:info 7/7/2019, 6:26:57 PM MQTT publish: topic 'zigbee2mqtt/Sensor Smarthings', payload '{"battery":"NaN","voltage":null,"linkquality":68,"temperature":20.97,"contact":true,"battery_low":false}'
  zigbee2mqtt:warn 7/7/2019, 6:27:05 PM Failed to configure Sensor Smarthings (0x286d970001010cd0) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #2)
  zigbee2mqtt:warn 7/7/2019, 6:27:05 PM This can be ignored if the device is working properly
  zigbee2mqtt:info 7/7/2019, 6:27:16 PM Successfully reenabled joining
```

mclaudiopt, 2019-07-07:
when i it the switch on the web interface
```
  zigbee2mqtt:warn 7/7/2019, 6:28:11 PM This can be ignored if the device is working properly
  zigbee2mqtt:error 7/7/2019, 6:28:44 PM Failed to find device with ieeeAddr: '0xd0cf5efffe475d33'
  zigbee2mqtt:info 7/7/2019, 6:28:44 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"0xd0cf5efffe475d33"}'
  zigbee2mqtt:warn 7/7/2019, 6:28:44 PM Cannot handle MQTT message on 'zigbee2mqtt/0xd0cf5efffe475d33/set' with data '{"warning": {"duration": 0, "mode": "stop", "strobe": false}}'
  zigbee2mqtt:error 7/7/2019, 6:28:46 PM Failed to find device with ieeeAddr: '0xd0cf5efffe475d33'
  zigbee2mqtt:info 7/7/2019, 6:28:46 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"0xd0cf5efffe475d33"}'
  zigbee2mqtt:warn 7/7/2019, 6:28:46 PM Cannot handle MQTT message on 'zigbee2mqtt/0xd0cf5efffe475d33/set' with data '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}'
  zigbee2mqtt:info 7/7/2019, 6:29:09 PM MQTT publish: topic 'zigbee2mqtt/Sensor Smarthings', payload '{"battery":"NaN","voltage":null,"linkquality":68,"temperature":20.96,"contact":true,"battery_low":false}'
```

JLFN, 2019-07-07:
@mclaudiopt can you get the logs that it has been paired also print screen where you put the device. js info and tozigbee.js

martin3000, 2019-07-10:
I bought the Sercomm SZ-SRN12N and it seems that it only can beep. Is this right? No other sound?

mclaudiopt, 2019-07-10:
i dont know but i cannot make it work it zigbee2mqtt yet :(

JLFN, 2019-07-10:
@martin3000 Have you added the code? for device.js and ToZigbee.js and be able to pair and it keeped paired?
You can try the payload 
{"warning": {"duration": 20, "mode": "emergency", "strobe": false, "level": "very_high"}}
you should be able to control it with
zigbee2mqtt/[DEVICE_ID]/set   {"warning": {"duration": 10, "mode": "emergency", "strobe": false, "level": "very_high"}
duration = time in seconds max 1800
mode = stop, burglar, fire, emergency, police_panic, fire_panic or emergency_panic
strobe = true or false
level = low, medium, high, very high

martin3000, 2019-07-10:
Sorry, I use ZHA.
This is my comand:
{
"ieee":"d0:cf:5e:ff:fe:43:cc:75"
,"endpoint_id":1
,"cluster_id":1282
,"cluster_type":"in"
,"command":1
,"command_type":"server"
,"args": "0"
}

mclaudiopt, 2019-07-11:
@JLFN anyway i can do further ? thanks

mclaudiopt, 2019-07-11:
tried this with o sucess
[image]

JLFN, 2019-07-12:
@mclaudiopt is you device still paired?

mclaudiopt, 2019-07-13:
hi @JLFN dont know why but it does not pair now :(

mclaudiopt, 2019-07-13:
@JLFN does this mean that it is paired?
https://paste.ubuntu.com/p/qx3pHznzMT/

JLFN, 2019-07-14:
@mclaudiopt upgrade to 1.5.0 and try again

mclaudiopt, 2019-07-14:
@JLFN still not avaialable on my hassio
[image]

mclaudiopt, 2019-07-21:
hi there
this is the status after 1.5.1
```
  zigbee2mqtt:warn 7/21/2019, 9:42:05 PM This can be ignored if the device is working properly
  zigbee2mqtt:error 7/21/2019, 9:42:20 PM Failed to find device with ieeeAddr: '0xd0cf5efffe475d33'
  zigbee2mqtt:info 7/21/2019, 9:42:20 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"0xd0cf5efffe475d33"}'
  zigbee2mqtt:warn 7/21/2019, 9:42:20 PM Cannot handle MQTT message on 'zigbee2mqtt/0xd0cf5efffe475d33/set' with data '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}'
  zigbee2mqtt:error 7/21/2019, 9:42:21 PM Failed to find device with ieeeAddr: '0xd0cf5efffe475d33'
  zigbee2mqtt:info 7/21/2019, 9:42:21 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"0xd0cf5efffe475d33"}'
  zigbee2mqtt:warn 7/21/2019, 9:42:21 PM Cannot handle MQTT message on 'zigbee2mqtt/0xd0cf5efffe475d33/set' with data '{"warning": {"duration": 0, "mode": "stop", "strobe": false}}'
  zigbee2mqtt:error 7/21/2019, 9:42:22 PM Failed to find device with ieeeAddr: '0xd0cf5efffe475d33'
  zigbee2mqtt:info 7/21/2019, 9:42:22 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"0xd0cf5efffe475d33"}'
  zigbee2mqtt:warn 7/21/2019, 9:42:22 PM Cannot handle MQTT message on 'zigbee2mqtt/0xd0cf5efffe475d33/set' with data '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}
```
@JLFN

JLFN, 2019-07-22:
@mclaudiopt was the settings home assistant =false and the device paired when you did this?

mclaudiopt, 2019-07-22:
hi !
yes it was :-( can you maybe teamview?

[54 more replies]

closed 2019-09-28 by stale[bot]