Koenkk/zigbee2mqtt #1089: IKEA and Philips lights won't connect (after a few dags without power to sniffer)

exetico, 2019-02-14:
Hi all

I had some issues for a couple of days, but i finally fixed that today, after many issues.

Overall everything is working fine - and all my buttons is OK again, however, both my IKEA and Philips lights are not working as expected with zigbee2mqtt, after i managed to get the system up and running. 

See the output here:
```
2019-2-14 17:45:07 - info: Logging to directory: '/app/data/log/2019-02-14.17-45-07'
2019-2-14 17:45:08 - info: Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
2019-2-14 17:45:08 - info: Starting zigbee-shepherd
2019-2-14 17:45:09 - info: zigbee-shepherd started
2019-2-14 17:45:09 - info: Coordinator firmware version: '20180815'
2019-2-14 17:45:09 - info: Currently 22 devices are joined:
2019-2-14 17:45:09 - info: switch1_mijia (0x00158d00023cd99e): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2019-2-14 17:45:09 - info: switch1_aqara (0x00158d0001712dbc): WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)
2019-2-14 17:45:09 - info: switch2_mijia (0x00158d0001b184ff): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2019-2-14 17:45:09 - info: switch3_mijia (0x00158d00023d1dda): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2019-2-14 17:45:09 - info: switch4_mijia (0x00158d00023cda09): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2019-2-14 17:45:09 - info: switch5_mijia (0x00158d00023d1db8): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2019-2-14 17:45:09 - info: switch6_mijia (0x00158d0001f36176): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2019-2-14 17:45:09 - info: switch7_mijia (0x00158d0001f3ac20): unkown - undefined unknown (EndDevice)
2019-2-14 17:45:09 - info: switch8_mijia (0x00158d00019d3d0c): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2019-2-14 17:45:09 - info: switch9_mijia (0x00158d0001b185f5): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2019-2-14 17:45:09 - info: switch2_aqara (0x00158d000103a06f): MFKZQ01LM - Xiaomi Mi/Aqara smart home cube (EndDevice)
2019-2-14 17:45:09 - info: sensor1_aqara (0x00158d00023482d2): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
2019-2-14 17:45:09 - info: light1_ikea (0x90fd9ffffe16a1f7): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
2019-2-14 17:45:09 - info: light2_ikea (0x90fd9ffffe4b6e32): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
2019-2-14 17:45:09 - info: light3_ikea (0x000b57fffef3b7e4): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
2019-2-14 17:45:09 - info: light4_ikea (0x90fd9ffffe00f938): LED1545G12 - IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (Route
r)
2019-2-14 17:45:09 - info: light5_ikea (0x90fd9ffffe0d1286): LED1536G5 - IKEA TRADFRI LED bulb E12/E14 400 lumen, dimmable, white spectrum, opal white (Router
)
2019-2-14 17:45:09 - info: light6_ikea (0x90fd9ffffe4d91e1): LED1545G12 - IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (Route
r)
2019-2-14 17:45:09 - info: unknown (0x0017880104e82196): 8718696598283 - Philips Hue white ambiance GU10 (Router)
2019-2-14 17:45:09 - info: light3_philips (0x0017880104e821a0): 8718696598283 - Philips Hue white ambiance GU10 (Router)
2019-2-14 17:45:09 - info: light4_philips (0x0017880103aeb867): 8718696598283 - Philips Hue white ambiance GU10 (Router)
2019-2-14 17:45:09 - info: sensor1_mijia (0x00158d0001a49a3c): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2019-2-14 17:45:09 - warn: `permit_join` set to  `true` in configuration.yaml.
2019-2-14 17:45:09 - warn: Allowing new devices to join.
2019-2-14 17:45:09 - warn: Set `permit_join` to `false` once you joined all devices.
2019-2-14 17:45:09 - info: Zigbee: allowing new devices to join.
2019-2-14 17:45:09 - info: Connecting to MQTT server at mqtt://IP-REMOVED
2019-2-14 17:45:09 - info: zigbee-shepherd ready
2019-2-14 17:45:09 - info: Connected to MQTT server
```

As you can see, all lights is loading (excepts a single light, which i removed for testing/debug purpose after is won't work as expected).

Are you able to point me in the right direction?

I didn't change a single thing - however, i updated my zigbee2mqtt container.

Any ideas?
Are it able to "time out" or something like that, if it's not active for a specific amount of time, or?

Koenkk, 2019-02-14:
What's the error/doesn't work here? (I don't see any  in the log)

exetico, 2019-02-14:
Waow. Sorry 😅...

After I turn on the spot, i can only find this in the logs, related to the event:
```
2019-2-14 19:51:13 - info: Device incoming...
2019-2-14 19:51:13 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
```
And it's the same for all the spots (both IKEA and Philips)

Koenkk, 2019-02-15:
Does the device turn on?

exetico, 2019-02-15:
All spots lights up as they should, but non of them are controlable by zigbee2mqtt - and nothing else is showing up in the logs, related to them... And it's more than a single one :-/ It's multiple, and both IKEA and Philips-branded.

Any ideas? One thing could be to clean the DB, but it will be a real pain with all my buttons which i would need to pair again :-( For that reason, i hoped that someone had the solution for me ☀️😅

Koenkk, 2019-02-15:
So what is logged when you try to turn a device on/off?

exetico, 2019-02-17:
_Felling a bit stupid at the moment..._

After debugging a bit more i could get all IKEA-lamps to work as expected, by using manual controls in Home Assistant, but not through the normal GUI (Lovelace ofc.). For that reason, i started up MQTT.fx like i normally do if my MQTT-solution are facing some issues.

I turned on every light while i was debugging - and all IKEA-spots worked as they should, if i manually sent the commands to zigbee2mqtt.

After an hour i spotted that i had swapped spot 1-2 with 3-4, and vice versa. Sooooo... After sending commands to the right spots, my Philips lights was working too.

However, Home Assistant doesn't work - but that's another store (and not a thing which should be located here at `zigbee2mqtt` :-)

I have a small request for a feature, if it's possible.

Would it be possible to report `unique_id` in the logs, and maybee identifier (is possible) in the `device incoming` message? I would love that, to give a better idea of which spot is connecting to the network.
, in the me to walk through all my devices, and test specific things. With 23 zigbee devices, it's hard to be 100% sure, whether it's a light, a button or another device :-)

Btw- Thanks for the answers.

_Did i learn something?_
Sure... Use **MQTT.fx** if something isn't working properly, and don't trust Home Assistant while you're debugging.

exetico, 2019-02-17:
_Another question..._
In near feature, i'll start using the [group-feature](https://www.zigbee2mqtt.io/information/groups.html). If i flash my CC2531 to the latest firmware, should i then pair all devices once again, or?

Koenkk, 2019-02-17:
The device incoming message do not contain what device is incoming (so we don't know the ieeeAddr). 

You can also reflash without re-pairing with the latest dev firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin

closed 2019-02-17 by Koenkk