Koenkk/zigbee2mqtt #1090: Could not send data to End device after connect to a router

dzungpv, 2019-02-14:
I am running the latest release version 1.1.1 and the edge version.
Coordinator is CC2530+CC2591 firmware version: '20190109'.

Step to produce bug:

First connect a zigbee end device (ZED)  (custom hardware running on cc2630 chip using TI sample switch , send report state each 1 minute) to coordinator (ZC) directly. Now it could be control normally with on/ off. turn on/off the ZED or ZC not affect.

Now connect 1 CC2530 router running SampleLight project or ptvoinfo firmware to the network. 
two of them work normally. Now turn off the ZC for 5 minute, ZED will find the router and connect to it.
And then turn on the ZC. From that moment the ZED switch could not control any more, it show:
```
Zigbee publish to device '0x00xxxx000d801379', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 1 failed with error Error: AF data request fails, status code: 240. MAC transaction expired.
```
Please note that from an CC2531 sniffer the network still running normally, ZED could send report each minute and zigbee2mqtt show it:
```
2019-2-14 17:45:47 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'DNCKAT_S004' (0x00xxxx000d801379)
```
Logs: https://pastebin.com/iZKBimxX

This bug prevent me to use any router to extend the network. Also cause the Livolo switch could not use any router and may be issue with control it as mention here https://github.com/Koenkk/zigbee2mqtt/issues/592

Koenkk, 2019-02-14:
@ptvoinfo could this be a bug in the router firmware?

dzungpv, 2019-02-14:
@Koenkk I also build my own router firmware base on TI samplelight  and sampleswitch project. Any device could be a router too.

dzungpv, 2019-02-14:
This also occur with xiaomi wall switch and plug.

dzungpv, 2019-02-23:
@Koenkk I guest that zigbee2mqtt or zigbee-sepherd send the packet (on/off command) not work when end device reconnect to the router or when the router reconnect to the coordinator.
The problem occur when the coordinator off line for 5 minutes and online after that (router and end device still online ), the command only receive but not send to end device, it only could send to router.
In zigbee it call hop, it may not handle fine.

We may use broad cast send with end device. I will check zigbee-sepherd api.

@kirovilya Do you have any idea?

kirovilya, 2019-02-25:
@dzungpv Hmm, strange. I saw this with xiaomi outlet. It was treated by turning off and turning on the outlet.
Now in iobroker.zigbee we send a reset when the adapter starts, and we also send ping to all routers every minute.
Maybe this approach will help in this case.

dzungpv, 2019-02-25:
@kirovilya  I don't know what you mean send a reset when adapter start.

You reset the coordinator or the router? How you do that?

kirovilya, 2019-02-25:
@dzungpv I mean this signal https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zigbee.js#L81 at coordinator. Some routers at this time re-pairing with the coordinator

kirovilya, 2019-02-25:
@dzungpv no

dzungpv, 2019-02-25:
@Koenkk can you check this?
Is z2m now send soft reset when coordinator start?

Koenkk, 2019-02-25:
@dzungpv no, I will add this.

kirovilya, 2019-02-25:
check first :)

Koenkk, 2019-02-25:
@dzungpv implemented, please check https://github.com/Koenkk/zigbee2mqtt/tree/fix/240_mac_expired

dzungpv, 2019-02-28:
@Koenkk I will try, but this is an Zigbee 1.2 issue, we must migrate to 3.0 to solve the problem:
https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/766240/2834664#2834664

Koenkk, 2019-03-22:
did this change have any effect, or is 3.0 the only solution?

closed 2019-05-28 by stale[bot]