Koenkk/zigbee2mqtt #1091: new tradfri ikea outlet 1702

zambujal, 2019-02-14:
Hello, I have no coding experience neither linux experience ... 

ok I managed to edit the device.js ... I included this:

 _
 {
        zigbeeModel: ['TRADFRI control outlet'],
        model: 'E1703',
        description: 'TRADFRI control outlet',
        supports: 'on/off',
        vendor: 'IKEA',
        fromZigbee: [fz.generic_state],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 1000, 0, cb),
                (cb) => device.foundation('genPowerCfg', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
    },
    {

and it works fine... it's so easy to connect, might there be weird stuff, but it works.... now where is the device.js on the hash.io installation...plugin...

zambujal, 2019-02-14:
many errors though... but it works I even made mistakes E1603 doesn't exist it's 1702

 Feb 14 22:55:42 hassbian npm[526]:   zigbee2mqtt:info 2/14/2019, 10:55:42 PM Zigbee publish to device '0x000d6ffffe2edfdc', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Feb 14 22:55:42 hassbian npm[526]:   zigbee2mqtt:info 2/14/2019, 10:55:42 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2edfdc', payload '{"state":"ON"}'
Feb 14 22:55:42 hassbian npm[526]:   zigbee2mqtt:warn 2/14/2019, 10:55:42 PM No converter available for 'E1603' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":1}}'
Feb 14 22:55:42 hassbian npm[526]:   zigbee2mqtt:warn 2/14/2019, 10:55:42 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Feb 14 22:55:43 hassbian npm[526]:   zigbee2mqtt:info 2/14/2019, 10:55:43 PM Zigbee publish to device '0x000d6ffffe2edfdc', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Feb 14 22:55:43 hassbian npm[526]:   zigbee2mqtt:info 2/14/2019, 10:55:43 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2edfdc', payload '{"state":"OFF"}'
Feb 14 22:55:44 hassbian npm[526]:   zigbee2mqtt:warn 2/14/2019, 10:55:44 PM No converter available for 'E1603' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":0}}'
Feb 14 22:55:44 hassbian npm[526]:   zigbee2mqtt:warn 2/14/2019, 10:55:44 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.

zambujal, 2019-02-14:
weird now it doesn't work when I put 1702 in the file devices.js  ... though it works when it says1603 don't know why I made that mistake...  I really don't understand. Not capable to pair if there the 1702... I do not understand ...but it works...  it's really cheap I payed in france not more than 10 euros I think

zambujal, 2019-02-15:
never mind, I was able to connect the outlet even without editing  in a diferent home assistance. I am really a newbie to this...  but enjoying it!

zambujal, 2019-02-15:
though it shows like 1603 :D 

zigbee2mqtt:info 2019-2-15 02:25:08 MQTT publish: topic 'homeassistant/switch/0x000d6ffffe2edfdc/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x000d6ffffe2edfdc/set","state_topic":"zigbee2mqtt/0x000d6ffffe2edfdc","name":"0x000d6ffffe2edfdc_switch","unique_id":"0x000d6ffffe2edfdc_switch_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffe2edfdc","name":"0x000d6ffffe2edfdc","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI control outlet (E1603)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'

zambujal, 2019-02-15:
I did nothing and maybe I thought I made something... the pairing sometimes is difficult... sometimes not... weird...
many thanks Koenkk for this wonderfull service for the community. I am learning in the process... now have to link arduino that control my aquarium via zigbee... ;)

Koenkk, 2019-02-15:
I'm confused because of all the messages.

- Could you provide an URL to this device (from the IKEA site)
- Can you share your `data/database.db` file?

zambujal, 2019-02-15:
I will do that when in home. Here the link to the product. I will make an photo of it when in home.

zambujal, 2019-02-15:
8 euros... Amazing and finally 16 amp

zambujal, 2019-02-15:
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/2871279/database.zip)
 from the non hass.io instalation...

Koenkk, 2019-02-16:
What zigbee2mqtt version are you running, this should already be supported in zigbee2mqtt 1.1.1

antti365, 2019-02-16:
> many errors though... but it works I even made mistakes E1603 doesn't exist it's 1702, I edited the file device.js with the 1603 name...

How does this not exist? I have just E1603 ...

zambujal, 2019-02-16:
sorry about my english.... 

> many errors though... but it works I even made mistakes E1603 doesn't exist it's 1702, I edited the file device.js with the 1603 name...
> 
> How does this not exist? I have just E1603 ...

it's my lawsy english ... it doesn't exist on my setup I have the 1702, but maybe it's the same produit just an different reference/label ...

zambujal, 2019-02-16:
I am using the 1.1.1 version now and it's working well... not even error messages. My problems arised because I needed several hours to pair ... why... don't know. and I tweaked edited to some of those files, without even knowing for sure what I was doing. so the erros, this is working well in hasi.io zig2mqtt 1.1.1 without even editing it though he see the Ikea outlet like the 1603, like I said maybe it's the same electronics and only changed some of the specifications... this one does 16 amp 220v of course.

Koenkk, 2019-02-17:
It's seems that there are multiple model numbers of this switch, so far I've found E1603, E1702, E1705, E1707. 

E1706 seems to be the US variant, E1702 the french, E1603 the german.

As the Zigbee model is identical, we cannot distinguish between the variant. That also means that all these plugs are supported.

closed 2019-02-17 by Koenkk

Simanias, 2019-05-20:
I get this message when I try to pair my Tradfri 1603 control outlet:

> No converter available for 'E1603' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"10":[69,49,54,48,51]}}'<

Any thoughts?

Regards