Koenkk/zigbee2mqtt #1092: Low range using CC2530

berrywhite96, 2019-02-14:
Today my CC2530 arrived, before I was using for months the CC2531. I flashed it and connect it like described here: https://www.zigbee2mqtt.io/how_tos/how_to_connect_a_cc2530_coordinator_using_a_usb_to_serial_adapter.html (GND connected to the left Pin, is this important?!)
And later it worked with the software, so no problems until here.

I upgraded because in my living room the connectivity was not really good, so I thought the upgrade would help the devices there to have a stable connection. But it felts like it has the same range like the CC2531, so maybe I did something wrong?

Thanks for help!

Koenkk, 2019-02-15:
https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html may help you (the cause might be e.g. wifi interference).

JLFN, 2019-02-15:
the included antenna for cc2530 is not very good try another 2.4ghz antenna.

berrywhite96, 2019-02-15:
> https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html may help you (the cause might be e.g. wifi interference).

My Wi-Fi AP is two rooms away from the CC2530, but I will try to change channels, thanks.

> the included antenna for cc2530 is not very good try another 2.4ghz antenna.

Do you have a model you can recommend?

JLFN, 2019-02-15:
@berrywhite96 any  2.4Ghz wifi antenna you have laying around.

krikk, 2019-02-16:
i also upgraded from a CC2531 to the CC2530, and i did see a major increase in the measured linkquality:

[image]

distance between the sensor and the cordinator: one reinforced-concrete floor, but only about 5 meters

i use channel 26 so wifi interference isn't a problem

krikk, 2019-02-16:
and i did use an usb extension cable for the cc2531

berrywhite96, 2019-02-16:
> https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html may help you (the cause might be e.g. wifi interference).

I tried channel 6 and 26, but my devices wont connect to the bridge anymore.
Here's the bottom of my config:
`    friendly_name: 'ikea_remote'
    retain: false
    coordinator_group: 0xa9a6
advanced:
  rtscts: false
  channel: 26`

closed 2019-04-25 by stale[bot]