Koenkk/zigbee2mqtt #1093: Can't start npm

shafiq786110, 2019-02-15:
I'm quite new to linux, im learning slowly but i cant understand why i cant get this to start

I followed the instructions here: https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

When it gets to `npm start` i get the following message:

> pi@raspberrypi:~ $ clear
> pi@raspberrypi:~ $ npm start
> npm ERR! path /home/pi/package.json
> npm ERR! code ENOENT
> npm ERR! errno -2
> npm ERR! syscall open
> npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
> npm ERR! enoent This is related to npm not being able to find a file.
> npm ERR! enoent
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/pi/.npm/_logs/2019-02-15T00_13_38_321Z-debug.log
> pi@raspberrypi:~ $

Here is the debug log referring to this: https://pastebin.com/qEMsx0Zc

My zigbee2MQTT file has passed the yamllint check 

I noticed that there wasn't the package.json or index.js files in the location "/home/pi" so i tried to copy them from the zigbee2MQTT folder to my home folder. When i do that, i get this message: 

>     at Function.Module._load (internal/modules/cjs/loader.js:507:25)
>     at Module.require (internal/modules/cjs/loader.js:637:17)
>     at require (internal/modules/cjs/helpers.js:22:18)
>     at Object.<anonymous> (/home/pi/index.js:1:78)
>     at Module._compile (internal/modules/cjs/loader.js:689:30)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
>     at Module.load (internal/modules/cjs/loader.js:599:32)
>     at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
>     at Function.Module._load (internal/modules/cjs/loader.js:530:3)
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! zigbee2mqtt@1.1.1 start: `node index.js`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the zigbee2mqtt@1.1.1 start script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> npm WARN Local package.json exists, but node_modules missing, did you mean to install?
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/pi/.npm/_logs/2019-02-15T00_18_31_870Z-debug.log
> pi@raspberrypi:~ $ clear
> pi@raspberrypi:~ $ npm start
> 
> > zigbee2mqtt@1.1.1 start /home/pi
> > node index.js
> 
> internal/modules/cjs/loader.js:583
>     throw err;
>     ^
> 
> Error: Cannot find module 'semver'
>     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
>     at Function.Module._load (internal/modules/cjs/loader.js:507:25)
>     at Module.require (internal/modules/cjs/loader.js:637:17)
>     at require (internal/modules/cjs/helpers.js:22:18)
>     at Object.<anonymous> (/home/pi/index.js:1:78)
>     at Module._compile (internal/modules/cjs/loader.js:689:30)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
>     at Module.load (internal/modules/cjs/loader.js:599:32)
>     at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
>     at Function.Module._load (internal/modules/cjs/loader.js:530:3)
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! zigbee2mqtt@1.1.1 start: `node index.js`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the zigbee2mqtt@1.1.1 start script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> npm WARN Local package.json exists, but node_modules missing, did you mean to install?
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/pi/.npm/_logs/2019-02-15T00_18_36_647Z-debug.log
> pi@raspberrypi:~ $

And here is the debug log for this: https://pastebin.com/C9QTDax2

Can anyone please help with this?? I feel like ive gotten so far with getting this all set up and im stuck at the last bit!! 

I have searched for this error on google but i cant find a solution that works for me

Thank you!

JLFN, 2019-02-15:
You are not follow the install guide! Do that and it should work.

shafiq786110, 2019-02-15:
> You are not following the install guide , step 2 step! Follow it and it will work!

I have, 4 times now, each with a fresh install of raspbian. And im copying the command lines, not typing and yet the same error each time

JLFN, 2019-02-15:
You can download hassbian script and install it
 ```.bash wget https://github.com/home-assistant/hassbian-scripts/releases/download/0.11.1/hassbian-scripts_0.11.1_all.deb```
Install it 
``` sudo dpkg -i hassbian-scripts_0.11.1_all.deb ```
edit zigbee2mqtt.sh
```sudo nano hassbian-scripts/package/opt/hassbian/suites/zigbee2mqtt.sh```
Before installation sudo nano hassbian-scripts/package/opt/hassbian/suites/zigbee2mqtt.sh
go to line 19 and change to from 8 to 10 
```bash curl -sL https://deb.nodesource.com/setup_10.x | bash - ```
Save it and then run ```sudo hassbian-config install zigbee2mqtt ``
That will go very fast using that alot.

clockbrain, 2019-02-15:
@shafiq786110 see this error

```
npm ERR! path /home/pi/package.json
```

You don't seem to be in the /opt/zigbee2mqtt directory that steps 2 and 4 of the instructions specify

```
cd /opt/zigbee2mqtt
npm start
```

shafiq786110, 2019-02-15:
@JLFN 

> @shafiq786110
> You can download hassbian script and install it
> https://github.com/home-assistant/hassbian-scripts/releases
> Download it
> ` wget https://github.com/home-assistant/hassbian-scripts/releases/download/0.11.1/hassbian-scripts_0.11.1_all.deb`
> Install it
> `sudo dpkg -i hassbian-scripts_0.11.1_all.deb`
> Edit zigbee2mqtt.sh
> `sudo nano hassbian-scripts/package/opt/hassbian/suites/zigbee2mqtt.sh`
> 
> Go to line [19](https://github.com/home-assistant/hassbian-scripts/blob/53d65eaacc325905937080c46705650f34e386b2/package/opt/hassbian/suites/zigbee2mqtt.sh#L19) and change to from 8 to 10
> `curl -sL https://deb.nodesource.com/setup_10.x | bash - `
> Save it and then run
> `sudo hassbian-config install zigbee2mqtt `
> Very nice and simple :)
> 
> Big thanks to @Landrash

It worked!! 
Thank you so much for your help! Now i just need to work out how to get my devices working with it lol

closed 2019-02-15 by shafiq786110