Koenkk/zigbee2mqtt #1094: All stops working after (seemingly) heavy control

wowtah, 2019-02-15:
I have recently setup zigbee2mqtt in Hassio and after some testing I find it a very promising project.

I do have one repeating issue:

seemingly after some heavy control, like turning on/off multiple lights at the same time (using light groups in Homeassistant), all control stops working. If I check the zigbee2mqtt log i get a lot of

`zigbee2mqtt:error 2019-2-15 01:42:47 Zigbee publish to device '0x000b57fffe97bbb4', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout` 

When I do a network scan at this point (i use zigbee2mqtt_networkmap) the log only has

```
zigbee2mqtt:info 2019-2-15 01:42:26 MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];
  "0x00124b001936942c" [style="bold", label="{0x00124b001936942c|Coordinator|No model information available|online}"];
  "0x000b57fffea69572" [style="rounded", label="{Tradfri hal 2|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x000b57fffea7152d" [style="rounded", label="{Tradfri hal 3|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x000b57fffe97bbb4" [style="rounded", label="{Tradfri hal 1|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x000b57fffea47637" [style="rounded", label="{Tradfri keuken 2|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x000b57fffea50004" [style="rounded", label="{Tradfri keuken 3|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x000b57fffe983f8b" [style="rounded", label="{Tradfri keuken 1|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x00158d0001e057a7" [style="rounded, dashed", label="{Aqara motion|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x000b57fffe9b37e2" [style="rounded", label="{Tradfri slaapkamer 4|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x000b57fffe974a12" [style="rounded", label="{Tradfri slaapkamer 3|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x000b57fffea4b062" [style="rounded", label="{Tradfri slaapkamer 1|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x000b57fffeb0e974" [style="rounded", label="{Tradfri slaapkamer 2|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x00158d0001e887fc" [style="rounded, dashed", label="{Aqara cube|EndDevice|Xiaomi Mi/Aqara smart home cube (MFKZQ01LM)|online}"];
}'
```

where usually this gives out a lot more information. When I check the graphical map, I still see al my devices, but all the connecting lines between the blocks are gone. 

[image]

after replugging the cc2531 USB dongle and restarting zigbee2mqtt through Hass.io, everything works again. 

Normally (when all is working) the network map looks like this:
[image]

My zigbee2mqtt configuration.yaml is pretty standard i think:
```
devices:
  '0x000b57fffe974a12':
    friendly_name: Tradfri slaapkamer 3
    retain: false
  '0x000b57fffe97bbb4':
    friendly_name: Tradfri hal 1
    retain: false
  '0x000b57fffe983f8b':
    friendly_name: Tradfri keuken 1
    retain: false
  '0x000b57fffe9b37e2':
    friendly_name: Tradfri slaapkamer 4
    retain: false
  '0x000b57fffea47637':
    friendly_name: Tradfri keuken 2
    retain: false
  '0x000b57fffea4b062':
    friendly_name: Tradfri slaapkamer 1
    retain: false
  '0x000b57fffea50004':
    friendly_name: Tradfri keuken 3
    retain: false
  '0x000b57fffea69572':
    friendly_name: Tradfri hal 2
    retain: false
  '0x000b57fffea7152d':
    friendly_name: Tradfri hal 3
    retain: false
  '0x000b57fffeb0e974':
    friendly_name: Tradfri slaapkamer 2
[... 26 lines cut ...]
{"id":7,"type":"Router","ieeeAddr":"0x000b57fffe983f8b","nwkAddr":46546,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 W 400lm","epList":[1],"status":"online","joinTime":1550184739,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb GU10 W 400lm","dateCode":"20170302","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":48,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":148,"lastMessageRssi":-63}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"fsoDQYPFUay8XyIq"}
{"id":8,"type":"EndDevice","ieeeAddr":"0x00158d0001e057a7","nwkAddr":47762,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"status":"online","joinTime":1550185062,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"jbdIZczguuBE1ZSv"}
{"id":5,"type":"Router","ieeeAddr":"0x000b57fffea47637","nwkAddr":48586,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 W 400lm","epList":[1],"status":"online","joinTime":1550184406,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb GU10 W 400lm","dateCode":"20170302","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"qOtwqKAHGZCQJyoM"}
{"id":12,"type":"Router","ieeeAddr":"0x000b57fffeb0e974","nwkAddr":48545,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 W 400lm","epList":[1],"status":"online","joinTime":1550185215,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb GU10 W 400lm","dateCode":"20170302","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":1,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"haDiagnos [... cut ...] facturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"DSW2QMrw5hJZCBjX"}

```

Any idea what causes this? Is this a known issue?

Koenkk, 2019-02-15:
@glentakahashi and @sylarevan hit the same problem. With also some possible solution: https://github.com/Koenkk/zigbee2mqtt/issues/764#issuecomment-462162875

Let's use this issue to tackle this problem.

@wowtah 
- can you post the log when the error happens
- how many commands per second do you send?

I checked with my own network. It seems to get unreliable when more than 6 group commands are send. This seems to be the same as what [@glentakahashi] experiences https://github.com/Koenkk/zigbee2mqtt/issues/764#issuecomment-462162875

Koenkk, 2019-02-15:
I've made some changes and got good results. 

Here is a log where I send 40 `set` MQTT messages in a second, which get handled by zigbee2mqtt without errors: https://hastebin.com/iwuhuverad.cs

To test this:
- Update to the latest zigbee2mqtt (dev/edge for hassio), which has improved sheduling
- Flash the `20190215` firmware (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin). A cool thing about this firmware is that it also doesn't not require to re-pair devices.

glentakahashi, 2019-02-15:
Very cool! I will test this this weekend as well.

@Koenkk did you also find that your bulbs were being responsive to the group commands as well? Even when I got to the point where i could send 2 requests a second without the CC2531 hitting limits, my bulbs didn't respond to requests anymore.

Koenkk, 2019-02-15:
@glentakahashi they indeed stop responding after a few commands, however I think, apart from adding extra delays for group controls, there is not much to do about this.

glentakahashi, 2019-02-15:
Yea that makes sense. It seems some bulbs don’t have the power to support
lots of group requests which is unfortunate.

EmilFlach, 2019-02-17:
> I've made some changes and got good results.

I have great results too!
I am now running on dev/edge with the `20190215` firmware.

This is my log where I sent 40 messages in one second with just hue bulbs in a group: [https://hastebin.com/anenoleleb.cs](https://hastebin.com/anenoleleb.cs)

And this is 40 messages in one second with hue bulbs and a tradfri bulb: [https://hastebin.com/jobamunewu.cs](https://hastebin.com/jobamunewu.cs)

@glentakahashi See this comparison for how a hue bulb and a tradfri bulb handle these messages:
Hue: https://gfycat.com/legaldescriptiveleveret
Tradfri: https://gfycat.com/impartialrecklessfrog

I'm very happy with these results as my dimmer is behaving perfectly with the hue bulbs:
https://gfycat.com/grandwellinformedcockerspaniel

It indeed looks like the Tradfri bulbs can't handle the same speed of messages, because during dimming mine freezes frequently. This behavior is not limited to groups; it has trouble with fast (4 per second) direct messages as well, which the hues are fine with. Before the Hue's would be fine when addressed directly, but not in groups. This issue with hues in groups has been resolved for me!

@Koenkk It seems like my bottlenecks on the zigbee2mqtt side are fixed, thanks a lot!

Koenkk, 2019-02-17:
I've verified that the TRADFRI bulbs can indeed not handle many commands per second.

I've made a setup were I paired a TRADFRI bulb directly to the remote (so there is no hub/zigbee2mqtt between) and started spamming the bulb with commands. Around 00:10 you see that the TRADFRI bulb does not respond anymore, after a few seconds, it starts responding again. Video: https://www.youtube.com/watch?v=LK9Llve8Xl8

As @EmilFlach mentions, I think the problems on the zigbee2mqtt side are indeed solved.

@glentakahashi please let me know your results.

glentakahashi, 2019-02-17:
@Koenkk @emilflach Just tested and got the same results as you two. Zigbee2mqtt supports about 5 commands a second, but it's the bulbs that can't keep up. The Sengled bulbs actually support even fewer commands than the TRADFRI bulbs in your video. I can only hit about 10 commands before it stops responding. Hopefully bulbs in the future better support group commands!

Koenkk, 2019-02-17:
Good, then I think this issue can be closed?

glentakahashi, 2019-02-17:
I think so, not much else to do from zigbee2mqtt side

closed 2019-02-18 by Koenkk