Koenkk/zigbee2mqtt #1095: Device compatibility with ZigBee channel 26

kiwinol, 2019-02-15:
I have been reading:
[https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html](url)

That says:
```
ZigBee channel 26 is usually relatively unaffected by WiFi, but many ZigBee devices do not support it.
```

Is there any way for me to verify if all my devices support channel 26?

JLFN, 2019-02-15:
@kiwinol Change your zigbee2mqtt config to channel 26 and then try to pair it!

khzd, 2019-02-15:
Hi
We are implementing support to livolo zigbee working Only on channel 26, so I test many Xiaomi devices and gledopto on this channel and work fine 
Didn’t have all other devices to test...

Koenkk, 2019-02-15:
As this is not an zigbee2mqtt issue I will close this issue.

Feel free to continue the discussion.

closed 2019-02-15 by Koenkk