Koenkk/zigbee2mqtt #1096: Crashing CC2531 / Zigbee2MQTT after restart

Vout, 2019-02-15:
Currently i'm experiencing an lot of issues with the CC2531, when working it seems to work quite ok but after an reboot of Hassio, the Zigbee2Mqtt addon or the pc i'm running it on i'm always having the following issue,

```[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-02-15T14:43:53: PM2 log: Launching in no daemon mode
2019-02-15T14:43:53: PM2 log: App [npm:0] starting in -fork mode-
2019-02-15T14:43:53: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /app
> node index.js
		Zigbee2mqtt requires node version 10, you are running v8.14.0!
  zigbee2mqtt:info 2019-2-15 14:43:53 Logging to directory: '/share/zigbee2mqtt/log/2019-02-15.14-43-53'
  zigbee2mqtt:debug 2019-2-15 14:43:53 Removing old log directory '/share/zigbee2mqtt/log/2019-02-09.15-40-40'
  zigbee2mqtt:debug 2019-2-15 14:43:53 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2019-2-15 14:43:53 Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 2019-2-15 14:43:53 Starting zigbee2mqtt version 1.1.1 (commit #3d15129)
  zigbee2mqtt:info 2019-2-15 14:43:53 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-15 14:43:53 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2019-2-15 14:44:54 Starting zigbee-shepherd
  zigbee2mqtt:error 2019-2-15 14:44:54 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2019-2-15 14:44:54 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2019-2-15 14:44:54 Failed to start
	{"message":"Error: Resource busy, cannot open /dev/ttyACM0","stack":"Error: Error: Resource busy, cannot open /dev/ttyACM0"}```

My setup is as follows,

- Lenovo M73 (i3 with 8GB ram and 256GB SSD)
- CC2531 as coordinator with the latest firmware
- HassIO version 0.87.x
- Zigbee2Mqtt addon, tried both the stable and edge version

To get it working again i have to turn of the pc, unplug the stick, replug it again, fire up the pc and fingers crossing that it works. Sometimes it works, most times it doesn't. I also tried replugging the stick without powering it off or pressing the reset button as mentioned in the docs, doesn't work.

I already tried reflashing the coordinator but that didn't seem to help... 

I also have an test setup with an simple atom pc and another CC2531 stick i had laying around but there also the issue with Zigbee shepherd nagging about `resource busy` quite some times.

Any suggestions what i can do to fix this?

Koenkk, 2019-02-15:
This doesn't seem to be a zigbee2mqtt, but rather an issue with the serialport already being in use.

You should find out which process locks this port.

Some resources I've found:
- https://unix.stackexchange.com/questions/11238/how-to-get-over-device-or-resource-busy/11241
- https://stackoverflow.com/questions/53163998/could-not-open-port-error-immediately-after-plugging-in-arduino-on-arch-linux

Vout, 2019-02-19:
> * https://stackoverflow.com/questions/53163998/could-not-open-port-error-immediately-after-plugging-in-arduino-on-arch-linux

That one seemed to resolve my issue, i disabled the service last friday and haven't had the issue since!

Can be closed if you ask me ;)

closed 2019-02-19 by Vout