Koenkk/zigbee2mqtt #1099: Multiple Ikea Tradfri Outlet failing to join

luke-browning, 2019-02-15:
Hi, I'm having a hard time getting a second Tradfri outlet to join.. The instructions say;

> To factory reset the TRADFRI control outlet, press and hold the reset button (pinhole underneath the light, located at the top of the outlet) with a paperclip until the white light starts fading. Hold onto the button for a few more seconds, then release. After this, the outlet will automatically connect.

This is what I followed to set up the first device. This worked straight away. I then wanted to add a second device but I get nothing in the log;

```
[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-02-15T15:18:50: PM2 log: Launching in no daemon mode
2019-02-15T15:18:50: PM2 log: App [npm:0] starting in -fork mode-
2019-02-15T15:18:50: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /zigbee2mqtt-1.1.1
> node index.js
  zigbee2mqtt:info 2019-2-15 15:18:52 Logging to directory: '/share/zigbee2mqtt/log/2019-02-15.15-18-52'
  zigbee2mqtt:debug 2019-2-15 15:18:52 Removing old log directory '/share/zigbee2mqtt/log/2019-02-13.13-33-51'
  zigbee2mqtt:debug 2019-2-15 15:18:52 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2019-2-15 15:18:52 Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 2019-2-15 15:18:53 Starting zigbee2mqtt version 1.1.1 (commit #unknown)
  zigbee2mqtt:info 2019-2-15 15:18:53 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-15 15:18:53 zigbee-shepherd started
  zigbee2mqtt:info 2019-2-15 15:18:53 Coordinator firmware version: '20190109'
  zigbee2mqtt:debug 2019-2-15 15:18:53 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ed3096","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190109},"startTime":1550243933,"joinTimeLeft":0}
  zigbee2mqtt:info 2019-2-15 15:18:53 Currently 1 devices are joined:
  zigbee2mqtt:info 2019-2-15 15:18:53 3782319979803407 (0x000d6ffffed2b30f): E1603 - IKEA TRADFRI control outlet (Router)
  zigbee2mqtt:info 2019-2-15 15:18:53 Zigbee: disabling joining new devices.
  zigbee2mqtt:info 2019-2-15 15:18:53 Connecting to MQTT server at mqtt://automation.browning.local:1883
  zigbee2mqtt:debug 2019-2-15 15:18:53 Configuring 3782319979803407 (0x000d6ffffed2b30f) ...
  zigbee2mqtt:info 2019-2-15 15:18:53 zigbee-shepherd ready
  zigbee2mqtt:info 2019-2-15 15:18:53 Connected to MQTT server
  zigbee2mqtt:info 2019-2-15 15:18:53 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-2-15 15:18:53 MQTT publish: topic 'zigbee2mqtt/3782319979803407', payload '{"state":"ON","linkquality":118}'
  zigbee2mqtt:info 2019-2-15 15:18:54 MQTT publish: topic 'homeassistant/switch/0x000d6ffffed2b30f/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/3782319979803407/set","state_topic":"zigbee2mqtt/3782319979803407","name":"3782319979803407_switch","unique_id":"0x000d6ffffed2b30f_switch_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffed2b30f","name":3782319979803407,"sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI control outlet (E1603)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:debug 2019-2-15 15:18:54 Mounted the cieApp (epId 11)
  zigbee2mqtt:info 2019-2-15 15:18:54 Succesfully configured 3782319979803407 (0x000d6ffffed2b30f)
  zigbee2mqtt:debug 2019-2-15 15:20:19 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0x000d6ffffed2b30f)
  zigbee2mqtt:info 2019-2-15 15:20:19 MQTT publish: topic 'zigbee2mqtt/3782319979803407', payload '{"state":"ON","linkquality":118}'
  zigbee2mqtt:debug 2019-2-15 15:20:19 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0x000d6ffffed2b30f)
  zigbee2mqtt:debug 2019-2-15 15:21:06 Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
  zigbee2mqtt:info 2019-2-15 15:21:06 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-2-15 15:21:06 MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":true}'
```

Am I doing something wrong? Or is there an issue adding multiple outlets?

Thanks

antti365, 2019-02-15:
You do everything correctly, except that after each pairing of one device, you need to reset the zigbee2mqtt stick. At least this is what happens to me with Ikea devices ...

Reset zigbee2mqtt:

* stop zigbee2mqtt
* unplug CC2531
* plug CC2531
* press reset button on CC2531 (the one close to the usb port, green light should go off)
* start zigbee2mqtt

antti365, 2019-02-15:
Maybe @Koenkk will help to deal with such a wrong behavior, but for now we have what we have ...

Koenkk, 2019-02-15:
@antti365 I haven't seen this yet, could this be only with the TRADFRI control outlet?

@luke-browning can you confirm that this indeed fixes it?

antti365, 2019-02-15:
@Koenkk No, it’s not just a outlet, I’m watching it with almost all TRADFRI products — a dimmer, an on / off switch, an outlet, etc.

Koenkk, 2019-02-15:
@antti365 also with the bulbs?

antti365, 2019-02-15:
@Koenkk Light bulbs today did not have time to check, tomorrow I will check

luke-browning, 2019-02-15:
@Koenkk I can confirm the steps provided by @antti365 work. After adding a single device I can no longer add another one.

I'm a little unsure if this log look correct either? It looks like it tries to connect more than 10 times before accepting the device..

```
  zigbee2mqtt:info 2019-2-15 22:09:55 Starting zigbee2mqtt version 1.1.1 (commit #unknown)
  zigbee2mqtt:info 2019-2-15 22:09:55 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-15 22:09:57 zigbee-shepherd started
  zigbee2mqtt:info 2019-2-15 22:09:57 Coordinator firmware version: '20190109'
  zigbee2mqtt:debug 2019-2-15 22:09:57 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ed3096","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190109},"startTime":1550268597,"joinTimeLeft":0}
  zigbee2mqtt:info 2019-2-15 22:09:57 Currently 1 devices are joined:
  zigbee2mqtt:info 2019-2-15 22:09:57 3782319979803407 (0x000d6ffffed2b30f): E1603 - IKEA TRADFRI control outlet (Router)
  zigbee2mqtt:info 2019-2-15 22:09:57 Zigbee: disabling joining new devices.
  zigbee2mqtt:info 2019-2-15 22:09:57 Connecting to MQTT server at mqtt://automation.browning.local:1883
  zigbee2mqtt:debug 2019-2-15 22:09:57 Configuring 3782319979803407 (0x000d6ffffed2b30f) ...
  zigbee2mqtt:info 2019-2-15 22:09:57 zigbee-shepherd ready
  zigbee2mqtt:info 2019-2-15 22:09:57 Connected to MQTT server
  zigbee2mqtt:info 2019-2-15 22:09:57 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-2-15 22:09:57 MQTT publish: topic 'zigbee2mqtt/3782319979803407', payload '{"state":"ON","linkquality":120}'
  zigbee2mqtt:info 2019-2-15 22:09:57 MQTT publish: topic 'homeassistant/switch/0x000d6ffffed2b30f/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/3782319979803407/set","state_topic":"zigbee2mqtt/3782319979803407","name":"3782319979803407_switch","unique_id":"0x000d6ffffed2b30f_switch_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffed2b30f","name":3782319979803407,"sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI control outlet (E1603)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:debug 2019-2-15 22:09:57 Mounted the cieApp (epId 11)
  zigbee2mqtt:info 2019-2-15 22:09:58 Succesfully configured 3782319979803407 (0x000d6ffffed2b30f)
  zigbee2mqtt:debug 2019-2-15 22:10:42 Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
  zigbee2mqtt:info 2019-2-15 22:10:42 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-2-15 22:10:42 MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 2019-2-15 22:11:17 Received zigbee message of type 'devInterview' with data '"0x000d6ffffed16f00"'
  zigbee2mqtt:info 2019-2-15 22:11:17 Connecting with device...
  zigbee2mqtt:info 2019-2-15 22:11:17 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-2-15 22:11:17 Received zigbee message of type 'devInterview' with data '"0x000d6ffffed16f00"'
  zigbee2mqtt:info 2019-2-15 22:11:17 Connecting with device...
  zigbee2mqtt:info 2019-2-15 22:11:17 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-2-15 22:11:17 Received zigbee message of type 'devInterview' with data '"0x000d6ffffed16f00"'
  zigbee2mqtt:info 2019-2-15 22:11:17 Connecting with device...
  zigbee2mqtt:info 2019-2-15 22:11:17 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-2-15 22:11:18 Received zigbee message of type 'devInterview' with data '"0x000d6ffffed16f00"'
  zigbee2mqtt:info 2019-2-15 22:11:18 Connecting with device...
  zigbee2mqtt:info 2019-2-15 22:11:18 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-2-15 22:11:18 Received zigbee message of type 'devInterview' with data '"0x000d6ffffed16f00"'
  zigbee2mqtt:info 2019-2-15 22:11:18 Connecting with device...
  zigbee2mqtt:info 2019-2-15 22:11:18 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-2-15 22:11:18 Received zigbee message of type 'devInterview' with data '"0x000d6ffffed16f00"'
  zigbee2mqtt:info 2019-2-15 22:11:18 Connecting with device...
  zigbee2mqtt:info 2019-2-15 22:11:18 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-2-15 22:11:18 Received zigbee message of type 'devInterview' with data '"0x000d6ffffed16f00"'
  zigbee2mqtt:info 2019-2-15 22:11:18 Connecting with device...
  zigbee2mqtt:info 2019-2-15 22:11:18 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-2-15 22:11:19 Received zigbee message of type 'devInterview' with data '"0x000d6ffffed16f00"'
  zigbee2mqtt:info 2019-2-15 22:11:19 Connecting with device...
  zigbee2mqtt:info 2019-2-15 22:11:19 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
[... cut ...]
  zigbee2mqtt:info 2019-2-15 22:11:49 MQTT publish: topic 'homeassistant/switch/0x000d6ffffed16f00/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x000d6ffffed16f00/set","state_topic":"zigbee2mqtt/0x000d6ffffed16f00","name":"0x000d6ffffed16f00_switch","unique_id":"0x000d6ffffed16f00_switch_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6ffffed16f00","name":"0x000d6ffffed16f00","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI control outlet (E1603)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

Koenkk, 2019-02-15:
@luke-browning log looks normal, so after the instructions from https://github.com/Koenkk/zigbee2mqtt/issues/1099#issuecomment-464099775 you can pair devices again?

luke-browning, 2019-02-15:
@Koenkk Yes. I have to do all of those steps to be able to connect to new devices.

Koenkk, 2019-02-15:
Great that this workaround at least fixes this.

I don't have any ideas yet how to solve this yet, restarting zigbee2mqtt automatically after a TRADFRI device is paired doesn't sound like a good idea.

antti365, 2019-02-16:
Hi @Koenkk 

1 test

delete devices from database.db via mqtt
stop zigbee2mqtt
delete database.db
delete state.json
reset stick
start zigbee2mqtt

add dimmer successfully
add on / off switch successfully
add outlet - failed (no log)

restart zigbee2mqtt - failed

in the log file:
• 2/16/2019, 9:13:05 AM - info: Logging to directory: 'data / log'
• 2/16/2019, 9:13:06 AM - info: Starting zigbee2mqtt version 1.1.1 (commit # f171d54)
• 2/16/2019, 9:13:06 AM - info: Starting zigbee-shepherd
• 2/16/2019, 9:13:08 AM - info: Error while starting zigbee-shepherd, attemping to fix ... (takes 60 seconds)

2 test

delete devices from database.db via mqtt
stop zigbee2mqtt
delete database.db
delete state.json
reset stick
start zigbee2mqtt
add on / off switch successfully
add outlet successfully
add dimmer - failed (no log)
stop zigbee2mqtt
start zigbee2mqtt successfully
add dimmer - failed (no log)
stop zigbee2mqtt
reset stick
start zigbee2mqtt successfully
add dimmer successfully
add LED bulb LED1649C5 successfully

3 test

delete devices from database.db via mqtt
stop zigbee2mqtt
delete database.db
delete state.json
reset stick
reset all devices to factory settings
start zigbee2mqtt
add LED bulb LED1649C5 successfully
add outlet successfully
add dimmer - failed (no log)
add on/off switch - failed (no log)
stop zigbee2mqtt
start zigbee2mqtt successfully
add dimmer - failed (no log)
add on/off switch - failed (no log)
stop zigbee2mqtt
reset stick
start zigbee2mqtt successfully
add on/off switch successfully
add dimmer - failed (no log)
stop zigbee2mqtt
start zigbee2mqtt successfully
add dimmer - failed (no log)
stop zigbee2mqtt
reset stick
[... 53 lines cut ...]
2/16/2019, 9:03:30 AM - debug: Configuring 0x000d6ffffeb1d3c5 (0x000d6ffffeb1d3c5) ...
2/16/2019, 9:03:30 AM - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI on/off switch' (0x000d6ffffeb1d3c5)

otherwise:
2/16/2019, 9:23:29 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":true}'
2/16/2019, 9:24:16 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:17 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:21 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:23 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:25 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:27 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:32 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:34 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:38 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:40 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:44 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:46 AM - debug: Received zigbee message of type 'devInterview' with data '"0x000d6ffffeb1d3c5"'
2/16/2019, 9:24:46 AM - debug: Received zigbee message of type 'devIncoming' with data '"0x000d6ffffeb1d3c5"' of device 'TRADFRI on/off switch' (0x000d6ffffeb1d3c5)
2/16/2019, 9:24:46 AM - info: Device incoming...
2/16/2019, 9:24:46 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
2/16/2019, 9:24:46 AM - debug: Configuring 0x000d6ffffeb1d3c5 (0x000d6ffffeb1d3c5) ...
2/16/2019, 9:24:46 AM - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI on/off switch' (0x000d6ffffeb1d3c5)
2/16/2019, 9:24:52 AM - info: Succesfully configured 0x000d6ffffeb1d3c5 (0x000d6ffffeb1d3c5)
2/16/2019, 9:24:52 AM - debug: Configuring 0x000d6ffffeb1d3c5 (0x000d6ffffeb1d3c5) ...
2/16/2019, 9:24:58 AM - info: Succesfully configured 0x000d6ffffeb1d3c5 (0x000d6ffffeb1d3c5)

antti365, 2019-02-16:
Tests were conducted on the following configuration:
info: Starting zigbee2mqtt version 1.1.1 (commit # f171d54)
info: Starting zigbee-shepherd
info: zigbee-shepherd started
info: Coordinator firmware version: '2018090201'

Is there any possibility to update the coordinator's firmware without a programmer?

Koenkk, 2019-02-16:
@antti365 thanks for the tests, when a device is already in `configuration.yaml` `2/16/2019, 9:03:17 AM - info: Connecting with device...` will not be logged.

Updating without a programmer is possible using the `.sbl` firmware. (http://processors.wiki.ti.com/index.php/SerialBootLoader)

antti365, 2019-02-16:
@Koenkk Any ideas after the tests? If anything, write, I will test ...

Koenkk, 2019-02-16:
Can you try if a reset via MQTT fixes the problem (https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/mqtt_topics_and_message_structure.md#zigbee2mqttbridgeconfigreset)?

antti365, 2019-02-16:
zigbee2mqtt/bridge/config/reset
Is something sent to this topic?

antti365, 2019-02-16:
@Koenkk 
git clone --branch dev
reset stick
start zigbee2mqtt
add outlet successfully
add dimmer successfully
add on / off switch - failed (no log)

2/16/2019, 9:46:38 PM - debug: Received MQTT message on 'zigbee2mqtt/bridge/config/reset' with data '' 
2/16/2019, 9:46:43 PM - info: Soft resetted ZNP 
2/16/2019, 9:46:43 PM - info: zigbee-shepherd ready 
2/16/2019, 9:46:43 PM - debug: Coordinator is already in group '99' 
2/16/2019, 9:46:43 PM - debug: Mounted the cieApp (epId 11)

add on / off switch - failed (no log)
add LED bulb LED1649C5 successfully
add on / off switch - failed (no log)
stop zigbee2mqtt
reset stick (press button)
start zigbee2mqtt successfully
add on / off switch successfully

:(

bizziebis, 2019-02-16:
I added my second Ikea tradfri outlet today and since then the coordinator has failed 5 times already. I had to unplug it, reset it and restart zigbee2mqtt.
I'm also loosing contact with several lights at this moment. I noticed the router still sees them, but they are not controllable anymore.

MrBramme, 2019-02-17:
Similar behaviour on my end as @bizziebis 
The issues started once I paired an IKEA outlet. did not have any issues after pairing IKEA bulbs.

I was hoping to use the outlets as routers for Xiaomi devices, but after reading several reports and noting behaviour on my network, the Xiaomi devices seem to be picky about routers and won't use the Ikea outlets :(

patriot1889, 2019-02-18:
I have an outlet in my system and keep losing control of a wemo bulb. I’m actually kind of glad that it isn’t just happening to me, means I haven’t royally messed something up.

Koenkk, 2019-02-19:
@antti365 so is this problem with the on/off switch or the outlet itself?

I'm having the feeling that the outlet somehow 'disrupts' the zigbee network.

bizziebis, 2019-02-19:
I've updated the coordinator to the latest DEV firmware two days ago, and since then no problems anymore. Don't know if it's a coincidence.. I noticed that during some odd behavior of the network my pan_id changed. That led me to update the frimware as it will keep the pan_id the same.

antti365, 2019-02-19:
@Koenkk Judging by the tests, the problem was observed both after adding an outlet and after adding a switch. I would not call it network break, if you look at the last test where I tested "Soft reset via MQTT", you can see that the switch cannot be added, and at the same time, the light bulb is quietly added (does the network work?). After adding the light bulb, I try to add the switch again and it is not added again.

antti365, 2019-02-19:
@bizziebis What is the DEV version of firmware? It is updated a hundred times a day ... :)
I tested on version commit #b08bf33.

antti365, 2019-03-07:
Has anything changed or not yet?

Sviests, 2019-03-17:
also experienced this issue when pairing the second outlet

timdonovanuk, 2019-05-06:
Hi - not sure why this is closed. I am facing the same thing adding a second Ikea outlet. I followed the instructions by antti365 (shutting down, unplugging stick etc) and it worked though.

Koenkk, 2019-05-19:
I've did some more investigation.

After pairing a E1603, it sends a permit join request with a time of 180 seconds. This means that after 3 minutes the network will be closed.

This should now be fixed in the latest dev branch, because zigbee2mqtt sends a permit join request every 160 seconds.

If anyone is still having issue with the latest dev branch, feel free to re-open.

closed 2019-05-19 by Koenkk