Koenkk/zigbee2mqtt #1100: Ikea Tradfri Repeater?

joshdinsdale, 2019-02-15:
Any idea if the new Ikea Tradfri repeater wil work with Zigbee2mqtt? I will buy one next time I am there to test myself, i cant see why it wouldnt?

https://www.ikea.com/gb/en/products/lighting/smart-lighting/tr%C3%A5dfri-signal-repeater-art-80424255/

SBTephan, 2019-02-15:
Yes, it will, it is added in the latest version.
I am also using it.

holgerebert, 2019-02-15:
@SBTephan How do you connect it to Zigbee2mqtt?

SBTephan, 2019-02-15:
There is a reset hole on the repeater. Set zigbee2mqtt to allowing devices and reset it.

Koenkk, 2019-02-15:
Device is already supported in the dev branch, pairing instructions: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/getting_started/pairing_devices.md#ikea-tradfri-signal-repeater-e1746

closed 2019-02-15 by Koenkk

holgerebert, 2019-02-15:
Which version of Zigbee2mqtt do you mean? Seems not to work with version 1.1.1

holgerebert, 2019-02-15:
Ok, got it, dev-branch. Thank you!

axiom00, 2019-02-19:
Hi @SBTephan - how is the Ikea working out for you? I am thinking of getting one or two.

SBTephan, 2019-02-19:
@axiom00 

Works great. Didn't had reception in the garage, now I have

axiom00, 2019-02-19:
@SBTephan Thanks for the feedback. Which coordinator are you using it with?

SBTephan, 2019-02-19:
@axiom00 could you define coordinator? Don't know exactly what you mean.

axiom00, 2019-02-19:
@SBTephan - the zigbee sniffer that zigbee2mqtt is talking to and the Ikea repeater is paired with. eg cc2530, cc2531.

SBTephan, 2019-02-19:
@axiom00 I am using the CC2530. I also ordered a CC2530+CC2591 to test if it will increase the range more  then a Ikea repeater ( https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html )

axiom00, 2019-02-19:
@SBTephan - Thanks. I was using a CC2530 as coordinator and CC2530 as router. This weekend, I switched to a CC2530+CC2591 as the coordinator and CC2530 as the router. I saw roughly 10-15% increase in linkquality. I wasn't impress with the initial setup so I tried switching out the antenna on the CC2530 router and saw an increase in linkquality between coordinator/router. I have another antenna on order for the CC2530+CC2591, hopefully it'll increase the range a bit more.

I'll go get the Ikea repeater tomorrow and try it out. Hopefully it'll have the same range or better. They are lot better looking than the CC2530 with a big antenna sticking out of it. 😄

axiom00, 2019-02-23:
@Koenkk  - if I want to add support for the E1746 repeater to my existing instant of zigbee2mqtt, do I just add the following to devices.js or do I need to update something else? Thanks.

zigbeeModel: ['TRADFRI signal repeater'],
model: 'E1746',
description: 'TRADFRI signal repeater',
supports: '',
vendor: 'IKEA',
fromZigbee: [],
toZigbee: [],

Koenkk, 2019-02-24:
Just use the dev branch, however adding support for it doesnt really matter, the only change that you will see is that the unsupported device message isnt shown. When paired, it already functions as a router.

dh-harald, 2019-03-12:
I've got this message from the repeater, and it looks, there's no link quality in the topic
```
  zigbee2mqtt:warn 3/12/2019, 8:46:15 PM No converter available for 'E1746' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"10":[69,49,55,52,54]}}'
  zigbee2mqtt:warn 3/12/2019, 8:46:15 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/12/2019, 8:46:16 PM No converter available for 'E1746' with cid 'haDiagnostic', type 'devChange' and data '{"cid":"haDiagnostic","data":{"lastMessageLqi":116,"lastMessageRssi":-71}}'
  zigbee2mqtt:warn 3/12/2019, 8:46:16 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

Koenkk, 2019-03-13:
@dh-harald you wont see that linkquality in that part of the message, it's retrieved here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/deviceReceive.js#L125

dh-harald, 2019-03-13:
In this case, those aren't connected...
I'm using the latest version 1.2.1, and I still don't have link quality... There's no topic either... (I've got only the config topic)

kennymc-c, 2019-03-22:
Same problem here. The router was found and added automatically but there are no updates of the device. However, the router can be seen on the network map and the link quality and online status also appears there.

aaamoeder, 2019-03-26:
same here, repeater works fine, but no reported linkquality..

omer1141, 2019-04-06:
I have tried to add 4 Ikea Tradfri Routers (E1746) on four different floors in my house (the CC2530 is on the first floor). I'm using version 1.2.1 with the suggested Firmware 20190223.
The router on the same floor as the CC2530 works without problems. The other three Ikea Routers I've 'learned in' close to the CC2530 and then I've moved them to the other floors. Unfortunately all of my Xiaomi-Clients on the three upper floors  (temperatur sensor, presence sensors, movement sensor, cube, ...) don't really continue to work (I wanted to replace my four Xiaomi Wired-Switch routers with the smaller Ikea rounters). Any help to get the Ikea routers to work would be appreciated.

See network diagram attached:
[image]

tiagofreire-pt, 2019-04-12:
The same here, no link quality, just reports "0".

sunflowerABB, 2019-05-04:
Tried for several hours to pair one of my Tradfri Repeaters E1746 with my Zigbee2mqtt instance without luck. The repeater is not even being detected it seems from the debug log output. My CC2531 is running on Coordinator firmware version: '20180507' - could this be the culprit and I need to be on a newer firmware version? Any other clues?

sunflowerABB, 2019-05-05:
Update: Updating the firmware to a 2019' version solved the problem and the Tradfri repeater was instantly recognised and paired!

samnup, 2019-05-06:
Did you update the tradfri repeaters firmware or zigbee2mqtt?

boojew, 2019-05-14:
@sunflowerABB did you do anything else than update the coordinator firmare? I'm running the latest coordinator + zigbee2mqtt and it won't pair :(

PuckStar, 2019-05-19:
Same here. Had it paired in the past but wasn't working (no devices would connect to it).
So I removed it with the zigbee2mqtt/bridge/config/remove command and wanted to pair it again. But now it's not pairing. Distance to coordinator is just 1,5m.

mclaudiopt, 2019-07-08:
hi there to all...my ikea is not repeating...what can i do?
my firmware of coordinator is 20190425

AdamWhittingham, 2019-07-20:
I'm also having problems with 3 new repeaters.

These were purchased at the start of July 2019 in the UK and are showing up as `E1746` devices.
I have paired them by pushing the reset pin until the LED pulses, and after a few seconds it has gone steady. I then moved the repeaters near my weak signal areas. 

No link status is showing up and I'm often seeing errors in the logs such as:
`7/20/2019, 11:50:02 AM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe9a72bd/availability', payload 'offline'`
and
`7/20/2019, 11:53:56 AM Failed network lqi scan for device: '0x000d6ffffe9a72bd' with error: 'Error: Timed out after 10000 ms'`

If I graph my network, I see everything is trying to link directly to the router and the repeaters show 0 link strength. I've tried resetting the repeaters one at a time and rejoining them but to no avail.

I'm not sure what else I can try- does anyone have any pointers please?

(Also apologies for adding to a closed issue- please let me know if opening a new issue would be helpful. And *thank you* for such a marvellous tool, it's the bedrock of my home automation and I love it!)

PuckStar, 2019-07-27:
> I'm also having problems with 3 new repeaters.
>
> These were purchased at the start of July 2019 in the UK and are showing up as `E1746` devices.
> I have paired them by pushing the reset pin until the [...]

Did you get it working?
I still have the same issue. No devices are connecting to the Ikea repeater :(.

AdamWhittingham, 2019-07-27:
@PuckStar Sadly not yet- I've left them running for a week, I've tried resetting & re-pairing them, and I've tried resetting and re-pairing some of my devices just in case.

I've occasionally see a device attach to them but it doesn't seem to last. My boosters are all about 3, 5 and 10 meters from my USB adaptor and each in different directions, yet none seem to mesh or stay connected. One of my lights, which is about 6 meters from the adaptor, seems to be favoured as a coordinator for half my devices now.

I might try setting up the Tradfri hub again and pair the boosters to that and see if any firmware updates are available. I'll let you know if that works!

VeeYarr, 2019-08-01:
Wish I'd seen this before I went and bought 5 IKEA repeaters - they pair fine but I can't get anything to pair to them.....

Any updates which might help?

PuckStar, 2019-08-01:
> 
> 
> Wish I'd seen this before I went and bought 5 IKEA repeaters - they pair fine but I can't get anything to pair to them.....
> 
> Any updates which might help?

On a [Dutch forum Koen replied](https://gathering.tweakers.net/forum/list_message/59572330#59572330) on my question if the repeater works or not:
"It depends on the devices that you want to pair. He expects tradfri devices to work fijn but for xiaomi/aqara it might be more difficult. It's not something that can be changed on the zigbee2mqtt side."

So I guess you can be lucky or not.
Ikea takes goods back for like a year or so. So you can also return them and get your money back.

VeeYarr, 2019-08-01:
I moved the repeater and the Xiaomi sensor is reporting much better link quality....

How do I tell if the Xiaomi sensor is using the repeater?

PuckStar, 2019-08-01:
> 
> 
> I moved the repeater and the Xiaomi sensor is reporting much better link quality....
> 
> How do I tell if the Xiaomi sensor is using the repeater?

With the map: https://community.home-assistant.io/t/zigbee2mqtt-show-the-networkmap-in-home-assistant/89116

VeeYarr, 2019-08-01:
> > I moved the repeater and the Xiaomi sensor is reporting much better link quality....
> > How do I tell if the Xiaomi sensor is using the repeater?
> 
> With the map: https://community.home-assistant.io/t/zigbee2mqtt-show-the-networkmap-in-home-assistant/89116

Can't seem to get the map to draw, I see the mqtt messages but I don't get a map in the iframe in lovelace, not sure why...

RIMperator, 2019-08-02:
You can always request the map data by sending the message zigbee2mqtt/bridge/networkmap and then use webgraphviz.com to visualize it

AdamWhittingham, 2019-08-25:
I've reset and re-paired my three with varying results:
2 of them seem to have paired correctly and the map is now showing other devices being routed from them!
The third one showed an error (see #1886 which I'm still figuring out).

I suspect that, despite buying mine together, one of them might have a slightly different firmware version or something. Either way, I'm seeing better performance now.

I had already reset & re-paired all of these a couple of times before and I didn't do anything different this time. Whilst that's not very helpful, I hope it helps someone to hear it!

boojew, 2019-08-25:
I think our expectations of how this should work may need to be tempered.
I’ve had similar results, but what I find is the mesh network changes over
time. Sometimes, nothing is using the repeaters, other times a handful of
devices. Interference and WiFi usage likely lead the network self-healing
and re-optimizing somewhat frequently. I think also different vendor zigbee
stack implementations likely play a big part of this.

PeeVv, 2019-11-25:
Today my Tradfri repeater suddenly totally stopped working after working without problems for months. No updates or changes made. It shows up in the map with a link quality of 0. Tried to reset it multiple times but it just won't connect anymore in any way. Keep getting the "Failed to execute LQI" error. Removing it is impossible (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms).

Is it better to just return it and get one of the bulbs as repeater?

radiocolin, 2022-08-12:
I just got some E1746s and I'm seeing them drop offline constantly. If I reboot them they come back for a couple hours.

2lman, 2022-12-09:
> @axiom00
> 
> Works great. Didn't had reception in the garage, now I have

How did you get devices to use it?  I can pair it with my sonoff v3 usb stick but devices don't see to use it.  Someone suggested pairing devices with the repeater which I did that no problem but then they were not seen by HA,.