Koenkk/zigbee2mqtt #1101: Stability problem (more month)

MikeVRT, 2019-02-15:
Perhaps I found "longtime" stability problem - zigbee2mqtt stay unstable after big log.txt (10 MB) and starts new log1.txt.

In my case its take more then 1 month, and then log looks like:
```
2019-2-12 07:19:12 - error: Not connected to MQTT server!
2019-2-12 07:19:12 - error: Cannot send message: topic: 'zigbee2mqtt/router_cc2531', payload: '{"state":false,"linkquality":0}
2019-2-12 07:19:15 - error: Not connected to MQTT server!
2019-2-12 07:19:25 - error: Not connected to MQTT server!
2019-2-12 07:19:30 - error: Not connected to MQTT server!
2019-2-12 07:19:30 - error: Cannot send message: topic: 'zigbee2mqtt/bedroom_thp', payload: '{"temperature":19.87,"linkquality":56,"humidity":23.94,"pressure":990,"battery":99,"voltage":3015}
2019-2-12 07:19:30 - error: Not connected to MQTT server!
2019-2-12 07:19:30 - error: Cannot send message: topic: 'zigbee2mqtt/bedroom_thp', payload: '{"temperature":19.83,"linkquality":56,"humidity":24.09,"pressure":990.5,"battery":99,"voltage":3015}
2019-2-12 07:19:35 - error: Not connected to MQTT server!
2019-2-12 07:19:45 - error: Not connected to MQTT server!
2019-2-12 07:19:55 - error: Not connected to MQTT server!
2019-2-12 07:20:05 - error: Not connected to MQTT server!
2019-2-12 07:20:12 - error: Not connected to MQTT server!
2019-2-12 07:20:12 - error: Cannot send message: topic: 'zigbee2mqtt/router_cc2531', payload: '{"state":false,"linkquality":43}
2019-2-12 07:20:15 - error: Not connected to MQTT server!
2019-2-12 07:20:25 - error: Not connected to MQTT server!
2019-2-12 07:20:35 - error: Not connected to MQTT server!
2019-2-12 07:20:45 - error: Not connected to MQTT server!
2019-2-12 07:20:55 - error: Not connected to MQTT server!
2019-2-12 07:21:05 - error: Not connected to MQTT server!
2019-2-12 07:21:12 - error: Not connected to MQTT server!
2019-2-12 07:21:12 - error: Cannot send message: topic: 'zigbee2mqtt/router_cc2531', payload: '{"state":false,"linkquality":43}
2019-2-12 07:21:15 - error: Not connected to MQTT server!
2019-2-12 07:21:25 - error: Not connected to MQTT server! 
```

Can you check this part (10M+ = log->log1) of code, please?

Koenkk, 2019-02-15:
Looking at the log, it seems that it cannot connect to the MQTT server, I don't see a direct relation with this and the log size. 

Are you sure that your MQTT server is still running? Can you connect with e.g. MQTT.fx (https://mqttfx.jensd.de/)

MikeVRT, 2019-02-16:
> Looking at the log, it seems that it cannot connect to the MQTT server
yes, but it's only final, I think.

In reality restart of only zigbee2mqtt service helps (new log started). And restart of HASS or power off/on of USB Stick (CC2531) not needed.

Koenkk, 2019-02-16:
So did you try if you can still connect to the MQTT server when this happens?

MikeVRT, 2019-02-17:
I use buildin HASS mqtt, in preview case it's works, yes. When situation repeated (for repeat we must don't touch anything more than month...) - I only restarted zigbee2mqtt service, HASS still working ok, so buildin mqtt too again, I think.

marc-gist, 2019-02-18:
Have you been able to repeat the issue? MQTT could be the issue as I've had the mosquitto mqtt broker go down without reason before.

MikeVRT, 2019-02-18:
@marc-gist, HASS contains an embedded MQTT broker called HBMQTT, it's still works when problem happens and not needed to restart after restart of zigbee2mqtt.
I repeated this issue only 2 times, because it's need to many time for happen...

prein2, 2019-02-20:
last week I had many things logging (debug mode and node-red dashboard generating lots of logging). within 1 day I had log1, log2 and log3. Everything still worked. So not shure the problem lies in the log names/filesize. hass integrated mqtt broker is often problematic, even within hass. hass also suggests using mosquitto is the better option. Mosquitto can be installed on the same instance as hass, so why not test it?

MikeVRT, 2019-02-20:
> hass integrated mqtt broker is often problematic
Only one known issue was about memory leak in some cases, and now it's problem deleted from description - solved. I have no any problem with it at all. in my env. memory leaks never happens - I can show mem graph. log.
(in my case not debug mode and month+ work - it's more stressful then one week, but about log was just a hunch of cause)

closed 2019-04-28 by stale[bot]