Koenkk/zigbee2mqtt #1102: npm update error

m4dm4xi, 2019-02-15:
I´ve been running version 1.1.0 and I´m trying to update to 1.1.1.

I´m following the steps as mentioned in [https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html](url) in step 6.

However, during command `npm install` I get the error:

` sudo npm install
npm WARN tar ENOENT: no such file or directory, open '/home/pi/Envs/zigbee2mqtt/node_modules/.staging/moment-222a9672/package.js'
npm WARN tar ENOENT: no such file or directory, open '/home/pi/Envs/zigbee2mqtt/node_modules/.staging/moment-222a9672/README.md'
npm WARN tar ENOENT: no such file or directory, open '/home/pi/Envs/zigbee2mqtt/node_modules/.staging/eslint-53bb16b8/LICENSE'
npm WARN tar ENOENT: no such file or directory, open '/home/pi/Envs/zigbee2mqtt/node_modules/.staging/eslint-53bb16b8/README.md'
npm WARN tar ENOENT: no such file or directory, open '/home/pi/Envs/zigbee2mqtt/node_modules/.staging/acorn-9d9a3368/dist/acorn.mjs'
npm WARN tar ENOENT: no such file or directory, open '/home/pi/Envs/zigbee2mqtt/node_modules/.staging/mqtt-ed360657/dist/mqtt.min.js'
npm WARN tar ENOENT: no such file or directory, open '/home/pi/Envs/zigbee2mqtt/node_modules/.staging/mocha-f3138ebb/README.md'
npm WARN tar ENOENT: no such file or directory, open '/home/pi/Envs/zigbee2mqtt/node_modules/.staging/sinon-97001719/pkg/sinon-no-sourcemaps.js'
npm ERR! err.code.match is not a function

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-02-15T18_46_40_392Z-debug.log
`

This Error ist also referneced here:
[https://github.com/ONLYOFFICE/DocumentServer/issues/478](url)

A hint over there is:

> In case anyone else runs in to this issue exact same error message, the culprit has to do with npm >trying to clone something that is not a valid git reference.
>
>In my case (not using DocumentServer) I was referencing a branch in an external repository that had >recently been forced-pushed to. Keep in mind that even tho you put a branch name in your >`package.json` file... npm will turn that in to a specific commit hash in `package-lock.json`. Since that >specific hash didn't exist... `npm install` (or `npm ci`) is not able to pull data and hence things fail.
>
>TL;DR: check your package.json file to see if it is referencing branches (in external repos) that recently >changed or no longer exist.

Where is the problem?

Koenkk, 2019-02-15:
Can you do a `rm -rf node_modules && npm --version && npm install` and provide the complete log?

sebastianheierhoff, 2019-02-15:
I think I am running into the same error:
[2019-02-15T21_25_07_852Z-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/2870936/2019-02-15T21_25_07_852Z-debug.log)

Koenkk, 2019-02-15:
Can you check if this issue still exists in the dev branch?

sebastianheierhoff, 2019-02-16:
Yes, the issue is still present in the current dev

sebastianheierhoff, 2019-02-16:
As you can see from the log, I was running node v11.10.0 and npm  v6.8.0. Downgraded to node v10.15.1 this morning, to make sure that this wasn't caused by the wrong node version. Now I get:

```
Unhandled rejection Error: Command failed: /usr/bin/git clone --depth=1 -q -b master https://github.com/Koenkk/zigbee-shepherd.git /root/.npm/_cacache/tmp/git-clone-ac9b3fac/@serialport/parser-regex-130478c5 (3525ms)
fatal: Konnte führende Verzeichnisse von '/root/.npm/_cacache/tmp/git-clone-ac9b3fac' nicht erstellen.: Keine Berechtigung

    at ChildProcess.exithandler (child_process.js:294:12)
    at ChildProcess.emit (events.js:189:13)
    at maybeClose (internal/child_process.js:970:16)
    at Socket.stream.socket.on (internal/child_process.js:389:11)
    at Socket.emit (events.js:189:13)
    at Pipe._handle.close (net.js:600:12)

Unhandled rejection Error: Command failed: /usr/bin/git clone --depth=1 -q -b master https://github.com/Koenkk/zcl-id.git /root/.npm/_cacache/tmp/git-clone-d3941b00
fatal: Konnte führende Verzeichnisse von '/root/.npm/_cacache/tmp/git-clone-d3941b00' nicht erstellen.: Keine Berechtigung

    at ChildProcess.exithandler (child_process.js:294:12)
    at ChildProcess.emit (events.js:189:13)
    at maybeClose (internal/child_process.js:970:16)
    at Socket.stream.socket.on (internal/child_process.js:389:11)
    at Socket.emit (events.js:189:13)
    at Pipe._handle.close (net.js:600:12)

Unhandled rejection Error: Command failed: /usr/bin/git clone --depth=1 -q -b master https://github.com/Koenkk/zcl-id.git /root/.npm/_cacache/tmp/git-clone-f43af1c2
fatal: Konnte führende Verzeichnisse von '/root/.npm/_cacache/tmp/git-clone-f43af1c2' nicht erstellen.: Keine Berechtigung

    at ChildProcess.exithandler (child_process.js:294:12)
    at ChildProcess.emit (events.js:189:13)
    at maybeClose (internal/child_process.js:970:16)
    at Socket.stream.socket.on (internal/child_process.js:389:11)
    at Socket.emit (events.js:189:13)
    at Pipe._handle.close (net.js:600:12)

Unhandled rejection Error: Command failed: /usr/bin/git clone --depth=1 -q -b master https://github.com/Koenkk/cc-znp.git /root/.npm/_cacache/tmp/git-clone-8130ee6f
fatal: Konnte führende Verzeichnisse von '/root/.npm/_cacache/tmp/git-clone-8130ee6f' nicht erstellen.: Keine Berechtigung

    at ChildProcess.exithandler (child_process.js:294:12)
    at ChildProcess.emit (events.js:189:13)
    at maybeClose (internal/child_process.js:970:16)
    at Socket.stream.socket.on (internal/child_process.js:389:11)
    at Socket.emit (events.js:189:13)
    at Pipe._handle.close (net.js:600:12)

Unhandled rejection Error: Command failed: /usr/bin/git clone --depth=1 -q -b master https://github.com/Koenkk/zcl-packet.git /root/.npm/_cacache/tmp/git-clone-0f87f79a
fatal: Konnte führende Verzeichnisse von '/root/.npm/_cacache/tmp/git-clone-0f87f79a' nicht erstellen.: Keine Berechtigung

    at ChildProcess.exithandler (child_process.js:294:12)
    at ChildProcess.emit (events.js:189:13)
    at maybeClose (internal/child_process.js:970:16)
    at Socket.stream.socket.on (internal/child_process.js:389:11)
    at Socket.emit (events.js:189:13)
    at Pipe._handle.close (net.js:600:12)

npm ERR! cb() never called!

npm ERR! This is an error with npm itself. Please report this error at:
npm ERR!     <https://npm.community>

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-02-16T11_41_26_176Z-debug.log
```

Koenkk, 2019-02-16:
This really looks like a npm error:

```
npm ERR! This is an error with npm itself. Please report this error at:
npm ERR!     <https://npm.community>
```

I'm afraid I can't help you with this.

sebastianheierhoff, 2019-02-17:
I think you're right. Did a clean install and now it's working again.

closed 2019-02-17 by Koenkk