Koenkk/zigbee2mqtt #1103: [Home Assistant] state_changed event fires twice on single MQTT message

dyrkin, 2019-02-15:
**Problem**
When I press a button on my Xiaomi Aqara Wireless Switch, single MQTT message is sent to the home assistant.
Home assistant generates two identical events and both trigger my automation.
I'm expecting only on event per one MQTT message.

**Root cause**

> And the issue in your case is the use of the same topic in json_attributes_topic and state_topic combined with force_update true.

> You'll get to state updates, one for the changed state and one for the changed attributes. That is how the MQTT sensor architecture works and won't be changed. The zigbee2mqtt creators should a) not use force_update (also improves HA's state machine performance or b) create separate sensors for each attribute.

> You can subscribe to the topic directly in the automation too. That way you're guaranteed to only get one trigger per message.

**More details** https://github.com/home-assistant/home-assistant/issues/21082

dyrkin, 2019-02-15:
Discovery payloads:

**Xiaomi Aqara Wireless Switch** - _click_
**topic**: homeassistant/sensor/0x00158d0002d7b862/click/config
```json
{
  "icon": "mdi:toggle-switch",
  "value_template": "{{ value_json.click }}",
  "force_update": true,
  "state_topic": "zigbee2mqtt/living_room_xiaomi_double_button_1",
  "json_attributes_topic": "zigbee2mqtt/living_room_xiaomi_double_button_1",
  "name": "living_room_xiaomi_double_button_1_click",
  "unique_id": "0x00158d0002d7b862_click_zigbee2mqtt",
  "device": {
    "identifiers": "zigbee2mqtt_0x00158d0002d7b862",
    "name": "living_room_xiaomi_double_button_1",
    "sw_version": "Zigbee2mqtt 1.1.1",
    "model": "Aqara double key wireless wall switch (WXKG02LM)",
    "manufacturer": "Xiaomi"
  },
  "availability_topic": "zigbee2mqtt/bridge/state"
}
```

**Xiaomi Aqara Wireless Switch** - _battery_
**topic**: homeassistant/sensor/0x00158d0002d7b862/battery/config
```json
{
  "unit_of_measurement": "%",
  "device_class": "battery",
  "value_template": "{{ value_json.battery }}",
  "state_topic": "zigbee2mqtt/living_room_xiaomi_double_button_1",
  "json_attributes_topic": "zigbee2mqtt/living_room_xiaomi_double_button_1",
  "name": "living_room_xiaomi_double_button_1_battery",
  "unique_id": "0x00158d0002d7b862_battery_zigbee2mqtt",
  "device": {
    "identifiers": "zigbee2mqtt_0x00158d0002d7b862",
    "name": "living_room_xiaomi_double_button_1",
    "sw_version": "Zigbee2mqtt1.1.1",
    "model": "Aqara double key wireless wall switch (WXKG02LM)",
    "manufacturer": "Xiaomi"
  },
  "availability_topic": "zigbee2mqtt/bridge/state"
}
```

Koenkk, 2019-02-15:
As mentioned by balloob [here](https://github.com/home-assistant/home-assistant/pull/20716#issuecomment-460368012), and as explained in the [zigbee2mqtt documentation](http://www.zigbee2mqtt.io/integration/home_assistant.html#responding-to-button-clicks) you should use the MQTT trigger.

closed 2019-02-15 by Koenkk