Koenkk/zigbee2mqtt #1104: New device Paulmann zigbee remote

MarcSN311, 2019-02-15:
Hello!

I am trying to add a new device, the Paulman zigbee remote: https://de.paulmann.com/innenleuchten/smart-home/zigbee/smarthome-zigbee-fernbedienung-gent-mit-wandhalterung/50067

This is my device entry:
```
    {
        zigbeeModel: ['RemoteControl '],
        model: '50067',
        vendor: 'Paulmann',
        descriptio: 'Zigbee Remote',
        supports: 'remote',
        fromZigbee: [],
        toZigbee: [],
    },
```

The log when pairing looks like this:
https://pastebin.com/tkM99syb

When pressing a button on the remote I get nothing. Any advice on debugging that?

Thanks,
Marc

Koenkk, 2019-02-15:
Maybe this device requires the coordinator to be added to a group, please follow https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/coordinator_group.md

MarcSN311, 2019-02-17:
Thank you! I'll get another CC2531 stick and try sniffing!

closed 2019-04-25 by stale[bot]

martin3000, 2019-08-16:
> Thank you! I'll get another CC2531 stick and try sniffing!

What groups did it use?