Koenkk/zigbee2mqtt #1105: zigbee2mqtt crashing when lights are turned on

SBTephan, 2019-02-15:
I bought the Ikea range extender, to support it I updated the zigbee2mqtt through the update script provided.

After that all the devices that where connected cannot be reached anymore. So I thought, maybe the CC2531 needed an update two. I saw there was a new version so I flashed the new version on the CC2531.

The devices still wouldn't connect so I removed all of the devices from the configuration.yaml and reconnected all devices. Everything worked fine.

Until the next day domoticz managed to switch the ikea lights on but then all kind of errors showed up in the logging and It looked liked the process crashed because the sensors could not deliver there messages anymore, Also the lights could not be turned on or off.

2/15/2019, 5:35:12 PM - error: Zigbee publish to device '0xd0cf5efffec159ad', lightingColorCtrl - moveToColorTemp - {"colortemp":492,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
2/15/2019, 5:35:32 PM - error: Zigbee publish to device '0xd0cf5efffec01924', genLevelCtrl - moveToLevelWithOnOff - {"level":137,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
2/15/2019, 5:35:33 PM - error: Zigbee publish to device '0xd0cf5efffec01924', lightingColorCtrl - moveToColorTemp - {"colortemp":486,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
2/15/2019, 5:35:33 PM - error: Zigbee publish to device '0xd0cf5efffec159ad', genLevelCtrl - moveToLevelWithOnOff - {"level":137,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
2/15/2019, 5:35:50 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00027a3c4d', payload '{"occupancy":false,"linkquality":55,"battery":100,"voltage":3005}'
2/15/2019, 9:49:10 PM - info: Zigbee publish to device '0x000b57fffec57fbc', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/15/2019, 9:49:14 PM - error: Zigbee publish to device '0x000b57fffec57fbc', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
2/15/2019, 9:49:18 PM - info: Zigbee publish to device '0x000b57fffec02d51', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/15/2019, 9:49:21 PM - info: Zigbee publish to device '0xd0cf5efffec01924', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/15/2019, 9:49:21 PM - error: Zigbee publish to device '0x000b57fffec02d51', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
2/15/2019, 9:49:22 PM - info: Zigbee publish to device '0xd0cf5efffec159ad', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2/15/2019, 9:49:24 PM - error: Zigbee publish to device '0xd0cf5efffec01924', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
2/15/2019, 9:49:27 PM - error: Zigbee publish to device '0xd0cf5efffec159ad', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout

I needed to reboot the pi so that everything could work again. (restarting the proces with systemctl didn't work)

What am I doing wrong here? Is there something with the new firmware? with the update process did I destroyed something (weird that I needed to readd the devices)

Thanks in advance,

Koenkk, 2019-02-15:
Can you try with the latest dev branch?

SBTephan, 2019-02-15:
I can try, Is it stable? And what is the best way to install it?

Also, does it require reconnecting the devices again?

Koenkk, 2019-02-15:
@SBTephan lets than 1.1 but still good, just use git to checkout the `dev` branch. Device don't have to be reconnected.

SBTephan, 2019-02-19:
Strangely enough it works fine for a few days now, didn't change anything. 
Think I am going to reinstall everything just to be sure the SD card is not corrupting..

closed 2019-04-27 by stale[bot]