Koenkk/zigbee2mqtt #1106: Need help understanding my networkmap

OmerZfira, 2019-02-16:
Hi, I had the following map outputted from zigbee2mqtt/bridge/networkmap/graphviz (and then http://www.webgraphviz.com )  but I can't seem to follow the topology.

https://imgur.com/IXZSQYC\

Assuming that the Coordinator is my cc2531 dongle...
1) How come an Endpoint device (sensor_door_main) is connected to 2 devices? Shouldn't endpoints connect only to one? 
2) Shouldn't the arrows be reversed? i.e. Should be: socket_1 --> sensor_door_main.

clockbrain, 2019-02-16:
1 - Network map is simply a scan of neighbor tables for the coordinator and each router. Although an end device is only paired (at any one time) with a single parent router, its signal may have been picked up by other routers and exist in their neighbor tables. At the moment the network map code doesn't distinguish between actual child end devices and other end devices it may have heard from. To fix would need to include neighbor.relationship in the lqi scan return from zigbee-shepherd here: [https://github.com/Koenkk/zigbee-shepherd/blob/98f73de091a0f0259a031750d893a077e1beb7f4/lib/shepherd.js#L415](https://github.com/Koenkk/zigbee-shepherd/blob/98f73de091a0f0259a031750d893a077e1beb7f4/lib/shepherd.js#L415)

Here's an example of the full neighbor table available:
```
     { extPandId: '0xdddddddddddddddd',
       extAddr: '0x00xxxxxxxxxxx',
       nwkAddr: 64964,
       deviceType: 1,
       rxOnWhenIdle: 1,
       relationship: 3,
       permitJoin: 1,
       depth: 1,
       lqi: 15 },
```
Then would have to change the network map code to filter links only for real children. Explanation of neighbor table is here [https://microchip.secure.force.com/microchipknowledge/articles/en_US/FAQ/Neighbor-table-in-ZigBee-BitCloud-SDK](https://microchip.secure.force.com/microchipknowledge/articles/en_US/FAQ/Neighbor-table-in-ZigBee-BitCloud-SDK).

2 - depends on your interpretation - routers hold messages addressed to their child end devices, it is up to the end device to poll their parent router and pick up any such pending messages so the arrow direction could be considered to reflect that polling.

OmerZfira, 2019-02-17:
Thanks for the elaborated answer!
I now know how to proceed :)

closed 2019-02-17 by OmerZfira