Koenkk/zigbee2mqtt #1107: OSRAM LIGHTIFY LED Classic A60 clear (AC03641) issue with turning on/off

nerdosity, 2019-02-16:
Hello,

Running latest Home Assistant in a docker (latest version of synology), running zigbee2mqtt in a docker (latest version of zigbee2mqtt, 1.1.1), bought the usb key from the cheaper Italian auction that I could find:

https://www.ebay.it/itm/132934011667

The seller said to me that it was flashed with latest january firmware.

So, the problem is this: I can pair and use without problem a couple of Xiaomi devices, but the Osram light is acting really strange. 

[image]
The light is actually OFF, but you can see in Home Assistant that the status is ON, and there is no relay behind. 

After pairing it was immediately visible, but it's not working really well. Sometimes I can turn it off easily but it will not turn on, sometimes the light level is "flashing" after changing it from Home assistant, sometimes it turns off but on home assistant it stay "ON" until turning on and off again.

https://pastebin.com/fbznWG3K

Koenkk, 2019-02-16:
- Can you check if it still happens with the `-dev` docker container? 
- Can you also post the start of the log.

nerdosity, 2019-02-20:
https://pastebin.com/2jNzzdW6 

This is with latest-dev docker container. Behaviour is exactly the same.

Koenkk, 2019-02-20:
Can you also try with the latest dev firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin (doens't require repairing: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/what_does_and_doesnt_require_repairing.md)

nerdosity, 2019-02-21:
https://pastebin.com/0pDB7zEK

I have two more dimmable smart bulb, I can try them, maybe is it an hardware issue?

nerdosity, 2019-02-21:
https://pastebin.com/2jhP7v7E

This is another bulb, same type, took out from the box and connected just now. Same issue.

Koenkk, 2019-02-22:
Somehow the bulbs don't respond.

Let's see if we can poll the state after the set.

Can you add to: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/toZigbee.js#L54
```
readAfterWriteTime: 250,
```

nerdosity, 2019-03-16:
Hi, was this modification included with latest version? I don't know how to edit the js file, I'm using a docker.
I updated zigbee2mqtt to latest version, and flashed the latest version of firmware on the CC2531.
It seems nothing changed, this is from 10 minutes ago:

https://pastebin.com/KUyfSxBu

Koenkk, 2019-03-16:
Can you replace the following configuration in `devices.js` https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L1150

```js
    {
        zigbeeModel: ['Classic A60 W clear - LIGHTIFY'],
        model: 'AC03641',
        vendor: 'OSRAM',
        description: 'LIGHTIFY LED Classic A60 clear',
        extend: generic.light_onoff_brightness,
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 3);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

You need to go inside the docker container and execute `nano /app/node_modules/zigbee-shepherd-converters/devices.js`

nerdosity, 2019-03-17:
Ok, didn't know I could edit core files inside a docker. I will try to edit it and report it later.

closed 2019-05-23 by stale[bot]