Koenkk/zigbee2mqtt #1110: Zigbee2mqtt does not quit when unable to start (docker)

peterforeman, 2019-02-17:
When zigbee2mqtt is unable to start, the application does not exist which prevents futher actions. In my case after restarting the container zigbee2mqtt is often unable to successfully start. After pkill-ing node in the container itself for a couple of times it does work. 

See log below in which, after getting a "Error while starting" I kill the node process so that it starts again. As you can see, after a couple of times zigbee2mqtt starts and everything is working.

Is it possible for zigbee2mqtt to just exit after starting up does not work?

```
> zigbee2mqtt@1.1.1 start /app
> node index.js

  zigbee2mqtt:info 2019-2-17 12:16:22 Logging to directory: '/app/data/log/2019-02-17.11-16-22'
  zigbee2mqtt:info 2019-2-17 12:16:22 Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
  zigbee2mqtt:info 2019-2-17 12:16:22 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-17 12:16:25 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2019-2-17 12:17:25 Starting zigbee-shepherd
  zigbee2mqtt:error 2019-2-17 12:17:28 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2019-2-17 12:17:28 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2019-2-17 12:17:28 Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
Starting... npm

> zigbee2mqtt@1.1.1 start /app
> node index.js

  zigbee2mqtt:info 2019-2-17 13:15:19 Logging to directory: '/app/data/log/2019-02-17.12-15-19'
  zigbee2mqtt:info 2019-2-17 13:15:20 Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
  zigbee2mqtt:info 2019-2-17 13:15:20 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-17 13:15:23 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
Starting... npm

> zigbee2mqtt@1.1.1 start /app
> node index.js

  zigbee2mqtt:info 2019-2-17 13:15:34 Logging to directory: '/app/data/log/2019-02-17.12-15-34'
  zigbee2mqtt:info 2019-2-17 13:15:35 Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
  zigbee2mqtt:info 2019-2-17 13:15:35 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-17 13:15:38 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
Starting... npm

> zigbee2mqtt@1.1.1 start /app
> node index.js

  zigbee2mqtt:info 2019-2-17 13:15:41 Logging to directory: '/app/data/log/2019-02-17.12-15-41'
  zigbee2mqtt:info 2019-2-17 13:15:42 Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
  zigbee2mqtt:info 2019-2-17 13:15:42 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-17 13:15:45 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
Starting... npm

> zigbee2mqtt@1.1.1 start /app
> node index.js

  zigbee2mqtt:info 2019-2-17 13:16:13 Logging to directory: '/app/data/log/2019-02-17.12-16-13'
  zigbee2mqtt:info 2019-2-17 13:16:13 Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
  zigbee2mqtt:info 2019-2-17 13:16:13 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-17 13:16:16 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
Starting... npm

> zigbee2mqtt@1.1.1 start /app
> node index.js

  zigbee2mqtt:info 2019-2-17 13:16:20 Logging to directory: '/app/data/log/2019-02-17.12-16-20'
  zigbee2mqtt:info 2019-2-17 13:16:20 Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
  zigbee2mqtt:info 2019-2-17 13:16:20 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-17 13:16:29 zigbee-shepherd started
  zigbee2mqtt:info 2019-2-17 13:16:29 Coordinator firmware version: '20190109'
  zigbee2mqtt:info 2019-2-17 13:16:29 Currently 10 devices are joined:
[snip]
```

Koenkk, 2019-02-18:
implemented!

closed 2019-02-18 by Koenkk

JLFN, 2019-02-25:
@peterforeman  did you enable this patch for ? 
https://hub.docker.com/r/forepe/zigbee2mqtt-socat
Can you att a devbranch for it or upload the docker file to @Koenkk

peterforeman, 2019-02-28:
I added a latest-dev branch to zigbee2mqtt-socat which uses the zigbee2mqtt:latest-dev as base.

JLFN, 2019-02-28:
@peterforeman  It doesnt work on raspberry pi 3  with docker.
errorlog:
standard_init_linux.go:207: exec user process caused "exec format error"

standard_init_linux.go:207: exec user process caused "exec format error"

[image]