Koenkk/zigbee2mqtt #1111: New Philips Hue Outdoor Sensor SML002

jonnycastaway, 2019-02-17:
Can anyone help to integrate the new Philips Hue Outdoor Motion Sensor?
It connects to zigbee2mqtt as Device with modelID 'SML002' is not supported.

nanosonde, 2019-02-18:
@jonnycastaway 
Have you read this? https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

nanosonde, 2019-02-18:
Just add this as a next step to node_modules/zigbee-shepherd-converters/devices.js:
```
{
    zigbeeModel: ['SML002'], // The model ID from: Device with modelID 'SML002' is not supported.
    model: '8718699625474', // Vendor model number, look on the device for a model number
    vendor: 'Philips', // Vendor of the device (only used for documentation and startup logging)
    description: 'Philips hue Motion Outdoor Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'motion', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},
```

nanosonde, 2019-02-18:
There is already an issue about this device: https://github.com/ioBroker/ioBroker.zigbee/issues/194

jonnycastaway, 2019-02-18:
Hi,
is i have read this.

> @jonnycastaway
> Have you read this? https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

But i was uncertain what data to be inserted. I do the changes later and test it.

jonnycastaway, 2019-02-18:
ok guys,
got it working! i followed @nanosonde hint, salted it with the configuration from the indoor hue motion sensor and bam. motion is triggered, temperature meassured.
But after nearly 20 Messages it stopped working with:
Cannot get the Node Descriptor of the Device: 0x0017880106442d61 (Error: Timed out after 10000 ms)
Then no mqtt messages where send anymore.

Any hints?

`    {
        zigbeeModel: ['SML002'],
        model: '9290019758',
        vendor: 'Philips',
        description: 'Hue motion Outdoor sensor',
        supports: 'occupancy, temperature, illuminance',
        fromZigbee: [
            fz.hue_battery, fz.generic_occupancy, fz.generic_temperature,
            fz.ignore_occupancy_change, fz.generic_illuminance, fz.ignore_illuminance_change,
            fz.ignore_temperature_change,
        ],
        toZigbee: [tz.generic_occupancy_timeout],
        ep: (device) => {
            return {
                '': 2, // default
                'ep1': 1,
                'ep2': 2, // e.g. for write to msOccupancySensing
            };
        },
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 2);

            const actions = [
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.bind('msIlluminanceMeasurement', coordinator, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.bind('msOccupancySensing', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 1000, 0, cb),
                (cb) => device.report('msOccupancySensing', 'occupancy', 0, 600, null, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
            ];

            execute(device, actions, callback);
        },
    },`

jonnycastaway, 2019-02-18:
ok, it seems like when the sensor is not triggered many times a second it is normaly functional.
after the changes in devices.js and another zigbee2mqtt restart i removed the device from zigbee2mqtt and pair it again and now all works as expected.

closed 2019-02-20 by jonnycastaway

Koenkk, 2019-02-20:
Can you create a PR for this? In this way it can be officially supported by zigbee2mqtt.

jonnycastaway, 2019-02-20:
Done ;-)

nickcj931, 2019-02-22:
Sorry guys, im being a bit thick here. Is this now officially supported by zigbee2mqtt? Was just hunting around for an outdoor sensor and came across it got super excited.

jonnycastaway, 2019-02-23:
@nickcj931 
i'm fighting a little bit with the sensor. my tests are on the way and i hope i found the source of my problems. u can follow the testing here:
[https://github.com/Koenkk/zigbee2mqtt/issues/1140](url)

nickcj931, 2019-02-24:
@jonnycastaway thanks a mil for the reply, following along with great interest!

papanirual, 2019-02-25:
Hi everyone! I also have trouble correctly including this sensor (with hassio on a Raspberry 3b+). It is recognized by zigbee2mqtt (I used a modified devices.js in the \share\zigbee2mqtt folder) and I can see the sensor data in the zigbee log as well as with mqtt.fx. However even though I tried to re-pair the sensor it is not shown in Home Assistant on the Mosqitto integration page as new device, as my other devices did. I tried to remove the sensor from database.db an pair it again but that doesn't change anything. What am I missing? 

BTW: I think the definition above is missing the illumination measurement:

                (cb) => device.report('msIlluminanceMeasurement', 'measuredValue', 0, 600, null, cb),

Setting this also shows the illumination readings in my zigbee2mqtt log.

Thanks a lot in advance!

jonnycastaway, 2019-02-28:
In my MQTT Logs with this setting i get illuminance, but its always 0.
How it works in your setup? @papanirual

papanirual, 2019-02-28:
Illuminance works just fine. I have three sensors and they are still firing at the moment (had a reset because somebody unplugged my raspi.. ;-)

Example:

2/28/2019, 6:05:51 PM - info: MQTT publish: topic 'zigbee2mqtt/0x0017880106443a0d', payload '{"battery":100,"linkquality":44,"occupancy":false,"temperature":22.98,"illuminance":11693}'

jonnycastaway, 2019-03-01:
Mhm, i checked the devices.js 4 times and can see no error. But theres no illuminance value > 0 in my mqtt messages:
`{"battery":100,"linkquality":57,"occupancy":false,"temperature":8.4,"illuminance":0}`
It is 16:50PM so i expect a higher value. Do I have to reconnect the sensor?

jonnycastaway, 2019-03-21:
After last Hue Update for the Motion Sensors and new pairing with the Zigbee Network i got illuminance now.