Koenkk/zigbee2mqtt #1112: Setting Up SmartThings Multipurpose Sensor IM6001-MPP02

shafiq786110, 2019-02-17:
Hi 

Im still struggling with a few bits but im getting there slowly. 

Can someone please help me with adding this device: [link](https://www.smartthings.com/gb/products/smartthings-multipurpose-sensor)

The zigbeeModel is `multiv4`

It seems to output a cid `msTemperatureMeasurement` and `ssIasZone`

When the window opens, i get this: 

  ```
zigbee2mqtt:warn 2/17/2019, 3:50:42 PM No converter available for 'IM6001-MPP02' with cid 'ssIasZone', type 'statusChange' and data '{"cid":"ssIasZone","zoneStatus":33}'
  zigbee2mqtt:warn 2/17/2019, 3:50:42 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
When it closes, i get this: 
```
zigbee2mqtt:warn 2/17/2019, 3:50:42 PM No converter available for 'IM6001-MPP02' with cid 'ssIasZone', type 'statusChange' and data '{"cid":"ssIasZone","zoneStatus":32}'
zigbee2mqtt:warn 2/17/2019, 3:50:42 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

The temperature measurements come up with this despite linking it to `fz.smartthings_temperature`
```
zigbee2mqtt:warn 2/17/2019, 4:16:05 PM No converter available for 'IM6001-MPP02' with cid 'msTemperatureMeasurement', type 'devChange' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2216}}'
zigbee2mqtt:warn 2/17/2019, 4:16:05 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

My problem is i cant work out what to do in fromZigbee.js to get this working properly for the open/close status and the temperature measurement

My current input on the devices.js is: 

```
	{
		zigbeeModel: ['multiv4'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
		model: 'IM6001-MPP02', // Vendor model number, look on the device for a model number
		vendor: 'SmartThings', // Vendor of the device (only used for documentation and startup logging)
		description: 'Multi Purpose Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
		supports: 'contact, temperature', // Actions this device supports (only used for documentation)
		fromZigbee: [fz.smartthings_temperature], // We will add this later
		toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
	}, 
```

**Could anyone please help with getting this working correctly please?**

Thank You!

Koenkk, 2019-02-19:
Looks promising:

Can you try with:

```js
{
    zigbeeModel: ['multiv4'],
    model: 'IM6001-MPP02',
    vendor: 'SmartThings',
    description: 'Multi Purpose Sensor',
    supports: 'contact, temperature, battery',
    fromZigbee: [fz.generic_temperature, fz.ignore_temperature_change, fz.IM6001_MPP02_contact, fz.generic_batteryvoltage_3000_2500],
    toZigbee: [],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('genPowerCfg', 'batteryVoltage', 0, 3, cb),
            ];

            execute(device, actions, callback);
        },
}, 
```

Add to  `fromZigbee.js`:

```js
IM6001_MPP02_contact: {
        cid: 'ssIasZone',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            return {contact: msg.data.zoneStatus === 32};
        },
    },
```

This should also give us the battery voltage.

shafiq786110, 2019-02-19:
@Koenkk 

I think you're close! 

This is what comes up now: 

```
 zigbee2mqtt:warn 2/19/2019, 11:06:54 PM No converter available for 'IM6001-MPP02' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryVoltage":20}}'
  zigbee2mqtt:warn 2/19/2019, 11:06:54 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

  zigbee2mqtt:info 2/19/2019, 11:07:00 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:03 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:06 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:09 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":15,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:12 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:15 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:18 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:21 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":5,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:21 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":false,"linkquality":5,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:24 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":false,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:26 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:27 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:30 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:33 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":0,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:36 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":2,"temperature":20.82,"battery":"0.00","voltage":2000}'
  zigbee2mqtt:info 2/19/2019, 11:07:36 PM MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d749', payload '{"contact":true,"linkquality":5,"temperature":20.73,"battery":"0.00","voltage":2000}'

```

Shortened a bit but the 2 bits i found interesting were there was another cid coming up that wasnt there before and also a constant feed of temperatures. I think temp thing is normal no? But what would be needed for the genPowerCfg?

Contact is working great though! Instant change between true/false when the window opens and closes :)

Koenkk, 2019-02-20:
The constant feed is because of `(cb) => device.report('genPowerCfg', 'batteryVoltage', 0, 3, cb),`, which will make the sensor report every 3 seconds (for debugging, we will increase this later). 

What kind of battery does this device run on?

dlashua, 2019-02-20:
This may be difficult... I have a device labelled as "IM6001-MPP01" (note: yours is a MPP02). It, physically, looks identical to yours. It's "model" is reported as "multi".

I have another device labelled as "STS-MLT-250". It is an older version of the SmartThings Multi Purpose Sensor. It does not physically look like your device. It's model is reported as "multiv4", which is the same as yours. 

That being said, I'm working on a device handler for my multiv4 at the moment. I'm having a little trouble getting Cie Enrollment to work, but, once it's done I'm happy to share and hopefully it'll work for you too.

shafiq786110, 2019-02-20:
@Koenkk 

It runs on a CR2450 :)

Koenkk, 2019-02-22:
Can you try with:

```js
{
    zigbeeModel: ['multiv4'],
    model: 'IM6001-MPP02',
    vendor: 'SmartThings',
    description: 'Multi Purpose Sensor',
    supports: 'contact, temperature, battery',
    fromZigbee: [fz.generic_temperature, fz.ignore_temperature_change, fz.IM6001_MPP02_contact, fz.generic_batteryvoltage_3000_2500],
    toZigbee: [],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 3, cb),
            ];

            execute(device, actions, callback);
        },
}, 
```

ideraid, 2019-03-25:
> I have another device labelled as "STS-MLT-250". It is an older version of the SmartThings Multi Purpose Sensor. It does not physically look like your device. It's model is reported as "multiv4", which is the same as yours.

Hi, any update on getting the "STS-MLT-250" sensors working?  Thanks!

mclaudiopt, 2019-04-27:
hello...is there any solution for adding this sensor? 
thank a lot

marc-gist, 2019-04-28:
Have you tried with the latest version? It may Need someone with the sensor to update the appropriate files as per the "add support for new device" page.

mclaudiopt, 2019-04-28:
hello. thanks for the answer... i have seen on this post that there is some code...but i dont know how to add it :-(

mclaudiopt, 2019-05-15:
hello 
it is working (not for vibration sensor), but gives some errors
can please check the log?
https://paste.ubuntu.com/p/bPv3kQtD4C/

mclaudiopt, 2019-07-14:
hi.
i cannot pair the device in a way that it updates my temperature.
anything i can do?
thanks

closed 2019-09-19 by stale[bot]