Koenkk/zigbee2mqtt #1113: CC2531 crashes on large amount of commands

Edzilla2000, 2019-02-17:
I am on dev latest version and the firmware is the latest dev version: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin
If I send too many commands to zigbee2mqtt, it crashes and the usb stick is not recognized anymore.
It typically happens when I use homeassistant to turn of every lights in the house.

This is the zigbee2mqtt log from the last crash: 
https://pastebin.com/8HCiQfuc
This is the kernel log from the latest crash: 
https://pastebin.com/NF2B3Din

jonnycastaway, 2019-02-18:
I had the same problem. In my case it was the Node.js Version.
Raspbian installed Version 8.x and i get crashed all day long.
Now i'm on Version 10.15.x, the latest Stick Firmware from 15.02.2019 and the zigbee2mqtt dev branch and all is ok!

Bruceforce, 2019-02-18:
Maybe somehow related to #1027 

You could try to increase the delay from 170 to a over 1000 in https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/zigbee.js#L35

Or you group every bulb in Zigbee2mqtt (NOT in HA) and then use HA to turn the zigbee group off.

Koenkk, 2019-02-19:
@Edzilla2000 can you try again with the latest dev firmware, with the version you are running, I made a mistake which resulted in a lower memory being allocated in the CC2531.

Edzilla2000, 2019-02-19:
> 
> 
> @Edzilla2000 can you try again with the latest dev firmware (`20190218`), with the version you are running, I made a mistake which resulted in a lower memory being allocated in the CC2531.

Will I need to repair everything or can I just flash it without worrying about that?

Koenkk, 2019-02-19:
Repair is not needed anymore with this firmware.

apeinspace, 2019-02-23:
> Repair is not needed anymore with this firmware.

Hi @Koenkk, that reads like a great improvement - the first time I hear about it (maybe my bad).

So I do not have to tinfoil the stick anymore and walk away from the house while starting after flashing ? 
Since which FW-version is that the case ?

Bruceforce, 2019-02-23:
@apeinspace see #1088 and https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/what_does_and_doesnt_require_repairing.md

Edzilla2000, 2019-02-26:
I just flashed it on the stick, I'll see if I encounter the issue again.

aurelijusrozenas, 2019-03-03:
I am having similar problems: switching lights one by one is fine, changing color etc. But if try to do same for all 3 lights (defined as HA goup) - it works for couple of times at best and than all other commands are timeouting. Stick firmware: 20190223.
I have to re-plug stick and restart add-on to make it work again.
>   zigbee2mqtt:error 3/3/2019, 11:40:02 AM Zigbee publish to device '0x000b57fffebccfd5', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.

Can my issue be related to this? I can open new issue and add logs and stuff. Or maybe my debugger stick is faulty? This is my first day using it.

Koenkk, 2019-03-03:
@aurelijusrozenas are you also on the latest dev branch?

aurelijusrozenas, 2019-03-03:
@Koenkk coordinator stick - yes. hassio addon - stable.

Koenkk, 2019-03-03:
@aurelijusrozenas please try with hassio edge.

aurelijusrozenas, 2019-03-03:
All right, seems to be working :) Great!

Edzilla2000, 2019-03-12:
I haven't had that issue again

closed 2019-03-12 by Edzilla2000