Koenkk/zigbee2mqtt #1115: Problem with Xiaomi Aqara double key wired wall switch (QBKG12LM)

rado0786, 2019-02-18:
Hi,
I installed the Zigbee2mqtt and when I paired the "Xiaomi", it only detects that it is a sensor. I can not use the 2 buttons. (or I do not know how to use them)
[image]
any solution?
Thank you

Bojan023, 2019-02-18:
The wall switch is connected correctly. It actually is a sensor since it 'measures' what button is pressed. If any button is pressed (left, right, long, twice) it will broadcast a message over Zigbee accordingly. For instance: left single click will broadcast "left" and the HA sensor value will be updated to 'left'. Twice right: broadcast "right_double", HA sensor "right_double". Good luck!

dudel, 2019-02-18:
You can use the mqtt trigger and template conditions for automations like this:

```
trigger:
  - platform: mqtt
    topic: zigbee2mqtt/0x001234567890
  condition:
  - condition: template
    value_template: '{{ "left" == trigger.payload_json.click }}'
```

rado0786, 2019-02-18:
> 
> 
> You can use the mqtt trigger and template conditions for automations like this:
> 
> ```
> trigger:
>   - platform: mqtt
>     topic: zigbee2mqtt/0x001234567890
>   condition:
>   - condition: template
>     value_template: '{{ "left" == trigger.payload_json.click }}'
> ```

But where do you configure it (in configuration.yaml in .homeassistant or in zigbee2mqtt / data / configuration.yaml)
Best regards

dudel, 2019-02-18:
Have a look at the "Responding to button clicks" section in https://www.zigbee2mqtt.io/integration/home_assistant.html. You either put the whole snippet including the "automation:" in the first line in the home assistant configuration.yaml or if you have a separate automations.yaml you can put it in that file without the first line "automation:".

Also have a look at https://www.home-assistant.io/docs/automation/ to get started with automations.

Vout, 2019-02-19:
I have the "battery powered switches" and my automation looks like this,

```- alias: Switch eetkamertafel
  trigger:
  - platform: mqtt
    topic: zigbee2mqtt/Switch eetkamertafel
  condition:
      condition: or
      conditions:
        - condition: template
          value_template: '{{ ''left'' == trigger.payload_json.click }}'
        - condition: template
          value_template: '{{ ''right'' == trigger.payload_json.click }}'
  action:
  - data:
      entity_id: light.eetkamertafel
    service: light.toggle
```

Currently both switches toggle the same light (still need to configure that). Besides an normal click you probably also have an double and long click available (i have them on the double version, single versions seems to have single click only).

> But where do you configure it (in configuration.yaml in .homeassistant or in zigbee2mqtt / data / configuration.yaml)

Under home assistant configuration > automations. You could do it also in the editor itself inside hassio (http://your-hassio-url/config/automation/new).

b4dpxl, 2019-02-21:
Don’t the wired switches know whether they’re on or off? I know the battery ones show as a sensor, but I’d expect the wired ones to show as a light. HA for example shouldn’t need to trigger an automation to turn the light on/off when the button is pressed, as it’s physically wired to the bulbs. I use the automation approach above with battery Xiaomi switches (WXKG02LM) to control ikea bulbs, but the wired switches not showing as a light puts me off them. If they’re showing as a sensor, can you even turn them on/off via HA?

b4dpxl, 2019-03-15:
@rado0786 FYI, in case you haven't fixed this, I had the same issue until I updated z2m. Looks like an out of date zigbee-shepherd-converter.

closed 2019-05-14 by Koenkk