Koenkk/zigbee2mqtt #1117: Pairing issues

marc-gist, 2019-02-18:
Using Coordinator firmware version: '20190109'
Pairing stopped working.  I have 7 devices paired for the past 24hrs or so.  This morning went to add more this morning... nothing.
I restarted the zigbee2mqtt... nothing

I stopped server... removed USB receiver, wait 10 sec, plugged back in, started server.
Pairing works again!

Things I noted on startup after doing this.  Got the `Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)` message, which it does recover from.

Is this because I have left pairing mode "true" in configuration?

khzd, 2019-02-18:
Hi
I got also this error this Sunday
The only solution that work for me was to reboot all the system :(
Did you try this ?

marc-gist, 2019-02-18:
@khzd didn't have to reboot. Just shutdown zigbee2mqtt, remove usb... Insert USB. Start zigbee2mqtt again.

Koenkk, 2019-02-19:
Does it work again when toggling the permit_join off/on via MQTT? http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigpermit_join

marc-gist, 2019-02-19:
@Koenkk no... tried a few times, even after a couple restarts of zigbee2mqtt.  I had to physically remove the USB controller, and put it back in, to get pairing to work again.  Nothing showing in logs so didn't save them. (set to debug).
I have pairing to false in config now and only turn on when needed. I'll post again if i see the same happen in this configuration.  I'm assuming its because I left pairing enabled for over a day (which who is really going to do that regularly anyway :)

marc-gist, 2019-02-23:
Happened again.. configuration defaults to allow join false, went to add new devices today, enable join via mqtt, nothing!
restarted zigbee2mqtt, nothing... logs in 'debug' mode, nothing shows up.
Shutdown zigbee2mqtt, removed usb, waited 10 sec, inserted again, start zigbee2mqtt.
enable pair, successful pair a couple of devices!

alinelena, 2019-02-24:
I had similar experinces with devices which were paired in the past. Got few new devices, xioami switches/sensors fo motion and ikea bulbs they paired without issues. With older devices what seems to help is to force remove them via mqtt. unfortunately this may be just a red herring i have no proof.

Bastiencc, 2019-02-26:
Same issue there. Your trick works great :

1. stop server
2. removed USB receiver
3. wait 10 sec
4. plugg back in
5. started server.
6. Pairing works again

closed 2019-05-04 by stale[bot]