Koenkk/zigbee2mqtt #1118: Adding Yale YMF40 Lock to Hass.io

matisaul, 2019-02-18:
Hi,

Continue with topic #1032 , I want to add my Yale YMF40 Lock to Hass.io,
I tried with previous solution of a different Yale lock model, but I just can open/close lock using "set" call, and read . I can't get lock status when change, and battery level.

I get to pair to coordinator, but I get this error:

``` 2019-2-14 20:29:28 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
2019-2-14 20:29:28 - debug: Configuring 0x000d6f00118c1895 (0x000d6f00118c1895) ...
2019-2-14 20:29:28 - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'iZBModule01' (0x000d6f00118c1895)
2019-2-14 20:29:38 - warn: Failed to configure 0x000d6f00118c1895 (0x000d6f00118c1895) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #2)
2019-2-14 20:29:38 - warn: This can be ignored if the device is working properly
```

My actual config:

**Device.js:**

```
    {
        zigbeeModel: ['iZBModule01'],
        model: 'YMF40',
        vendor: 'Yale',
        description: 'Real Living Lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.report('closuresDoorLock', 'lockState', 0, 3, 0, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 3, 0, cb),
            ];

            execute(device, actions, callback);
        },

    },

```
**FromZigbee:**
```
{
  YMF40_lockstatus: {
      cid: 'closuresDoorLock',
      type: 'devChange',
      convert: (model, msg, publish, options) => {
		      return {state: msg.data.data.lockState === 1 ? 'Lock' : 'Unlock'};
      },
  },
  battery_200: {
      cid: 'genPowerCfg',
      type: 'devChange',
      convert: (model, msg, publish, options) => {
		  console.log(msg.data);
          const batt = msg.data.data.batteryPercentageRemaining;
          const battLow = msg.data.data.batteryAlarmState;
          const results = {};
          if (batt != null) {
              const value = Math.round(batt/200.0*10000)/100; // Out of 200
              results['battery'] = value;
          }
          if (battLow != null) {
              if (battLow) {
                  results['battery_low'] = true;
              } else {
                  results['battery_low'] = false;
              }
          }
          return results;
      },
  },

```

**toZigbee:**
```
{
      key: ['state'],
      convert: (key, value, message, type, postfix) => {
          const cid = 'closuresDoorLock';
          const attrId = 'lockState';

          if (type === 'set') {
              if (typeof value !== 'string') {
                  return;
              }

              return {
                  cid: cid,
                  cmd: `${value.toLowerCase()}Door`,
                  cmdType: 'functional',
                  zclData: {
                      'pincodevalue': '',
                  },
                  cfg: cfg.default,
                  readAfterWriteTime: 200,
              };
          } else if (type === 'get') {
              return {
                  cid: cid,
                  cmd: 'read',
                  cmdType: 'foundation',
                  zclData: [{attrId: zclId.attr(cid, attrId).value}],
                  cfg: cfg.default,
              };
          }
      },
  },

```

Here is part of my log:

https://hastebin.com/mafedunepo.js

Thanks!!

Koenkk, 2019-02-19:
Could you post the `data/database.db` entries of this device?

matisaul, 2019-02-19:
Sure:
```
{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f00118c1895","nwkAddr":58062,"manufId":4491,"manufName":"ASSA ABLOY iRevo","powerSource":"Battery","modelId":"iZBModule01","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,10,32,257],"outClusterList":[10,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{}},"genTime":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"closuresDoorLock":{"dir":{"value":1},"attrs":{}}}}},"_id":"aWGDHbkdzEWPmXQ6"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e32c7c","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1550496812,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[0,1281,3],"outClusterList":[1280,1282,3],"clusters":{"genBasic":{"dir":{"value":1},"acls":{"zclVersion":"R","hwVersion":"R","manufacturerName":"R","modelId":"R","dateCode":"R","powerSource":"R","locationDesc":"RW","physicalEnv":"RW","deviceEnabled":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"g3LpXo3sHkAyFGtN"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e32c7c","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1550498542,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"g3LpXo3sHkAyFGtN"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f00118c1895","nwkAddr":58062,"manufId":4491,"manufName":"ASSA ABLOY iRevo","powerSource":"Battery","modelId":"iZBModule01","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,10,32,257],"outClusterList":[10,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{}},"genTime":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"closuresDoorLock":{"dir":{"value":1},"attrs":{}}}}},"_id":"aWGDHbkdzEWPmXQ6"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e32c7c","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1550498542,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[0,1281,3],"outClusterList":[1280,1282,3],"clusters":{"genBasic":{"dir":{"value":1},"acls":{"zclVersion":"R","hwVersion":"R","manufacturerName":"R","modelId":"R","dateCode":"R","powerSource":"R","locationDesc":"RW","physicalEnv":"RW","deviceEnabled":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"g3LpXo3sHkAyFGtN"}

```

Koenkk, 2019-02-20:
So just to be sure:
- You are able to lock/unlock the lock
- The only thing that isn't working are the status updates?

If so, please try with
```js
{
        zigbeeModel: ['iZBModule01'],
        model: 'YMF40',
        vendor: 'Yale',
        description: 'Real Living Lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.report('closuresDoorLock', 'lockState', 0, 3, 0, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 3, 0, cb),
            ];

            setTimeout(() => {
                execute(device, actions, callback);
            }, 3000)
        },

    },
```

matisaul, 2019-02-20:
>* You are able to lock/unlock the lock
>* The only thing that isn't working are the status updates?

Yes, using "zigbee2mqtt/0x000d6f00118c1895/set" I can lock/unlock
and using "zigbee2mqtt/0x000d6f00118c1895/get" I receive status in that moment.

I use your code, but I still receiving the same error.

https://hastebin.com/acuponovuy.sql

I get a lot of this messages:

```
2019-02-20T19:36:34.877Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]

Is this maybe the status?

Koenkk, 2019-02-20:
Could you do some zigbee sniffing to see if something is send when manually (physically) unlocking/locking the lock? http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Because the configure fails, the device doesn't seem to report it's status (I don't know why this fails yet).

matisaul, 2019-02-20:
Ok, let me try with that.

matisaul, 2019-03-31:
Hi @Koenkk ! I finally receive a second CC2531 stick for sniffing.

When I open and close the door, I receive this message:

```
Frame 1066: 58 bytes on wire (464 bits), 56 bytes captured (448 bits) on interface 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x81b2
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x81b2
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
    Destination: 0x0000
    Source: 0x81b2
    Radius: 30
    Sequence Number: 107
    [Extended Source: Ember_00:11:8c:18:95 (00:0d:6f:00:11:8c:18:95)]
    [Origin: 818]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 1
    Cluster: Door Lock (0x0101)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 204
ZigBee Cluster Library Frame

```

Do you need more information? 

Thanks

matisaul, 2019-04-01:
I receive new data on Hassio add-on logs:

```
2019-03-31T23:48:41.752Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":257,"srcaddr":33202,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":63,"securityuse":0,"timestamp":5081100,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[25,89,32,2,2,0,0,0,150,21,87,0,0]}}
2019-03-31T23:48:41.756Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-03-31T23:48:41.756Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":257,"srcaddr":33202,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":63,"securityuse":0,"timestamp":5081100,"transseqnumber":0,"len":13,"data":{"0":25,"1":89,"2":32,"3":2,"4":2,"5":0,"6":0,"7":0,"8":150,"9":21,"10":87,"11":0,"12":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":89,"cmdId":"operationEventNotification","payload":{"opereventsrc":2,"opereventcode":2,"userid":0,"pin":0,"zigbeelocaltime":5707158,"data":0}}}
```

Koenkk, 2019-04-04:
Could you add to https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L691 `operationEventNotification`? 

It's perhaps easier to setup a bare metal zigbee2mqtt installation (http://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html), file can then be found under `/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js`.

he902, 2019-04-06:
> Could you add to https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L691 `operationEventNotification`?
> 
Koenkk, this does helped me. Thanks!
I'm trying to add my Yunmi Lock, and met the same issue.
After fixed the af.js followed your guide, it can decode the unlock door message correctly.
Now, my af.js has:
`            const cmdIDs = ['on', 'offWithEffect', 'step', 'stop', 'hueNotification', //Philips-RWL020-RWL021
                            'off', 'stepColorTemp', 'moveWithOnOff', 'move', 'moveHue', 'moveToSaturation', //OSRAM-Switch 4x EU-LIGHTIFY
                            'stopWithOnOff', 'moveToLevelWithOnOff', 'toggle', 'tradfriArrowSingle', 'tradfriArrowHold', 'tradfriArrowRelease',
                            'stepWithOnOff', 'moveToColorTemp', 'moveToColor', 'onWithTimedOff', 'recall', 'operationEventNotification'];
`

matisaul, 2019-04-07:
> Could you add to https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L691 `operationEventNotification`?
>
> It's perhaps easier to setup a bare metal zigbee2mqtt installation [...]

Yeah!! That did the trick...

**opereventcode** is the state of the lock. Number 2 is Unlock, and 10 is Lock.
**opereventsrc** is the opening source. "0" is open from Keypad, "2" open from inside, and "4" is from fingerprint. 
**userid** is the id of registered user who opens the door.

Finally, This is my config:

**devices.js:**

```
    {
        zigbeeModel: ['iZBModule01'],
        model: 'YMF40',
        vendor: 'Yale',
        description: 'Real Living Lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.YMF40_lockstatus],
        toZigbee: [tz.YMF40_lockcmd],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.report('closuresDoorLock', 'lockState', 0, 3, 0, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 3, 0, cb),
            ];

            execute(device, actions, callback);
        },

    },

```

**fromZigbee.js:**

```
    YMF40_lockstatus: {
      cid: 'closuresDoorLock',
      type: 'cmdOperationEventNotification',
      convert: (model, msg, publish, options) => {
		  const stateCode = msg.data.data['opereventcode'];
		  const user = msg.data.data['userid'];
		  const sourceCode = msg.data.data['opereventsrc'];
		  const source = sourceCode === 0 ? 'Keypad' : 2 ? 'Inside' : 4 ? 'Fingerprint' : 'Other';
		  const state = stateCode === 2 ? 'Unlock' : 'Lock'; 
		      return {state: state, user: user, source: source };
      },
    },

```

toZigbee.js:

```
      YMF40_lockcmd: {
      key: ['state'],
      convert: (key, value, message, type, postfix) => {
          const cid = 'closuresDoorLock';
          const attrId = 'lockState';

          if (type === 'set') {
              if (typeof value !== 'string') {
                  return;
              }

              return {
                  cid: cid,
                  cmd: `${value.toLowerCase()}Door`,
                  cmdType: 'functional',
                  zclData: {
                      'pincodevalue': '',
                  },
                  cfg: cfg.default,
                  readAfterWriteTime: 200,
              };
          } else if (type === 'get') {
              return {
                  cid: cid,
                  cmd: 'read',
                  cmdType: 'foundation',
                  zclData: [{attrId: zclId.attr(cid, attrId).value}],
                  cfg: cfg.default,
              };
          }
      },
  },

```

Tell me if you need more information to add this lock to the official devices.

Thanks for everything @Koenkk !

he902, 2019-04-08:
Any easy way to do the mapping from opereventsrc and userid into  a person who unlocked the door?
e.g.:
(opereventsrc, userid)   --> Person
(2, 0) --> "Me"
(2, 1) --> "Wife"
(3, 1) --> "Mother"
(3, 2) --> "Me"
(4, 0) --> "Mother"
...
Do it in zigbee2mqtt or Home Assistant?
Better to have a config file for such mapping.

he902, 2019-04-08:
@matisaul , I saw:
`                  zclData: {
                      'pincodevalue': '',
                  },`
Does that mean your YMF40 has no need of PIN to unlock via zigbee?

tb-killa, 2019-04-08:
> Could you add to https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L691 `operationEventNotification`?
>
> It's perhaps easier to setup a bare metal zigbee2mqtt installation [...]

@Koenkk 
To be prepared for the future and especially to help the end user in case of support we would need something like a function which checks if the cmdId Value is present in the cmdIDs list.

You could iterate over each element in the array list of cmdIDs and look for the entry with a matching value.
If this is not the case, output a hint with a logger.error such as
``"Attention: Unknown cmdID: ${Value}, this may be needed in the list "cmdIDs" to ensure functionality."``

Here I need your advice and implementation :)

msaulAyT, 2019-04-08:
> @matisaul , I saw:
> ` zclData: { 'pincodevalue': '', },`
> Does that mean your YMF40 has no need of PIN to unlock via zigbee?

Yes, I do not need a pin to unlock it, just I have to be paired with the lock.

matisaul, 2019-04-08:
> @matisaul , I saw:
> ` zclData: { 'pincodevalue': '', },`
> Does that mean your YMF40 has no need of PIN to unlock via zigbee?

Yes, I do not need a pin to unlock it, just I have to be paired with the lock.

Koenkk, 2019-04-09:
@matisaul great, can you make a PR (https://github.com/koenkk/zigbee-shepherd-converters)?

@tb-killa I've added the extra logging.

matisaul, 2019-04-09:
@Koenkk PR added !

BrunoZumba, 2019-10-26:
Hey guys. 
I'm not sure if I should post this here or open a new thread. Since I think this is just a doubt, not an issue, I'll post it here. If that's not correct, let me know and I'll open a new thread.

I've been strugling with YMF40 lock for almost a week now. Thanks to this thread and #1032 I've been able to pair, lock, unlock and get status. 

But I'm facing a problem with YMF40 that it doesnt send any zigbee messages when I close it from inside.

When I open it, I get everything correct: 
opereventcode = 2 for unlock.
opereventsrc 0 for keypad unlock, 2 from inside and 4 from fingerprint.
I even get everything correct for autolock:
opereventcode = 10 for autolock

But I get nothing when I lock it from the knob on the inside.

Is that correct? 
@matisaul (sorry to @ you directly) but I know you have this exact same lock. Does it send a lock message when locking from inside?

BrunoZumba, 2019-10-26:
Here are my config files
(for some reason my zigbeeModel is 0700000001 instead of iZBModule01)
```
fromZigbee.js
generic_lock_operation_event: {
    cid: 'closuresDoorLock',
    type: 'cmdOperationEventNotification',
    convert: (model, msg, publish, options) => {
        return {
            state: msg.data.data['opereventcode'] == 2 ? 'UNLOCK' : 'LOCK',
            user: msg.data.data['userid'],
            source: msg.data.data['opereventsrc'],
        };
    },
},

    battery_200: {
        cid: 'genPowerCfg',
        type: ['attReport', 'readRsp'],
        convert: (model, msg, publish, options) => {
            const batt = msg.data.data.batteryPercentageRemaining;
            const battLow = msg.data.data.batteryAlarmState;
            const voltage = msg.data.data.batteryVoltage;
            const results = {};
            if (batt != null) {
                const value = Math.round(batt/200.0*10000)/100; // Out of 200
                results['battery'] = value;
            }
            if (battLow != null) {
                if (battLow) {
                    results['battery_low'] = true;
                } else {
                    results['battery_low'] = false;
                }
            }
            if (voltage != null) {
                results['voltage'] = voltage * 100;
            }
            return results;
        },
    },

toZigbee.js
generic_lock: {
        key: ['state'],
        convert: (key, value, message, type, postfix, options) => {
            const cid = 'closuresDoorLock';
            const attrId = 'lockState';

            if (type === 'set') {
                if (typeof value !== 'string') {
                    return;
                }

                return [{
                    cid: cid,
                    cmd: `${value.toLowerCase()}Door`,
                    cmdType: 'functional',
                    zclData: {
                        'pincodevalue': '',
                    },
                    cfg: cfg.default,
                    readAfterWriteTime: 200,
                }];
            } else if (type === 'get') {
                const teste = [{attrId: Zcl.getAttributeLegacy(cid, attrId).value}];
                return [{
                    cid: cid,
                    cmd: 'read',
                    cmdType: 'foundation',
                    zclData: [{attrId: Zcl.getAttributeLegacy(cid, attrId).value}],
                    cfg: cfg.default,
                }];
            }
        },
    },

devices.js
    {
 //       zigbeeModel: ['iZBModule01'],
        zigbeeModel: ['0700000001'],
        model: 'YMF40',
        vendor: 'Yale',
        description: 'Real living lock',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.generic_lock_operation_event, fz.battery_200],
        toZigbee: [tz.generic_lock],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.report('closuresDoorLock', 'lockState', 0, 60, 0, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 30, 0, cb),
            ];

            execute(device, actions, callback);
        },

    },

```

Koenkk, 2019-10-26:
I may have fix that issue in the latest zigbee2mqtt dev branch, can you check?

BrunoZumba, 2019-10-27:
Yep, that's correct!!! for dev branch it's working like a charm. Thanks!! :)

Since you're here just one more question: I have zwave lock (schale) and with zwave I can perform a lot more actions: include a new user code to unlock it, delete a user code, change some attributes (like alarm, volume level and stuff).

You know zigbee devices infinetly more than me: 
Is it possible to do the same with zigbee? (I know it will depend on the model I have). If so, can you point me to the direction where I can find more information about this?

Or is it just impossible doing this with Zigbee? (like, "structurally" zigbee doesn't support these functions, so it's not possible for now and that's it).

Thanks again!

Koenkk, 2019-10-27:
Zigbee also probably supports these, please take a look at the zigbee cluster library specification; it should be in the door lock cluster.

closed 2019-10-27 by Koenkk