Koenkk/zigbee2mqtt #1119: Question: IKEA TRADFRI Control Outlet vs Signal Repeater

tynor88, 2019-02-18:
Why would one buy the Signal Repeater over the Control Outlet ? Both will function as an ZigBee Router, but at the same price the Control Outlet seems like a more valuable option, as it offers a button as well.

Did i miss anything?

SBTephan, 2019-02-19:
Maybe just because you don't need an controlled outlet there? Also the repeater is an USB charger for your mobile phone,

JLFN, 2019-02-19:
@SBTephan Is the  Signal the same? Do you own one?

SBTephan, 2019-02-19:
I own one, it has extended the range to my garage successfully. Don't know if the signal strength is the same as the outlet. I do not have the measuring tools for that.

JLFN, 2019-02-19:
@SBTephan You have linkquality in mqtt :)

SBTephan, 2019-02-19:
@JLFN I know but that bounces up and down, couldn't get a good measurement, but I have tried :-)

tynor88, 2019-02-19:
Did anyone try the Control Outlet as a range extender for the Zigbee Network?

JohGub, 2019-02-19:
In Germany you can't buy the outlet. Therefore I have bought a signal repeater. Works fine in zigbee2mqtt, pairing was no problem, but I don't know how to configure to get status and link quality. Any ideas ?

SBTephan, 2019-02-19:
@JohGub 
at /opt/zigbee2mqtt/data/log/%startdatetime%/log.txt you will find all the info about signal quality etc.
just set a tail -f on the log.txt.

JohGub, 2019-02-19:
@SBTephan: The IKEA repeater only turns up once in the log file, at starttime. On the device map it is listed. Is there a way to make him produce feedback like the CC2530 router (ping and payload) ?  I've tried to add something to devices.js, no success.

JLFN, 2019-02-19:
did you guys try the  graph?

nikkolade, 2019-02-19:
Has anyone checked if the for example xiaomi end devices are happy with the signal repeater ? It looks to me that none of the xiaomi devices that I have, are willing to use the control outlet as a router. The ikea light bulbs are working fine as routers, but control outlet is being ignored.

SBTephan, 2019-02-19:
@JLFN what graph?

@nikkolade , I know it uses the lightbulbs, will try to see if it will use the signal repeater

JohGub, 2019-02-19:
In my net 2 Xiaomi temp sensors and 1 switch are hooked on the signal repeater. and give feedback. Still wonder if there is a way checking repeaters state.

pcb1962, 2019-02-19:
@nikkolade not sure what the difference is between the Xiaomi sensors and the Aqara ones, some seem to be listed as both. I have had success today with the Aqara door sensor which previously could not reach my coordinator and is now communicating via a Tradfri Outlet.
I picked up 3 outlets yesterday and have distributed them around the house. It looks like they are meshed together nicely on the networkmap, but I was hoping to see some of my other sensors move over to the new routers, which should improve the link quality. They don't seem to be doing that - apart from FrontDoor which has no choice as it can't reach the coordinator directly.
[image]

SBTephan, 2019-02-19:
@pcb1962 how do you create a map like that?

pcb1962, 2019-02-20:
@SBTephan most of what you need is here: https://community.openhab.org/t/zigbee2mqtt-network-map/57784 and more info right here if you search 'networkmap'

SBTephan, 2019-02-20:
@pcb1962 thanks, will check it out

nikkolade, 2019-02-20:
Ok, it seems that after several days of playing around with all these devices, a few of xiaomi aqara devices decided to use the ikea outlet as a router. Nice.

tynor88, 2019-02-20:
I bought the IKEA Control Outlet + On / Off button. Works great with as repeater, and I can pair the On/Off button directly with the CC2531 with the latest dev firmware, and use it independently of the Control Outlet.

Great stuff! 👍

djchen, 2019-02-22:
+1 on the Control Outlet

If you don't need the phone charger, the control outlet is a great choice for extending the networking range.

closed 2019-02-28 by tynor88

clausjensen, 2019-05-05:
Did any of you get any results as to whether there's a difference in regards to how much range extension is achieved with the repeater vs. using an outlet?