Koenkk/zigbee2mqtt #1120: ZCTS-808 statusChange results in "Ignoring message from coordinator"

srdansuka, 2019-02-18:
zigbee2mqtt version 1.1.1 (commit #92d88b6)
Coordinator firmware version: '20190109'

We have an issue with ZCTS-808 - Trust Wireless contact sensor.
It's successfully paired and sends `statusChange` messages but  it seems messages are rejected due to this issue: `Ignoring message from coordinator`

Detailed debug log at https://pastebin.com/XyzwdS7R

0x00158d0001cc89b3 is ZCTS-808 while 0x00124b0002c5a5ab is our coordinator address. 

The result is `Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":1}' of device 'undefined' (0x00124b0002c5a5ab)` ?!

On the other hand we are using `ZPIR-8000 - Trust Motion Sensor` and it works without any issues.

Any hints? 

Thanks

Koenkk, 2019-02-19:
Can you check if it works with the latest dev firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin (doesn't require repairing of devices).

Kryzek, 2019-02-20:
> Can you check if it works with the latest dev firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin (doesn't require repairing of devices).

Sorry to hijack the thread like this, but one can flash this new firmware with old flashing instructions without needing to drive to park to flash the stick? Or is there another method to do this?

Kryzek, 2019-02-20:
I'm not sure if this has any effect, but initially when I tried to pair these same things I couldn't get them work properly at first without doing this. Seemingly they paired but I needed to keep "opening and closing" (pulling and inserting the sensor parts next to eachother) the device for 10 seconds or so to get them configured. Opening/closing keeps sensors awake during pairing process and configuration went through like this.

Koenkk, 2019-02-20:
@Kryzek https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/what_does_and_doesnt_require_repairing.md

srdansuka, 2019-02-20:
> I'm not sure if this has any effect, but initially when I tried to pair these same things I couldn't get them work properly at first without doing this. Seemingly they paired but I needed to keep [...]

Thanks @Kryzek !!!!
This worked.

@Koenkk This would be great hint to have in docs for other users that run into same issue.

closed 2019-02-20 by srdansuka

Koenkk, 2019-02-20:
@srdansuka or @Kryzek could you make a PR to update the documentation here: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/getting_started/pairing_devices.md ?

srdansuka, 2019-02-20:
> 
> 
> @srdansuka or @Kryzek could you make a PR to update the documentation here: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/getting_started/pairing_devices.md ?

Sure I can! However, credits go to @Kryzek !