Koenkk/zigbee2mqtt #1121: - null failed with error Error: rsp error: 194

alinelena, 2019-02-18:
```

2/18/2019, 7:59:50 PM - info: Logging to directory: '/home/homeassistant/zigbee2mqtt/data/log/2019-02-18.19-59-50'
2/18/2019, 7:59:50 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/home/homeassistant/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":false}}'
2/18/2019, 7:59:50 PM - debug: Loaded state from file /home/homeassistant/zigbee2mqtt/data/state.json
2/18/2019, 7:59:50 PM - info: Starting zigbee2mqtt version 1.1.1 (commit #a20ec7c)
2/18/2019, 7:59:50 PM - info: Starting zigbee-shepherd
2/18/2019, 7:59:51 PM - info: zigbee-shepherd started
2/18/2019, 7:59:51 PM - info: Coordinator firmware version: '20190109'
2/18/2019, 7:59:51 PM - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a63","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ed19ff","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190109},"startTime":1550519991,"joinTimeLeft":0}
2/18/2019, 7:59:51 PM - info: Currently 32 devices are joined:
```
after 1h i got this
2/18/2019, 8:59:05 PM - error: Zigbee publish to group '3', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: rsp error: 194

this is freshly flashed after the previous flashing refused to boot (probbaly unreleated)

Koenkk, 2019-02-19:
Error 194 = ZNWK_INVALID_REQUEST 

- How many times do you get this? For al request or only for one?
- What hardware are you using?

alinelena, 2019-02-21:
@Koenkk was for all requests. the hardware is cc2531. 

reflased and all seems ok now
or at least run ok since 18

```
2/18/2019, 10:07:47 PM - info: Logging to directory: '/home/homeassistant/zigbee2mqtt/data/log/2019-02-18.22-07-47'
2/18/2019, 10:07:47 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/home/homeassistant/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":false}}'
2/18/2019, 10:07:47 PM - debug: Loaded state from file /home/homeassistant/zigbee2mqtt/data/state.json
2/18/2019, 10:07:47 PM - info: Starting zigbee2mqtt version 1.1.1 (commit #a20ec7c)
2/18/2019, 10:07:47 PM - info: Starting zigbee-shepherd
2/18/2019, 10:07:48 PM - info: zigbee-shepherd started
2/18/2019, 10:07:48 PM - info: Coordinator firmware version: '20190109'
2/18/2019, 10:07:48 PM - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a63","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ed19ff","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190109},"startTime":1550527668,"joinTimeLeft":0}
2/18/2019, 10:07:48 PM - info: Currently 32 devices are joined:

```

closed 2019-04-29 by stale[bot]