Koenkk/zigbee2mqtt #1123: Error reporting in MQTT message

pepp86, 2019-02-19:
Hi,

is it possible to report error from devices within the MQTT message? For example if the server responds with "Mac no ack" while trying to switch a device?

If so, my home automation can trigger on that and report the issue to me or try to switch on/off again.

andreasbrett, 2019-02-19:
Great idea! Would be very helpful. Not sure if it's better to have this on a per-device basis or globally. For my use-case it would be better to have an error message in the log subtree rather than having one additional error attribute for each device (as I would have to configure my home automation system to watch for ALL of them).

Koenkk, 2019-02-20:
Implemented, example of the message published to MQTT: https://hastebin.com/lakubupume.apache

andreasbrett, 2019-02-20:
Looks good, will use that in my smart home setup. Thanks for this rapid implementation!

pepp86, 2019-02-21:
Great! Awesome fast Implementation. 

Now I only need to get the dev tree up and running on my environment :-)

closed 2019-02-21 by pepp86