Koenkk/zigbee2mqtt #1126: how make the ZNP security disable

msadeghz, 2019-02-19:
Hi friends,
I want to add a device to the zigbee2mqtt. the device security capability is on "no security" but the ZNP sends the security key. how can I set the ZNP of zigbee2mqtt to disable security.

[log_ir transmitter.zip](https://github.com/Koenkk/zigbee2mqtt/files/2880337/log_ir.transmitter.zip)

msadeghz, 2019-02-20:
today, I've changed bdb configuration in the ZNP source code then change back to the default value. now, the issue is solved and I dont know why the ZNP dont send correct frame. you can see the "transport key" frame sent from ZNP to the device

old (when I faced the issue)
[image]

as you see, the source IEEE address is 00:00:00:00:00:00:00:00 which is incorrect and the aps frame hasn't auxheader.

new (the issue is solved)
[image]

as you see. the source IEEE address is correct and it has auxheader.

I dont know why the ZNP doesn't work stable in different times !!

Koenkk, 2019-02-20:
To disable security, you should compile the firmware with https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/firmware.patch#L100 `#define SECURE 0`.

Good to see that the problem is solved without changing this.

closed 2019-02-20 by Koenkk

msadeghz, 2019-02-21:
@Koenkk but why isn't it stable? I just change a definition in z-stack source code and change back to default value, and the pairing issue is solves?

Koenkk, 2019-02-21:
I'm not sure, perhaps somebody at the TI forum knows this.

msadeghz, 2019-06-12:
@Koenkk is there any other way to disable security? I want to not transport key to end device. because when I send the key it will reset.

Koenkk, 2019-06-12:
@msadeghz from what I know not, but you should check this on the TI forum (I'm no Z-stack expert).

msadeghz, 2019-06-12:
[image]
why source address is 00 in transport key message? can't it be resolved?

msadeghz, 2019-06-12:
@Koenkk I've flashed the dongle with [the newest version firmware you proposed](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/CC2531_20190425.zip) and the issue is resolved. thank you.