Koenkk/zigbee2mqtt #1127: Supporting new devices / Docker

ReneHonig, 2019-02-19:
Dear Koen, community
Thanks for this great development!

I am running zigbee2mqtt in Docker and find that configuring/testing new devices is quite cumbersome. I have read that you made it easier for Hass.io users ([https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html](url)) and wanted to ask if it would be possible to scan for custom device and converter files in the config directory. Ideally these should only contain new devices/converters which would be added to the full device configuration files (so you will get the latest updates when you pull a new zigbee2mqtt release.

Thanks for your consideration.

Regards, Rene

j4c3, 2019-02-19:
I just came here to check on the same thing!

snjoetw, 2019-02-20:
Hope this will help you ... I'm running z2m with docker compose like this:

```
version: "2.1"

services:
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt:latest
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /zigbee2mqtt:/app/data
      - /zigbee2mqtt/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters
      - /zigbee2mqtt/lib/extension/homeassistant.js:/app/node_modules/lib/extension/homeassistant.js
    devices:
      - /dev/ttyACM1:/dev/ttyACM0:rwm
```

Then I just git clone zigbee-shepherd-converters in "/zigbee2mqtt" and make the modification there to support new device

The last volume (/zigbee2mqtt/lib/extension/homeassistant.js) is so that I can integrate new device to HomeAssistant. I basically just copy that from z2m repo and put it in that path

Koenkk, 2019-02-20:
@snjoetw this is the best way to do this, would you mind updating the documentation with this?https://github.com/Koenkk/zigbee2mqtt.io/blob/master/information/docker.md

j4c3, 2019-02-20:
It does seem like a reasonable way to handle it.  You still have to be careful cloning the zigbee-sheperd-converters, so as to not overwrite your modifications. 

Is the concern with looking for / including an additional file that it will cause conflicts?

Koenkk, 2019-02-20:
You still have to be careful, but I think this is the easiest possible solution, it should only be used for developing/prototyping anyway.

j4c3, 2019-02-20:
Pulling the converters out of docker does make sense, as they'd likely be updated much more frequently.

Thanks!

ReneHonig, 2019-02-20:
thanks all, good way forward!

snjoetw, 2019-02-20:
@Koenkk here you go: https://github.com/Koenkk/zigbee2mqtt.io/pull/15

closed 2019-02-23 by ReneHonig