Koenkk/zigbee2mqtt #1130: Unable to install: serialport problem

SimplyGardner, 2019-02-20:
**System:**
- Rasp Pi 3b+
- Running DietPi - running Home Assistant in a virtual environment

I am following the instructions for installing zigbee2mqtt in a virtual environment and everything runs perfectly, up until the ``npm install`` script. Seems like there is a problem with the serialport@6.2.2 install script?

**Here's a copy of where I get up too (and the breaking point)**
```
(zigbee2mqtt) dietpi@DietPi:/opt/zigbee2mqtt$ npm install

> serialport@6.2.2 install /opt/zigbee2mqtt/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.15.1 runtime=node arch=arm platform=linux)
gyp ERR! configure error
gyp ERR! stack Error: Command failed: /opt/zigbee2mqtt/bin/python -c import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack   File "<string>", line 1
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                                ^
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:294:12)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:970:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
gyp ERR! System Linux 4.14.98-v7+
gyp ERR! command "/opt/zigbee2mqtt/bin/node" "/opt/zigbee2mqtt/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/serialport
gyp ERR! node -v v10.15.1
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/dietpi/.npm/_logs/2019-02-20T15_19_34_105Z-debug.log
```

### Log

Here are the last few lines from the log mentioned above, although it seems like a repeat.
```
3675 verbose cwd /opt/zigbee2mqtt
3676 verbose Linux 4.14.98-v7+
3677 verbose argv "/opt/zigbee2mqtt/bin/node" "/opt/zigbee2mqtt/bin/npm" "install"
3678 verbose node v10.15.1
3679 verbose npm  v6.4.1
3680 error code ELIFECYCLE
3681 error errno 1
3682 error serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
3682 error Exit status 1
3683 error Failed at the serialport@6.2.2 install script.
3683 error This is probably not a problem with npm. There is likely additional logging output above.
3684 verbose exit [ 1, true ]
```

andreasbrett, 2019-02-20:
Just a quick guess as I couldn't thoroughly check your log file right now (using smartphone): is your nodejs package too new maybe? I remember having some issue like yours and could solve that by downgrading nodejs to I guess 6.x

andreasbrett, 2019-02-20:
Documentation says...

# Setup Node.js repository
sudo curl -sL https://deb.nodesource.com/setup_10.x | sudo -E bash -

# Install Node.js
sudo apt-get install -y nodejs git make g++ gcc

# Verify that the correct nodejs and npm (automatically installed with nodejs)
# version has been installed
node --version  # Should output v10.X
npm --version  # Should output 6.X

SimplyGardner, 2019-02-20:
Thanks @andreasbrett .

I followed the instructions initially and assumed the Install Node.js part ``sudo apt-get install -y nodejs git make g++ gcc`` is installing the version best suited the the setup?

I can go back and try and make it lower. What version would you recommend and what command would I use to download the lower version?

Could I also check that to install in a virtual environment... I **first** follow the instructions from the docs: 

```
# Setup Node.js repository
sudo curl -sL https://deb.nodesource.com/setup_10.x | sudo -E bash -

# Install Node.js
sudo apt-get install -y nodejs git make g++ gcc

# Verify that the correct nodejs and npm (automatically installed with nodejs)
# version has been installed
node --version  # Should output v10.X
npm --version  # Should output 6.X
```

**Then**... go ahead and follow the virtual environment instructions: https://www.zigbee2mqtt.io/information/virtual_environment.html

andreasbrett, 2019-02-20:
Oh okay, not sure how things work for virtual environments. Better wait for someone with better knowledge before messing something up with my advice that might not be supposed for your setup.

What you can do however is checking the versions of nodejs and npm. Just to be sure that those are fine (10.x and 6.x).

SimplyGardner, 2019-02-20:
Appreciate your reply anyways @andreasbrett. 

I have just went ahead _(for the 3rd time today 😩)_ wiped my SD card and started again with a fresh install. Just got Home Assistant up and running perfectly again...

I'll hang fire before installing nodejs and npm for now until someone can confirm but I can say for sure that both versions were about 10.x and 6.x.

Koenkk, 2019-02-20:
What is the output when running `python --version`?. This should be Python 2, otherwise you will get the syntax error like above.

SimplyGardner, 2019-02-20:
Ah, this is embarrassing. I don’t remember installing Python - I assumed it would have been covered / installed automatically when I installed home Assistant. HA is up and running fine.

These are the instructions I followed for installing in a virtual environment: https://www.home-assistant.io/docs/installation/raspberry-pi/

Do I have to install it separately?

Koenkk, 2019-02-20:
The problem is that it is probably running python 3 instead of 2, this can be confirmed by running `python --version`.

SimplyGardner, 2019-02-20:
Intresting - I get this:

```
root@SmartHome:~# python --version
-bash: python: command not found
```

Koenkk, 2019-02-20:
@SimplyGardner you are following this: http://www.zigbee2mqtt.io/information/virtual_environment.html right?

SimplyGardner, 2019-02-20:
Yeh, I am. But it fails at the npm install.

Any steps I can take to fix?

SimplyGardner, 2019-02-20:
Just checked ``/usr/bin`` and I have the following folders

```
python3 
python3.5m
python3m
pyvenv
pyvenv-3.5 
```

SimplyGardner, 2019-02-20:
ah, @Koenkk - looks like I am running Python 3.5.3 - does it not work with this version?

```
root@SmartHome:~# python3 --version
Python 3.5.3
```

JLFN, 2019-02-21:
Iam running Python 3.5.3 and its working perfect!

SimplyGardner, 2019-02-21:
Then I don’t think it’s an issue with Python (3)... wondering what it could be...

JLFN, 2019-02-21:
Have you tried the install.sh from script folder? on dev

SimplyGardner, 2019-02-21:
No? How do I do that?

JLFN, 2019-02-21:
1.  wget https://github.com/Koenkk/zigbee2mqtt/blob/master/scripts/install.sh
2. sudo  ./install.sh

SimplyGardner, 2019-02-21:
Okay, I think I managed to get it installed. I think it was because I did not have Python 2 installed as @Koenkk suggested. I have followed the documentation ([for installing in a virtual environment](https://www.zigbee2mqtt.io/information/virtual_environment.html)) from end-to-end and [here is the paste from the SSH terminal](https://pastebin.com/Hpvnh6Sy).

Please could you kindly check and let me know if this is good? I am not sure if it installed correctly?

I just need to go ahead and configure now but want to make sure that this is good to go. The only thing I managed to pick up from the SSH session is that some packages provided warnings and that it recommended I run ``npm audit fix`` - should I do that?

SimplyGardner, 2019-02-21:
Fixed.

Ultimately - it was down it was down to not having **Python 2** installed.

Thanks for your help guys @Koenkk and @andreasbrett

closed 2019-02-21 by SimplyGardner

ndfred, 2020-07-18:
I've been running into the same issue, though I feel like I brought it onto myself:

* I am using a vanilla Ubuntu Server install
* it is arm64
* I am running from a zigbee2mqtt user that has no shell and no home directory

As mentioned above, the solution is to run `sudo apt install python2 build-essential`