Koenkk/zigbee2mqtt #1132: Feature request: Ability to set device specific options in runtime using mqtt

Filipok-A, 2019-02-20:
First I would like to thank Koen for zigbee2mqtt, I've started using it about two weeks ago and it has been very stable since then.   All of the devices I am using are Xiaomi and it is great than I can go away from using their gateway.   There is only one feature that is preventing me to completely switch to zigbee2mqtt and switch off Xiaomi gateway; therefore a feature request:

Ability to set device specific options in runtime using mqtt.

My use case:
Almost all of my lights are controlled by presence sensors with no physical switches.   In some areas it is more complex to accurately detect if there is no one present or just the people there are not moving and I am using auxilary presense indicators to control the timeout when presense (and therefore lights) are turned off.   For example in living room default timeout is 180s but if TV is on the timeout is changed to 600s which allows enough time to detect movement without accidently switching lights off.  This works great for me using Openhab with addon for Xiaomi gateway.  

But with zigbee2mqtt the occupancy_timeout can only be set to a fixed value per device and can not be changed with mqtt and without restarting the service.

If you can add this feature it would also be very benefitial to restart the timeout timer, if one is already running for specific device.

Thanks in advance,
Philip

Koenkk, 2019-02-22:
Implemented, please check out the dev branch, documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/mqtt_topics_and_message_structure.md#zigbee2mqttbridgeconfigdevice_options

Filipok-A, 2019-02-24:
@Koenkk thank you for implementing this feature!  Works very well.

Some observations/possible enhancements:

1.  When changing a value of an option that is not set in the configuration.yaml the option is not changed - which is correct according to documentation.  But in the log a successful 'Changed device specific options' message is posted which is probably better logged with a warning.
2. Now that it is possible to set device options it would be great to have a way to retrieve options through mqtt.  Maybe introduce a new mqtt parameter in configuration similar to include_device_information which would add device options to all posted device messages.   This would allow to syncronise options after a restart. 
3.  At least for me using Openhab, it would be a lot more convinient if the options could be set individually through zigbee2mqtt/[DEVICE_ID]/option/+ path.

Consider the following example:

Current Openhab mqtt channel configuration
`Type number : motionOffTimer "Motion Off Timer" [ commandTopic="zigbee2mqtt/bridge/config/device_options", formatBeforePublish="
        {
            \"friendly_name\": \"0x00158d00016c3333\",
            \"options\": {
                \"occupancy_timeout\": %s
            }
        }"]`

Would be changed to
`Type number : motionOffTimer "Motion Off Timer" [ commandTopic="zigbee2mqtt/0x00158d00016c3333/option/occupancy_timeout"]`

closed 2019-05-02 by stale[bot]