Koenkk/zigbee2mqtt #1137: failed to join due to error: Unable to read genBasic, likely communication error

msadeghz, 2019-02-21:
Hi friends,
I want to add a new device to zigbee2mqtt, but after some data exchange, finally I see "failed to join due to error: Unable to read genBasic, likely communication error" message and the joining procedure is not finished. what's the problem? should I provide a new application like cieApp?

Koenkk, 2019-02-22:
I've never seen this error yet, can you provide the log?

msadeghz, 2019-02-23:
OK
you can find it in attached file.
the device address is 0x00124b0018dd053d
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2896680/log.txt)

Koenkk, 2019-02-23:
What device are you trying to pair?

msadeghz, 2019-02-23:
a temperature and humidity sensor which sends data repeatedly

Koenkk, 2019-02-23:
Can you provide an url to the device?

msadeghz, 2019-02-23:
[http://www.fbeeinside.com/zigbee-temperature-and-humidity-sensor-fzb870ath-s1-p00015p1.html](url)
the company says its zigbee profile is Standard HA 1.2. It seems that it can;t join and tries and tries to join

Koenkk, 2019-02-23:
Did you keep the device awake while joining? (by continuously pressing a button on it?)

msadeghz, 2019-02-23:
yes, the device doesn't need to be kept awake, it retries the join procedure automatically.

msadeghz, 2019-02-24:
An abstract log!
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2897778/log.txt)

Koenkk, 2019-02-24:
This device doesn't seem to be zigbee HA compliant, otherwise it would allow reading of the `genBasic` cluster. With what bridge is this device normally used?

msadeghz, 2019-02-24:
I think I found the problem
I use a zigbee sniffer. the ZNP wants to read genBasic attribute but the device responds the command is not supported.
so the zigbee2mqtt sends leave command to the device

[image]

closed 2019-02-24 by msadeghz

msadeghz, 2019-02-24:
I flashed the ZNP by z-stack 3, which wants the forked zigbee-shepherd. the data request command used for fined the modelID and ... in the forked version is different from default version of zigbee shepherd.

the default zigbee-shepherd command (uses read attribute command)
[default.txt](https://github.com/Koenkk/zigbee2mqtt/files/2897983/default.txt)

the forked version (uses discover attribute command)
[forked.txt](https://github.com/Koenkk/zigbee2mqtt/files/2897984/forked.txt)