Koenkk/zigbee2mqtt #1138: Not able to run zigbee2mqtt as a service

SimplyGardner, 2019-02-21:
Followed the **[Running Zigbee2mqtt in Virtual Environment](https://www.zigbee2mqtt.io/information/virtual_environment.html)** docs perfectly. Then at step 5: Running as a daemon with systemctl....

I create the file``sudo nano /etc/systemd/system/zigbee2mqtt.service`` and add
```
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/bin/bash -c 'source /opt/zigbee2mqtt/bin/activate; /opt/zigbee2mqtt/bin/npm start'
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```
I check start ``sudo systemctl start zigbee2mqtt``

### Upon checking the status, I get this error
```
dietpi@SmartHome:~$ sudo systemctl status zigbee2mqtt
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-02-21 15:37:26 GMT; 1min 32s ago
  Process: 1479 ExecStart=/bin/bash -c source /opt/zigbee2mqtt/bin/activate; /opt/zigbee2mqtt/bin/npm start (code=exited, status=217/USER)
 Main PID: 1479 (code=exited, status=217/USER)

Feb 21 15:37:26 SmartHome systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Feb 21 15:37:26 SmartHome systemd[1]: Stopped zigbee2mqtt.
Feb 21 15:37:26 SmartHome systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Feb 21 15:37:26 SmartHome systemd[1]: Failed to start zigbee2mqtt.
Feb 21 15:37:26 SmartHome systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Feb 21 15:37:26 SmartHome systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
```

SimplyGardner, 2019-02-21:
Actually... the above error was down to the fact I don't have a ``pi`` user on my system - instead I use ``diepi`` user. I corrected this and rebooted - however it does not start upon boot.

```
dietpi@SmartHome:~$ sudo systemctl status zigbee2mqtt
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
```

I started again ``sudo systemctl start zigbee2mqtt`` and checked the status and it looks as though its online.

Do you know how I can get this to run after a restart?

SimplyGardner, 2019-02-21:
Fixed. Had to enable using ``sudo systemctl enable zigbee2mqtt.service`` - wasn't clear in the virtual environment documentation. Having to switch between to pages.

closed 2019-02-21 by SimplyGardner

mercs600, 2019-04-03:
For everyone else who doesn't use virtual env. I got the same issue but with clean installation. I used node as runner instead of npm. 
```
ExecStart=/home/pi/.nvm/versions/node/v10.15.3/bin/node index.js
```

Please make sure your node path is correct by:
```
whereis node
```

Sir-Ironic, 2020-05-25:
WORKS !!!!
Really thanks.

2 days to read and try everything
nodejs (v12.16.3) is installed from root and haven't pi user, juste 'root' and i created 'jeed'

So, i just replace : 

ExecStart=/usr/bin/npm start
by
ExecStart=/root/.nvm/versions/node/v12.16.3/bin/node index.js  
and
User=pi
by
User=root

Now, service is launched at boot

JaraZem, 2024-06-19:
2024.06.19 WORKS !!!

So, i change :

User=pi  >>>>>  User=root

my file today looks:
--------------------------------------------------------------
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment=NODE_ENV=production
Type=notify
ExecStart=/usr/bin/node index.js
WorkingDirectory=/opt/zigbee2mqtt
# StandardOutput=inherit
StandardOutput=null
# Or use StandardOutput=null if you don't want Zigbee2MQTT messages filling syslog, for more options see systemd.exec(5)
StandardError=inherit
WatchdogSec=10s
Restart=always
RestartSec=10s
User=root

[Install]
WantedBy=multi-user.target
--------------------------------------------------------------
AČIŪ !!!