Koenkk/zigbee2mqtt #1139: Samsung SmartThings Multipurpose Sensor (2018)

pfischbach, 2019-02-21:
Samsung SmartThings Multipurpose Sensor (2018)? M/N is IM6001-MPP01 ([link](https://www.smartthings.com/products/smartthings-multipurpose-sensor)).

#545 (closed)
Started working on this Sensor again.
Had to setup a Raspberry outside of HAS

Can you provide the log when running with `DEBUG=* npm start` (need to figure out why configuring fails).

[debug](https://pastebin.com/mY3BmdmT)
[device.js](https://pastebin.com/JR5kZBwB)

Now I am getting messages after restart
But configuration still fails
zigbee2mqtt:debug 2/20/2019, 4:46:56 PM Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
zigbee2mqtt:warn 2/20/2019, 4:46:56 PM No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
zigbee2mqtt:warn 2/20/2019, 4:46:56 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.

The message appears cyclical but no message when open/closing the sensor magnet.

Koenkk, 2019-02-22:
Can you provide the log when running with `DEBUG=* npm start`, in this way we can see which configure statement fails.

pfischbach, 2019-02-22:
Should be in the debug pastbin above, or I can run it again.
zigbee-shepherd-converters:devices Configured '(cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb)' with result 'Error: AF data request fails, status code: 240. MAC transaction expired.' +16ms
  zigbee2mqtt:error 2/20/2019, 4:19:00 PM Failed to configure 0x286d9700010285ef 0x286d9700010285ef

Koenkk, 2019-02-22:
Can you wakeup the device while start zigbee2mqtt? (press a few button on it)

pfischbach, 2019-02-23:
Here is the log pushed the reset every 1s very short. during debug [debug](https://pastebin.com/gE0a1Nk5) 
0x286d9700010285ef: IM6001-MPP01 - Samjin-SmartThings MultiPurpose Sensor
Temperature works but no open close telegram

Koenkk, 2019-02-24:
Can you try with:

```js
   {
        zigbeeModel: ['multi'],
        model: 'IM6001-MPP01',
        vendor: 'Samjin-SmartThings',
        description: 'MultiPurpose Sensor',
        supports: 'contact, temperature, tilt, vibration',
        fromZigbee: [fz.generic_temperature],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                        (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 23}, cb),
                        (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                        (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
            ];
   
            execute(device, actions, callback);
        },
    },
```

pfischbach, 2019-02-24:
[Samsung Multi Log ](https://pastebin.com/aSaJj0eu)
Don't get any message when open and closing the sensor
Temperature is working

Koenkk, 2019-02-25:
The `mac transaction expired error` is probably due to #1090, until that is fixed, we need to work around this.

Can you remove the device from the network and re-pair it while keeping it close to the coordinator?

pfischbach, 2019-03-04:
I have the device very close to the coordinator. Will try again.

Koenkk, 2019-03-04:
When adding support for another device, I found out that you need to factory reset it and re-pair it once you have the device to `devices.js`. It seems that i only allows the `ssIasZone` configure right after pairing.

ericstewart22, 2019-03-10:
Home Assistant's ZHA now supports quirks and there is a quirk for this device. I can see the acceleration cluster from the ZHA control panel but I'm not sure how to enable it or do anything else. The YouTube links is a screen recording of all the configuration options available in Home Assistant.

https://youtu.be/5Tt33v-POLg

I had to unpair and repair the device to get it to work with Home Assistant and after renaming the device I had to restart Home Assistant.

marc-gist, 2019-03-22:
So is this sensor now working in zigbee2mqtt, or @ericstewart22 you just have it working in native HA ZHA component?

ericstewart22, 2019-03-23:
I haven't tried to use it with zigbee2mqtt in quite a while. Last time I tried, many months ago, open/close worked and temperature worked sometimes.

Native HA ZHA has support for open/close and temperature. It also seems to have support for the acceleration cluster but I think it needs to be enabled and I don't know how to do that.

pfischbach, 2019-04-29:
Still trying to get this to work.
Will update the firmware with a Samsung hub next week.
Got a sniffer stick going as well.
Will try again with updated firmware

mclaudiopt, 2019-04-30:
hi there
congrats for the work
any news about getting it work with zigbee2mqtt? thanks a lot

Fabiancrg, 2019-05-07:
Hi
You should have it if you upgrade Z2M to latest dev build.
We added it a few weeks ago:
https://github.com/Koenkk/zigbee-shepherd-converters/issues/406

Only thing, I am not too sure about tilt and vibration, does not look like mine is reporting anything else, only contact and temperature (checked with zigbee sniffer); but I don't have a ST Hub to upgrade to the latest FW so there might be a bug in mine.

Could you please check and let me know if yours is giving more data ?
Please take close look at the temperature, at first it was never changing and it sunddenly started sending regular temperature updates after several Z2M restarts... cannot explain why.

pfischbach, 2019-05-10:
Will update the sensor ona ST Hub next week and test it

mclaudiopt, 2019-05-10:
hi. this means that it maybe it will work better with z2mqtt?

mclaudiopt, 2019-05-11:
hello
i have this error
zigbee2mqtt:warn 5/11/2019, 7:08:44 PM Failed to configure Smarthings MultiSensor (0x286d970001010cd0) ('Error: Timed out after 30000 ms') (attempt #1)

also it only updates if i restart zigbee2mqtt. what can i do?
thanks

marc-gist, 2019-06-15:
@mclaudiopt i got the same on initial pairing... its working, but the temperature reading is always the same, 28.68.  Contact true/false works.  Also doesn't appear the sensor sends a regular status update (or at least one that zigbee2mqtt picks up :( )

mclaudiopt, 2019-06-15:
i already reseted...one time it worked ok during 2 weeks
now cannot make it work ....@Koenkk can you help?

Koenkk, 2019-06-15:
@mclaudiopt @marc-gist probably the configure didn't when right.

When starting zigbee2mqtt, make sure the device is awake (by pressing some buttons on it) so that the configure goes successful.

Fabiancrg, 2019-06-15:
I had the same poblem at the begining, temperature was not updated then once I restarted Z2M and while it was starting I was opening and closing the contact and since it's updating the temperature.
Z2M was restarted several time since, I am not doing this anymore but temp is still updated.
The only error I see in the log is the same (as above (Failed to configure...) but it does not seem to be a problem.

mclaudiopt, 2019-06-15:
i am on edge but temperature does not update
´``
zigbee2mqtt:warn 6/15/2019, 9:38:55 PM Failed to configure 0x286d970001010cd0 (0x286d970001010cd0) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
zigbee2mqtt:warn 6/15/2019, 9:38:55 PM This can be ignored if the device is working properly
zigbee2mqtt:debug 6/15/2019, 9:39:10 PM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":32}' of device 'multi' (0x286d970001010cd0) of endpoint 1
zigbee2mqtt:info 6/15/2019, 9:39:10 PM MQTT publish: topic 'zigbee2mqtt/0x286d970001010cd0', payload '{"contact":true,"battery_low":false,"linkquality":84,"temperature":24.07,"battery":"0.00","voltage":2500}'
zigbee2mqtt:debug 6/15/2019, 9:39:10 PM Configuring 0x286d970001010cd0 (0x286d970001010cd0) ...
zigbee2mqtt:debug 6/15/2019, 9:39:10 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x286d970001010cd0"' of device 'multi' (0x286d970001010cd0) of endpoint 1
zigbee2mqtt:warn 6/15/2019, 9:39:12 PM Failed to configure 0x286d970001010cd0 (0x286d970001010cd0) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #2)
zigbee2mqtt:warn 6/15/2019, 9:39:12 PM This can be ignored if the device is working properly

@Koenkk

mclaudiopt, 2019-06-15:
what i am saying is that uppon restart of zigbee2mqtt it updates, but then temperature remains the same forever

Koenkk, 2019-06-16:
@mclaudiopt  The temperature updates on startup because zigbee2mqtt send cached values (these do not come from the device). I see that the configure still fails. What firmware version are you running?

marc-gist, 2019-06-16:
i'm running 20190223 and have the same/similar issue.  Restart doesn't have an effect. only Contact true/false works for me.  And the samsung device never "checks-in" (device issue? or Z2M not seeing the heartbeat/check-in payload?)

mclaudiopt, 2019-06-16:
@Koenkk in my case i have this firmware:
M Coordinator firmware version: '20190425'
open/close works, temperature does not update
https://pastebin.ubuntu.com/p/B3xbmxxKPs/

Koenkk, 2019-06-17:
@mclaudiopt please remove all occurences of `0x286d970001010cd0` from `database.db` and try to re-pair the device. When the configures succeeds it will probably work.

Fabiancrg, 2019-06-17:
I am runing Z2M 1.4.0 with  CC2531 coordinator (FW 20190425) and it is working fine for me even if each time I restart Z2M I have these kind of messages:

````
6/11/2019, 10:42:55 AM - info: MQTT publish: topic 'zigbee2mqtt/SmartThings', payload '{"battery":"40.00","voltage":2700,"linkquality":31,"temperature":20.9,"contact":true,"battery_low":false}'
[...]
6/11/2019, 10:44:08 AM - debug: Configuring SmartThings (0x286d97000105d957) ...
6/11/2019, 10:44:09 AM - warn: Failed to configure SmartThings (0x286d97000105d957) ('Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.') (attempt #1)
6/11/2019, 10:44:09 AM - warn: This can be ignored if the device is working properly
[...]
6/11/2019, 10:45:07 AM - info: MQTT publish: topic 'zigbee2mqtt/SmartThings', payload '{"battery":"40.00","voltage":2700,"linkquality":39,"temperature":20.9,"contact":true,"battery_low":false}'
[...]
6/11/2019, 10:45:14 AM - warn: Failed to configure SmartThings (0x286d97000105d957) ('Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.') (attempt #2)
6/11/2019, 10:45:14 AM - warn: This can be ignored if the device is working properly
6
[...]
6/11/2019, 10:46:13 AM - info: MQTT publish: topic 'zigbee2mqtt/SmartThings', payload '{"battery":"40.00","voltage":2700,"linkquality":47,"temperature":20.9,"contact":true,"battery_low":false}'
[...]
6/11/2019, 10:46:44 AM - warn: Failed to configure SmartThings (0x286d97000105d957) ('Error: Timed out after 30000 ms') (attempt #3)
6/11/2019, 10:46:44 AM - warn: This can be ignored if the device is working properly
[...]
6/11/2019, 10:47:19 AM - info: MQTT publish: topic 'zigbee2mqtt/SmartThings', payload '{"battery":"40.00","voltage":2700,"linkquality":39,"temperature":20.9,"contact":true,"battery_low":false}'
````
But it is working properly as I have regular updates (you cannot see in the log above but it does work).

mclaudiopt, 2019-06-17:
@Koenkk i deleted from db but still same problem:-( temperature only updates uppon addon restart :-(

pfischbach, 2019-06-17:
Updated the Sensor Firmware on a Samsung hub to the latest version last week.
Tested with samsung hub   (Contact, Temperature and Vibration)
Would be good to add vibartion as well to Z2MQTT

Tested Today
Deleted the database entry and repaired.
Same issue, contact updates correctly but temperature is not updating as reported above.
zigbee2mqtt version 1.4.0  Coordinator firmware version: '20190515'
[Log](https://pastebin.com/TPv7Ka6k)

mclaudiopt, 2019-06-18:
in my case i have not updated the firmware since i do not have the hub.
sometime agora the sensor was working with all the updates (besides the vibration) the i had to reinstall zigbee2mqtt and stopped working.
already deleted the .db and removed device and re-paired but still not working

Koenkk, 2019-06-18:
@pfischbach the configure failed (on line 15), make sure that it **at least** configured it successfully (see https://github.com/Koenkk/zigbee2mqtt/issues/1139#issuecomment-502638934). If you have seen a successfully configured once, but the sensor it still not working, please let me know.

mclaudiopt, 2019-06-20:
@Koenkk this is my log
https://paste.ubuntu.com/p/4kC52TzCtd/

i already add it paired in the past with temperature updatinha but now seems not to be correctly paired

pfischbach, 2019-06-20:
Success!
Tested again, removed sensor including data base entry.
Paired it while continued to open close during the pairing process to keep it awake. 
Finally it worked and the temperature is updating appr. every 1 minute.  
The process is not very clean some errors show up during the process.
Would be god if we can get the vibration signal to work as well.

mclaudiopt, 2019-06-20:
hi @pfischbach can you please make a step-by step so that i can use also?
thanks

pfischbach, 2019-06-20:
@mclaudiopt 
This is what worked for me:
Publish device ID to zigbee2mqtt/bridge/config/remove
Stop Z2MQTT
Remove the 2 device data bases entrees from device.db
Remove device from zigbee configuration.yaml if still there.
Remove battery from device
Start Z2MQTT
Enable joining devices
Put battery back into device 
Open and close the contact every 1-2s until pairing is finished to keep awake.10-20s

mclaudiopt, 2019-06-21:
@pfischbach that worked for be ! thanks a lot !
@Koenkk is there anyway to improve this pairing method? also to add vibration sensor ?
is it already merged on the non edge version?
thanks a lot

Koenkk, 2019-06-23:
@mclaudiopt not at the moment (however maybe in the future as I'm refactoring the underlying zigbee library).

On vibration, does the sensor send any message? (seen by setting `log_level: debug` in configuration) http://www.zigbee2mqtt.io/configuration/configuration.html.

I've added the pairing instructions here: http://www.zigbee2mqtt.io/devices/IM6001-MPP01

mclaudiopt, 2019-06-23:
i will check that
altough the sensor is loosing connection and does not update :-(

truglodite, 2019-07-02:
I don't think this is directly related, but I found some older info on getting the accel to work with zha using an older version of the sensor (via sql sensor):
https://community.home-assistant.io/t/zigbee-zha-get-a-cluster-attribute-that-isnt-normally-tracked/122199

Unfortunately the newer samjin's don't have a Centraline cluster like that, just a ManufacturerSpecificCluster with the same address... and none of it's attributes come up in hassio zha config. My guess is some command/attribute needs to be sent to enable the accel output on the MSC cluster similar to the older Centralite versions?

I'm a newb regardling zha vs stIDE etc... would be nice if one could simply sniff the commands an st setup sends/recieves and emulate that in zha to setup the accel on these things.

closed 2019-09-07 by stale[bot]

thardie, 2020-09-09:
I was having the temperature not showing up problem. I did the paring with the suggested magnet every 1 second thing, but do it close to the coordinator. That made things work properly for me. Hopefully that helps someone.

Robertdecker12, 2021-05-29:
how do i get my sensor to report vibration I want it to report vibration on my washing machine and when stops let me know

crowbarz, 2021-05-30:
> how do i get my sensor to report vibration I want it to report vibration on my washing machine and when stops let me know

Vibration is reported as the `acceleration` Zigbee attribute, which is translated to the `moving` property by z2m (see State tab in the UI). This property turns to 1 when vibration is detected. If your sensor (Zigbee model `multi`) is not reporting this attribute, then try reconfiguring, and if that doesn't work then remove (force if necessary) and re-pair the sensor. If it is set up correctly, you should see an entry in the sensor's Reporting tab with cluster `manuSpecificSamsung` and attribute `acceleration`, and Zigbee message debug logs should show `acceleration` being sent whenever the sensor moves.

Note also that it is only the part of the sensor that houses the battery that can detect vibration/movement, the smaller part is just a magnet for the contact sensor and doesn't in itself trigger vibration detection.