Koenkk/zigbee2mqtt #1140: Philips Hue Outdoor Motion Sensor SML002 errors

jonnycastaway, 2019-02-21:
Hey @all,
seems like the challenge is not over.
After some playing and testing i cannot bind the Sensor with zigbee2mqtt again.
Message:
`Cannot get the Node Descriptor of the Device: 0x0017880106442d61`

Anyone an idea?

jonnycastaway, 2019-02-21:
Ok, got it.
You have to 'click' the switch on the backside repeatedly.
Click, wait for red light, then green light, then click and so on.
After several times (5-10) the Sensor connects again ;-)

Koenkk, 2019-02-22:
Did you remove if from that `data/database.db` before re-pairing?

jonnycastaway, 2019-02-22:
so journalctl says the sensor stops working at 00:22 AM.
i restarted the system at 06:30 AM with no luck. no sensor data.
magicaly since 14:10 PM the sensor started working again.
i have no idea where to search for the problem. any ideas?

Koenkk, 2019-02-22:
Using zigbee sniffing, can you verify that indeed something is send when motion is detected (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

jonnycastaway, 2019-02-23:
ok, here my further results.
is sniffed the traffic and saw that the zigbee channel is 11.
after checking my fritzbox i saw that my wifi-channel is 1.
oh, maybe some inteference problems? i switched the wifi-channel to 11 but there is a problem. i use also a fritzrepeater in a mesh-system and when the fritzbox-wlan is on channel 11 the repeater automatically switches to wifi-channel 1. when i swich the fritzbox wifi-channel to 1 the repeater automatically switch to wifi-channel 11. so there is no channel that not intefer with zigbee.
for further testing i removed the repeater and switched the fritzbox-wifi channel to 11.after reconnecting the sml002 sensor it seems to work. now i have to wait how long the sensor will do his job.

jonnycastaway, 2019-02-23:
> 
> 
> Using zigbee sniffing, can you verify that indeed something is send when motion is detected (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

After nearly 2 hours the Sensor stopped working again.
I sniffed the traffic and see many Network Address Request, Rejoin Request and Device Announcement.
I don't have the skills to interpret teh things right. So was can be done to bring the Sensor alive?

jonnycastaway, 2019-02-23:
i checked the communication for hours and i see that the sensor sends to the coordinator:
`Data Request` 
and
`Network Address Request`
but he never becomes an answer.

Koenkk, 2019-02-24:
Using the network map, what is the linkquality? It seems to drop off the network somehow, is this still the case when keeping it close to the coordinator?

jonnycastaway, 2019-02-24:
This morning, I put the senor 1m away from the coordinator and check him.
I thought the sensor would use one of my Osram smart + plugs as a repeater, but it seems that this is not the case.
In a few hours I know more.

jonnycastaway, 2019-02-24:
like i said, i installed the sensor outdoor again and switched some locations from wlan and zigbee devices. At the moment the sensor is working.
My actual [Network Map](https://www.dropbox.com/s/te3wcxnq5lzicuw/zigbeenetwork2.png?dl=0). Lets see how long it works.

jonnycastaway, 2019-02-25:
Ok, the sensor works now since 24h without a problem.
Let's see what happenes the next 24h.

jf-freeze, 2019-02-26:
Hi everyone! I also have trouble correctly including this sensor (with hassio on a Raspberry 3b+). It is recognized by zigbee2mqtt (I used a modified devices.js in the \share\zigbee2mqtt folder) and I can see the sensor data in the zigbee log as well as with mqtt.fx. However even though I tried to re-pair the sensor it is not shown in Home Assistant on the Mosqitto integration page as new device, as my other devices did. I tried to remove the sensor from database.db an pair it again but that doesn't change anything. What am I missing?

BTW: I think the definition in devices.js is missing the illumination measurement:

```
            (cb) => device.report('msIlluminanceMeasurement', 'measuredValue', 0, 600, null, cb),
```

Setting this also shows the illumination readings in my zigbee2mqtt log.

Thanks a lot in advance!

jonnycastaway, 2019-02-26:
Oh cool, Illuminance. I copied the Section from Hue Indoor Sensor and there is no Section for Illuminance.
So its pretty cool that you share this info. Can you put it in a Pull Request?

In HomeAssistant i integrate the Sensor by Hand with platform: mqtt. This works perfectly for me. I played a bit around with discovery but it's not reliable in my point of view.

papanirual, 2019-02-26:
So far the discovery worked without any problems. For some reason with this sensor it doesn't. The indoor sensor also has illuminance btw, I just copied exactly the same definition. I think that the capabilities of the two sensors in terms of what they measure are identical.

By manually you mean in configuration.yaml? Do you have an example of that for this sensor?

Pull request: Haven't done that yet, going to have a look.

jonnycastaway, 2019-02-26:
this is for motion.
for battery, temperature, illuminance and link quality you have to change some lines.

`- platform: mqtt
  name: "Hue Outdoor Motion"
  state_topic: "zigbee2mqtt/0x0017880106442d61"
  payload_on: true
  payload_off: false
  value_template: "{{ value_json.occupancy }}"
  device_class: motion
  qos: 0
  unique_id: 0x0017880106442d61_motion`

`- platform: mqtt
  name: "Hue Outdoor Temperature"
  state_topic: "zigbee2mqtt/0x0017880106442d61"
  unit_of_measurement: '°C'
  value_template: "{{ value_json.temperature }}"
  unique_id: 0x0017880106442d61_temp`

`- platform: mqtt
  name: "Hue Outdoor Battery"
  state_topic: "zigbee2mqtt/0x0017880106442d61"
  unit_of_measurement: '%'
  value_template: "{{ value_json.battery }}"
  unique_id: 0x0017880106442d61_battery`

jonnycastaway, 2019-02-26:
At the moment i have some sad news.
after nearly 72h of operating the last status from the sensor arrived at 19:59 PM. (now its 21:45 PM here).
So the sensor is actually not working. In the Map it has connection to 5 routers and the coordinator but with linkquality of 20 to 45. not much. it is stated as offline. i will check tomorrow with a osram smart+ plug on 1m distance. it seems the sensor has a bad zigbee range or the osram plugs won't work well as router.

papanirual, 2019-02-26:
OK, going to have a try, let's see if my sensor behaves.

jonnycastaway, 2019-02-26:
After i restarted my complete RaspberryPi the sensor starts working again.
How long? Surprise me.

thissiac, 2019-03-01:
The Philips SML002 (Philips Hue motion Outdoor sensor (EndDevice)) is installed in my Openhab2 enviroment. After updating my device.js it is working very well. But for illumination measurement I can't get any values.

Is there any entry in device.js I forgot?

**journalctl -u zigbee2mqtt.service -f**

Mär 01 18:41:35 openHABianPi npm[579]:   zigbee2mqtt:info 2019-3-1 18:41:35 MQTT publish: topic 'zigbee2mqtt/0x0017880106444268', payload '{"temperature":7.73,"linkquality":42,"occupancy":true,"battery":100}'

**devices.js**

{
        zigbeeModel: ['SML002'],
        model: '9290019758',
        vendor: 'Philips',
        description: 'Hue motion Outdoor sensor',
        supports: 'occupancy, temperature, illuminance',
        fromZigbee: [
            fz.hue_battery, fz.generic_occupancy, fz.generic_temperature,
            fz.ignore_occupancy_change, fz.generic_illuminance, fz.ignore_illuminance_change,
            fz.ignore_temperature_change,
        ],
        toZigbee: [tz.generic_occupancy_timeout],
        ep: (device) => {
            return {
                '': 2, // default
                'ep1': 1,
                'ep2': 2, // e.g. for write to msOccupancySensing
            };
        },
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 2);

        const actions = [
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.bind('msIlluminanceMeasurement', coordinator, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.bind('msOccupancySensing', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 1000, 0, cb),
                (cb) => device.report('msOccupancySensing', 'occupancy', 0, 600, null, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
                (cb) => device.report('msIlluminanceMeasurement', 'measuredValue', 0, 600, null, cb),
        ];

                execute(device, actions, callback);

        },
},

One more question: Is it possible to shorten the entries in the journal a bit? The date and time are displayed twice. Maybe only once is enough ;-)

**_Mar 01 18:41:45 openHABianPi npm[579]: zigbee2mqtt:info 2019-3-1 18:41:45 Zigbee publish to device...._**

thissiac, 2019-03-02:
**UPDATE:**

After many tests I decided to reinstall z2m. After a restart, "illuminance" is now also recognized.

_"... payload '{"temperature":6.75,"linkquality":55,"occupancy":true,"battery":100,"illuminance":17713}' "_

How do I find out which values are available? What is the range? If I have an "illuminance" of 15794, how many lux is that? Where do you get this information from?

The same applies to the value for "linkquality". What corresponds to 100%?

Koenkk, 2019-03-02:
@thissiac 100% = 255

papanirual, 2019-03-03:
@jonnycastaway: Unfortunately my three sensors are also behaving unreliable (on hassio, but I don't think it has to do with the system). The messages of the devices in the zigbee2mqtt log just stop, today 20hs after the last raspi restart, while for example the indoor sensor works just fine. Also the outdoor sensors don't always light up anymore on movement, so it seems they just hangs, but later they work again. It is frustrating, as the device is great as long as it works. 

If there should be an update from Philips (firmware) I would only be able to update the sensors using a hue bridge, right? At the moment I don't own one as my target is to avoid cloud based services (as do many considering zigbee2mqtt, I assume).

papanirual, 2019-03-04:
In my case it may have to do with the link quality. The moment it drops to zero (meaning a very bad [no] connection) for the sensor, the last state transmitted stays the known state while it should be "Not available", or am I misunderstanding something? What also strikes me as strange is that afterwards, when the connection gets better, still no message can be found in the log until a reset. 

Any ideas?

papanirual, 2019-03-04:
I tested it with another outdoor sensor and one indoor sensor as well as two xiaomi sensors. The problem is the same for all. The moment I move a sensor out of reach it stays with the last state. And when I move it back near the zigbee2mqtt stick it still stays the same, no updates. I am quite sure the sensors work but there is not entry in mqtt (also checked with mqttfx). I did not check the mesh network as I don't have a second stick. For me it looks like the devices are not accepted back after the connection was lost once. :-(

Is there anything I can do to help to analyse this problem?

In the meantime I will try to extend the range of my zigbee network to prevent low link quality.

papanirual, 2019-03-14:
In the meantime I made some progress. First I moved from hass.io to hasbian as I had performance issues with hass.io. Even though I found it nice and comfortable to have some work done for me, the lack of control and the performance issues bugged me. Well, I am better now. ;-) 

Second I actually read "How to pair new devices" now. Previously I thought that all steps had been made, but as the binary sensors of the outdoor sensors did not work anymore with Home Assistant on hasbian I looked again at why the sensors have not been automatically discoverred via mqtt. The reason is simple and can be found at the end of the "How to support..." documentation: "In order to automatically discover this device in home assistant your device needs to be added to mapping in lib/extension/homeassistant.js..

So i copied the part from the indor sensor and corrected the ID, leading to this in \opt\zigbee2mqtt\lib\extension\homeassistant.js:

`    '9290019758': [
        configurations.binary_sensor_occupancy, configurations.sensor_temperature,
        configurations.sensor_illuminance, configurations.sensor_battery,
    ],
`

Now the sensor is actually discovered. I still had some trouble with the stability of the connection, currently I am waiting for my CC2530 router to improve signal quality.

Will let you know how it works out. ;-}

jonnycastaway, 2019-03-21:
@papanirual 
has the cc2530 improved link quality?
in my environment it turns from 130 to 0 and back over the day. there are periods of time from 1 to 5 hours where the sensor nothing reports and then its working for 1 to 5 hours without a problem.
i ordered a 2530 also.

papanirual, 2019-03-22:
It hasn't arrived yet... Still in delivery. My system has been running for a couple of days now with all sensors near the zigbee2mqtt stick. Sometimes one or the other hang but after some time display correctly again (without restart). I will have to wait to see how it works with the 2530.

papanirual, 2019-04-05:
So... I got the CC2530+CC2591 and flashed it as router. It works and the general link quality and the reach of the network increased. 

But: The outdoor sensor still seems to loose connection at some point and it seems that it can even happen with a generally good link quality in proximity of the zigbee2mqtt stick. @Koenkk: is it possible that the sensor is discarded at some point, maybe because it goes into a sleep mode or something? Is there any way to "refresh" a connection to a device?

If you look at the last 24hs the link quality (as well as all other data from this entity) just stays at the last known value:

[image]

According to the zigbee2mqtt.log, the last known data from the sensor was:

2019-4-4 18:28:11 - info: MQTT publish: topic 'zigbee2mqtt/0x0017880106443a0d', payload '{"occupancy":false,"linkquality":107,"illuminance":31525,"battery":100,"temperature":17.47}'

There was no error (for this device) afterwards, everything else continued to work.

zigbee2mqtt version is: 1.2.1, home assistant version is: 0.89.1, mosquitto version is: 3.1.1.

I am very frustrated with that as I wanted to use the zigbee network for motion triggered lights, but at least with these sensors I don't think I can implement that reliably. :-(

Koenkk, 2019-04-07:
Perhaps things will get better with zigbee 3.0, please keep an eye on #211

thissiac, 2019-04-17:
2019-04-17 16:21:15.202 [vent.ItemStateChangedEvent] - O1_Door_Sensor_01_Signal changed from 0 to 70

2019-04-17 16:21:15.205 [vent.ItemStateChangedEvent] - O1_Door_Sensor_01_Helligkeit changed from 34790 to 35115

For my understanding the SML002 has a sleep mode. If no changes are detected, it goes in sleep mode.
If there is an update, it turns on the radio, send the message and goes bag to sleep mode.

Fabiancrg, 2019-05-02:
Same issue for me, works well (without any value for "illuminance" but today was not a sunny day) but after a few hours it's not sending any information. I see the detection is working as the red light turns on and off when movement is detected but nothing is published by zigbee2mqtt. Will try to run a sniffer and see what happens.

Fabiancrg, 2019-05-05:
From factory, motion sensor was using FW 6.1.0..25261, I did an upgrade today to version 6.1.1.27575 using the Hue bridge.
Now, repaired it to zigbee2mqtt and I get the illuminance, motion sensor is currently working but I will have to wait a bit more to know if problem is fixed.

Fabiancrg, 2019-05-06:
Unfortunately it's the same issue with the latest FW, after a few hours, no updates are reaching zigbee2mqtt.
I can see the sensor detecting the presence but noting is logged in z2m, anymoe has an idea how to fix this ?

Koenkk, 2019-05-06:
Perhaps this is the same issue as #1408. This should be fixed in the latest dev firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin

papanirual, 2019-05-06:
I will give it a try, thanks!

Fabiancrg, 2019-05-06:
FW and SW updated to the latest dev branch.
It's currently working, will see in a few hours if device is still reporting the temp, motion and illumination.

Fabiancrg, 2019-05-07:
@Koenkk After 16 hours it's still sending data regularly and detecting motion so problem is fixed for me.
Thank you !

Koenkk, 2019-05-07:
@Fabiancrg great, if @papanirual can also confirm this I guess we can close this issue.

papanirual, 2019-05-07:
My first attempt did not work, used new firmware and current dev zigbee2mqtt and now everything is unstable... :-/ Will try again with stable zigbee2mqtt tomorrow or the day after tomorrow.

papanirual, 2019-05-13:
My problem seems to be that I have a lot of devices in my existing network and that only works stable as long as my CC2530 router is connected. Unfortunately my CC2530 also looses connection after some time and from there on other devices cannot connect anymore. I will try to reduce the network back to a much smaller number of devices and try again. I am hoping for the CC2652R to become stable so that I can make an attempt with it... :-)

papanirual, 2019-05-28:
@Koenkk:  OK. It seems, that the outdoor sensors keep on working, mostly. Sometimes they are "gone" for a bit, even when they seem to work (red LED blinks on movement). But they always come back soon and work reliably most of the time. The indoor sensor doesn't work, though. I removed it from the database and the configuration and repaired it. The pairing works -  but the occupancy just stays in state "occupancy": true (according to state.json).

Attached is the zigbee2mqtt.log.
[hue_indoor.txt](https://github.com/Koenkk/zigbee2mqtt/files/3228675/hue_indoor.txt)

Looks like it doesn't send anything after the pairing has succeeded?

papanirual, 2019-06-27:
So, after a while of watching I can now confirm the hue outdoor sensor is working stably with zigbee2mqtt! Thanks!!

Fabiancrg, 2019-06-27:
@papanirual  which version of Z2M and which firmware are you using ?

papanirual, 2019-06-27:
2019-6-16 11:56:26 AM - info: Starting zigbee2mqtt version 1.4.0 (commit #6b75465)
2019-6-16 11:56:28 AM - info: Coordinator firmware version: '20190425'

Fabiancrg, 2019-06-29:
Mine setup uses a CC2531 with coordinator firmware version: '20190619' and zigbee2mqtt version 1.4.0 (commit #927c4db), it's the master version from two days ago.

@Koenkk I was running Z2M from dev branch for a while and upgraded a few days ago but had to fall back to master version as it was not working anymore for me, Z2M stopped working after a few minutes or hours without any message being logged (even in debug mode). 
Note also that my Philips Hue motion sensor is still working fine with this FW version (C2531_SOURCE_ROUTING_20190619).

papanirual, 2019-07-07:
Question: Has anyone been able to find out the control commands to reduce the sensitivity of the sensor? With wind in the bushes and trees near the sensor, it tends to trigger far too often...

Koenkk, 2019-07-07:
@papanirual https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/devices/9290019758.md#motion-sensitivity (only in dev branch).

Fabiancrg, 2019-07-09:
I tested this and it looks to work pretty well, I an now reduce the sensitivity during the day (to avoid false positive) and increase at night.

closed 2019-09-14 by stale[bot]