Koenkk/zigbee2mqtt #1141: Xiaomi QBKG03LM got undefined device

pkedvessy, 2019-02-21:
I moved my coordinator from Windows to Linux with around 60 devices in the configuration.yml but most of my QBKG03LM wall switches (not all of them) got undefined:
`  zigbee2mqtt:debug 2/21/2019, 10:08:51 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'undefined' (0x00158d00016da703)
  zigbee2mqtt:warn 2/21/2019, 10:08:51 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2/21/2019, 10:08:51 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

Some of them still working as before on Windows. When I'm moving back the coordinator to Windows, the issue disappears (on Windows, but still exists on Linux with the same configuration).

Do you have any idea on how I can move my coordinator with it's configurations to Linux?

Koenkk, 2019-02-22:
Can you try to
-  Stop zigbee2mqtt
- Remove all entries of `0x00158d00016da703` from `data/database.db`
- Start zigbee2mqtt
- Pair again.

roygbiv856, 2019-02-22:
@Koenkk I had some problems pairing xiaomi devices yesterday, so I deleted my database.db. Once I did that, I was successfully able to pair the troublesome sensors again. When I started repairing my devices, the new database.db was very organized with each device ID taking up consecutive lines. For example, 1, 1, 1, 2, 2, 2, 3, 3, 3, etc. Once I got to about 8 devices, my database.db became unorganized, as in, 1, 1, 8, 2, 3, 2. I also began having trouble pairing devices and maintaining a connection with already paired sensors. Does the "organization" of the database.db have any influence on the stability of the network or is my problem likely elsewhere?

pkedvessy, 2019-02-22:
@Koenkk Thanks for the suggestion, it helped me. Could you please detail what happened and why effected only my QBKG03LM switches? Thanks in advance.

Koenkk, 2019-02-22:
@roygbiv856 the `database.db` is just a lookup list, so shouldn't influence the stability network (the CC2531 takes care of this).

@pkedvessy are you sure you correctly moved the `database.db` file when migrating?

pkedvessy, 2019-02-22:
@Koenkk What do you mean on correctly moded? It was a simple copy paste...

Koenkk, 2019-02-23:
@pkedvessy then it was correctly moved, I honestly don't know why you got the 'undefined' device error..

closed 2019-04-26 by Koenkk