Koenkk/zigbee2mqtt #1143: device list doesn't contain coordinator

andreasbrett, 2019-02-22:
When requesting the list of devices through **zigbee2mqtt/bridge/config/devices** the resulting list contains all routers and end devices but not the coordinator itself.

Koenkk, 2019-02-22:
This is on purpose, what would be the use case for this?

andreasbrett, 2019-02-22:
Use case for me is that I regularly check the list of devices against my list of known devices. Some kind of combined security check and monitoring if you will. My smart home will let me know if there's an unknown device in my zigbee network (security) and also if a known device is not in the network anymore (monitoring).

What was the initial purpose on omitting the coordinator? I guess one could argue from both sides. Have it in the list of devices because devices are coordinators, routers and end devices. But also: omit it from the list as devices could also mean "all devices connected to the coordinator".

Koenkk, 2019-02-22:
The current device list is a list of all clients, so all devices - coordinator.

For the above use case, what would be the benefit of having the coordinator? As the coordinator will always be there, there is nothing to monitor there.

andreasbrett, 2019-02-22:
Yes, you're right. The only benefit would be that I could re-use my whole list of zigbee devices. I think we can close this, I will work around with my script.

closed 2019-02-22 by andreasbrett