Koenkk/zigbee2mqtt #1144: Messages rate (is this a zigbee2mqtt configuration, or a device one?)

wsw70, 2019-02-22:
This is probably (hopefully) not a bug but rather an uncertainty on my side. 

I see that `zigbee2mqtt` is sending its messages to the MQTT broker approximatively every second:

```
Feb 22 06:25:05 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 6:25:05 AM Connected to MQTT server
Feb 22 06:25:05 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 6:25:05 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
(and then the status messages)
```

The documentation just mentions the kind of messages which are sent, not the rate. Specifically, I was wondering whether the messages are always sent, every second, from `zigbee2mqtt` to the broker --  or whether they are sent when a message from the Zigbee device is received by `zigbee2mqtt`.

The latter would be very problematic as it would mean that the device is sending data continuously, as opposed to do it only when its readings change (and I do not know if this is something which is built-into the device, or that it is forced to do so by `zigbee2mqtt`)

Koenkk, 2019-02-22:
This message doesn't have anything to do with the zigbee devices. It should only be send when zigbee2mtt reconnects to the MQTT broker. Zigbee2mqtt checks the connection with MQTT every second (therefore you probably see it every second).

However, this should ofcourse only happend when your MQTT broker is down, therefore it's very strange that you see this every second.

Can you post your complete log?

wsw70, 2019-02-22:
The log from the last few minutes is below. The messages are repeated all the time at the same rate.

The `wlx00c0ca384bd9` is an unconfigured Wifi card, otherwise unused (the python3 script error is related to it but otherwise does not matter to me for the moment). I left it in the log for completeness.  Connectivity (in particular to the MQTT broker) is fine, a lot of devices and services use it continuously (from the host running `zigbee2mqtt` and others)

```
Feb 22 11:36:40 srv wpa_supplicant[879]: wlx00c0ca384bd9: CTRL-EVENT-SCAN-FAILED ret=-100 retry=1
Feb 22 11:36:41 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:41 AM Connected to MQTT server
Feb 22 11:36:41 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:41 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Feb 22 11:36:41 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:41 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002c77517', payload '{"temperature":24.03,"linkquality":97,"humidity":36.38,"pressure":1017,"battery":80,"voltage":2965}'
Feb 22 11:36:41 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:41 AM MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_temperature","unique_id":"0x00158d0002c77517_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Feb 22 11:36:41 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:41 AM MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_humidity","unique_id":"0x00158d0002c77517_humidity_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Feb 22 11:36:41 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:41 AM MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/pressure/config', payload '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_pressure","unique_id":"0x00158d0002c77517_pressure_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Feb 22 11:36:41 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:41 AM MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_battery","unique_id":"0x00158d0002c77517_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Feb 22 11:36:41 srv python3[8326]: command failed: Network is down (-100)
Feb 22 11:36:41 srv python3[8326]: 2019-02-22 11:36:41    ERROR (simplemonitor) monitor failed: WMQTT (Command '['iw', 'dev', 'wlx00c0ca384bd9', 'scan']' returned non-zero exit status 156.)
Feb 22 11:36:41 srv wpa_supplicant[879]: wlx00c0ca384bd9: CTRL-EVENT-SCAN-FAILED ret=-100 retry=1
Feb 22 11:36:42 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:42 AM Connected to MQTT server
Feb 22 11:36:42 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:42 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Feb 22 11:36:42 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:42 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002c77517', payload '{"temperature":24.03,"linkquality":97,"humidity":36.38,"pressure":1017,"battery":80,"voltage":2965}'
Feb 22 11:36:42 srv npm[1967]: #033[32m  zigbee2mqtt:info#033[39m 2/22/2019, 11:36:42 AM MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_temperature","unique_id":"0x00158d0002c77517_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_versio [... cut ...] e_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_battery","unique_id":"0x00158d0002c77517_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'

```

Koenkk, 2019-02-22:
Can you provide the log from the `data/log` directory?

wsw70, 2019-02-22:
The last few events from `data/log` are below, they do not seem to be different form the ones in `/var/log/syslog`.  Would you like me to send more events? Maybe in a debug mode? (I chhecked https://github.com/Koenkk/zigbee2mqtt/issues/25 but the page referenced over there does not seem to be available)

```
2/22/2019, 12:30:42 PM - info: Connected to MQTT server
2/22/2019, 12:30:42 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2/22/2019, 12:30:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002c77517', payload '{"temperature":24.24,"linkquality":94,"humidity":33.16,"pressure":1017.4,"battery":80,"voltage":2965}'
2/22/2019, 12:30:42 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_temperature","unique_id":"0x00158d0002c77517_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2/22/2019, 12:30:42 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_humidity","unique_id":"0x00158d0002c77517_humidity_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2/22/2019, 12:30:42 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/pressure/config', payload '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_pressure","unique_id":"0x00158d0002c77517_pressure_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2/22/2019, 12:30:42 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_battery","unique_id":"0x00158d0002c77517_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2/22/2019, 12:30:43 PM - info: Connected to MQTT server
2/22/2019, 12:30:43 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2/22/2019, 12:30:43 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002c77517', payload '{"temperature":24.24,"linkquality":94,"humidity":33.16,"pressure":1017.4,"battery":80,"voltage":2965}'
2/22/2019, 12:30:43 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_temperature","unique_id":"0x00158d0002c77517_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2/22/2019, 12:30:43 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_humidity","unique_id":"0x00158d0002c77517_humidity_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2/22/2019, 12:30:43 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/pressure/config', payload '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_pressure","unique_id":"0x00158d0002c77517_pressure_zigbee [... cut ...]
2/22/2019, 12:30:44 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002c77517/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d0002c77517","json_attributes_topic":"zigbee2mqtt/0x00158d0002c77517","name":"0x00158d0002c77517_battery","unique_id":"0x00158d0002c77517_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c77517","name":"0x00158d0002c77517","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'

```

Koenkk, 2019-02-22:
Can you provide the debug log (`log_level: debug` http://www.zigbee2mqtt.io/configuration/configuration.html)

wsw70, 2019-03-02:
Sorry for the delay, I was on vacation.

I lowered the log level to `debug `and restarted `zigbee2mqtt`. There are no more continuous updates :-|
I have no idea why (having of course restarted  `zigbee2mqtt` before).

I will close this issue for now and re-open if I see them reappearing (I will keep the debug log running).

Sorry for the noise.

closed 2019-03-02 by wsw70