Koenkk/zigbee2mqtt #1149: Changing what router end device is connected to.

GForce2010, 2019-02-23:
I have just successfully setup the dev version of Zigbee2MQTT and have gotten three Ikea bulbs, one Innr bulb and a Xiaomi button paired.

to make testing easier I have placed all the items around me. On checking the network map that zigbee2mqtt provides I can see that my button has connected to one of the Ikea bulbs as this is also a router.

Unfortunately the final physical locations of the bulb and the button would be quite far away from each other. Once moved would the button automatically connect to a closer router or would I have to re-pair the button in its final location to do so?

If I do have to re-pair the button what is the process I need to follow? Do I have to remove it from zigbee2mqtt by sending it's name to zigbee2mqtt/bridge/config/remove before I can re-pair or can I just hit the pairing button on the back to cause it to connected to closer routers?

Thanks.

Koenkk, 2019-02-24:
- First try if they automatically reconnect when moving them further away.
- If you need to re-pair just repeat the pairing procedure. (if that works, there is no need for the remove).

GForce2010, 2019-02-24:
Thank you for the clarification.

closed 2019-02-24 by GForce2010