Koenkk/zigbee2mqtt #1152: Paired device RTCGQ11LM  disappears.

jvdpas, 2019-02-23:
Paired device RTCGQ11LM  disappears.

Moment it disappears 
**2/24/2019, 12:08:27 AM - warn: Message without device!**
followed by 
**2/24/2019, 12:08:52 AM - error: Cannot get the Node Descriptor of the Device: 0x00158d00022b942c (Error: Timed out after 10000 ms)**

Version info log
2/24/2019, 12:25:02 AM - info: Starting zigbee2mqtt version 1.1.1 (commit #92d88b6)
2/24/2019, 12:25:02 AM - info: Starting zigbee-shepherd
2/24/2019, 12:25:03 AM - info: zigbee-shepherd started
2/24/2019, 12:25:03 AM - info: Coordinator firmware version: '20190217'

failure log
2/23/2019, 9:21:01 PM - warn: `permit_join` set to  `true` in configuration.yaml.
2/23/2019, 9:21:01 PM - warn: Allowing new devices to join.
2/23/2019, 9:21:01 PM - warn: Set `permit_join` to `false` once you joined all devices.
2/24/2019, 12:08:27 AM - warn: Message without device!
2/24/2019, 12:08:52 AM - error: Cannot get the Node Descriptor of the Device: 0x00158d00022b942c (Error: Timed out after 10000 ms)
2/24/2019, 12:10:35 AM - error: Cannot get the Node Descriptor of the Device: 0x00158d00022b942c (Error: Timed out after 10000 ms)
2/24/2019, 12:20:36 AM - error: Cannot get the Node Descriptor of the Device: 0x00158d00022b942c (Error: Timed out af

Koenkk, 2019-02-24:
The device somehow remove itself from the network.

- Do you have another zigbee network running? (e.g. Xiaomi hub).
- Did you try replacing the batteries?
- Does this still happen when the device is close to the coordinator?
- What devices do you have in your network?

jvdpas, 2019-02-24:
I paired 3 times 3 times same result. See your questions below.(Thanx for the quick response on this sunny sunday)

* Do you have another zigbee network running? (e.g. Xiaomi hub).
 **No**
* Did you try replacing the batteries?
**Yes**
* Does this still happen when the device is close to the coordinator?
**It was already close. Aprox 40 cm.**
* What devices do you have in your network?
**They are in the database.db : 
Name:	ieeeAddr:	Model:	Type:
Router_1	0x00124b0019365ed7	CC2530.ROUTER	Router
Stopcontact_1	0x000d6ffffec3a945	E1603	Router
Lamp_kleur_1	0xd0cf5efffe29766f	LED1624G9	Router
Schakelaar_1	0x00158d0002c715d5	WXKG01LM	EndDevice**

Koenkk, 2019-02-24:
After +- how long does it disapear?

jvdpas, 2019-02-24:
1st pair a few minutes.
2nd pair(with deleting device fromdatabase.db) ~2 hours
3rd ~4 hours when starting to think it worked it disappeared again.

Could  you provide some info where too look?

Koenkk, 2019-02-25:
Do you have multiple `RTCGQ11LM`? Does this happen with all of them?

jvdpas, 2019-02-25:
I have one. It happens to all of them ;-).

Koenkk, 2019-02-26:
When removing the E1603, does this issue still happen (just plug it of the power)?

jvdpas, 2019-03-01:
Measured the 2 used CR2450's. The Open Circuit Voltages of the used CR2450.
1st 2.78v
2nd 2.84v 
When looked at the specs (attached below) i concluded both cells must be empty.
Bought 2 new cells measured an OCV of 3.1V
Pairing went much smoother sensor is working now. I will report back when it's still working tommorow.
[CR2450.pdf](https://github.com/Koenkk/zigbee2mqtt/files/2919818/CR2450.pdf)

closed 2019-03-01 by jvdpas

jvdpas, 2019-03-10:
After one week of correct operation it can be concluded that it was the battery,

InToSSH, 2022-02-11:
I have this same issue. My network is paired with around 10 of these sensors. Two of them died already. My issue unfortunatelly persists even after replacing the batter with a fresh one, which has 3.3V.