Koenkk/zigbee2mqtt #1153: zigbee2mqtt and Ikea Outlet Switch

antti365, 2019-02-24:
I want to return to the topic of the Ikea switch again.
I asked my friend to sniff packets transmitted by the Ikea switch with a short and long press on and off, the results are as follows:
short and long press on 1:
Command: On (0x01)
Move mode: Up (0x00)

Zigbee2mqtt log:
publish: topic 'zigbee2mqtt/Ikea_Switch', payload '{"battery":87,"linkquality":39,"elapsed":34442,"click":"on"}'
publish: topic 'zigbee2mqtt/Ikea_Switch', payload '{"battery":87,"linkquality":39,"elapsed":2386,"click":"brightness_down"}'

short and long press on 0:
Command: Off (0x00)
Move mode: Down (0x01)

Zigbee2mqtt log:
publish: topic 'zigbee2mqtt/Ikea_Switch', payload '{"battery":87,"linkquality":44,"elapsed":1389023,"click":"off"}'
publish: topic 'zigbee2mqtt/Ikea_Switch', payload '{"battery":87,"linkquality":42,"elapsed":3277,"click":"brightness_up"}'

@Koenkk , please change up and down in zigbee2mqtt. It will be more logical.

Koenkk, 2019-02-24:
Changing this should be pretty easy, just change the converters here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L150

Could you test this and make a PR?

antti365, 2019-02-24:
Test successfully, PR created, thanks!

Koenkk, 2019-02-24:
Merged, thanks!

closed 2019-02-24 by Koenkk