Koenkk/zigbee2mqtt #1154: Light level RTCGQ01LM possible?

iFloris, 2019-02-24:
Did you read the FAQ?
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device?
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

Koenkk, 2019-02-24:
No, the RTCGQ01LM doesn't have an illuminance sensor, buy the RTCGQ**1**1LM  if you want illuminance.

closed 2019-02-24 by Koenkk

iFloris, 2019-02-24:
Thanks @Koenkk for the quick response. I am a bit confused, as I was able to get light level information from the sensor with the xiaomi hub. I will look into the model you suggested, thanks!

thomashogema, 2020-07-03:
I just started converting to zigbee2mqtt (working great so far!), but can also confirm the RTCGQ01LM human body sensor _does_ include illumination when interacting via the Aqara hub. Is it possible to add this functionality, or was the Aqara gateway faking data somehow before?

Koenkk, 2020-07-03:
@thomashogema strange that the gateway shows values, an easy check to see if they make sense is to first put the sensor in dark (check values of gateway) and then put it in full day light (check values again), do the values on the gateway significantly change?

thomashogema, 2020-07-03:
@Koenkk thanks for the quick reply!
I've just removed the old entry from HA so unfortunately I can't look back at the history anymore. However, I am quite sure that the sensor values made sense (value between 0 and 1000 depending on the time of day).

If I can be of any help by reconnecting it to the aqara gateway and check some values let me know!