Koenkk/zigbee2mqtt #1155: State reported twice

GForce2010, 2019-02-24:
I just wanted to check to see if the behaviour I am seeing is what is supposed to happen.

I have zigbee2mqtt setup and I am controlling it via OpenHab. 

When I send a command to change the brightness of a lamp I get two responses in the state topic, the first one is the current brightness and the second one is the new brightness I have requested. Everything is working but it causes the on screen control in OpenHab to flicker during the state change.

If that's how it works then fine but is there a way to disable to first state report?

Thanks.

Koenkk, 2019-02-25:
Can you post the log of this?

Kryzek, 2019-02-25:
I experience this too so I think I can also provide some log:
```
  zigbee2mqtt:debug 2/25/2019, 12:08:30 PM Received MQTT message on 'zigbee2mqtt/0x00158d0001cded1c/set' with data '{"state": "ON"}'
  zigbee2mqtt:info 2/25/2019, 12:08:30 PM Zigbee publish to device '0x00158d0001cded1c', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2/25/2019, 12:08:30 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":108,"linkquality":34,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:08:30 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)
  zigbee2mqtt:info 2/25/2019, 12:08:30 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":108,"linkquality":34,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:08:30 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)
  zigbee2mqtt:info 2/25/2019, 12:08:30 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":108,"linkquality":34,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:08:30 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)
  zigbee2mqtt:info 2/25/2019, 12:08:30 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":108,"linkquality":34,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:08:50 PM Received MQTT message on 'zigbee2mqtt/0x00158d0001cded1c/set' with data '{"state": "ON", "brightness": 93}'
  zigbee2mqtt:debug 2/25/2019, 12:08:50 PM Skipping 'state' because of 'brightness'
  zigbee2mqtt:info 2/25/2019, 12:08:50 PM Zigbee publish to device '0x00158d0001cded1c', genLevelCtrl - moveToLevelWithOnOff - {"level":93,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2/25/2019, 12:08:50 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":108,"linkquality":34,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:08:50 PM Received zigbee message of type 'attReport' with data '{"cid":"genLevelCtrl","data":{"currentLevel":93}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)
  zigbee2mqtt:info 2/25/2019, 12:08:50 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":93,"linkquality":36,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:08:50 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":93}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)
  zigbee2mqtt:info 2/25/2019, 12:08:50 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":93,"linkquality":36,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:info 2/25/2019, 12:08:50 PM Zigbee publish to device '0x00158d0001cded1c', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:debug 2/25/2019, 12:08:50 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":93}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)
  zigbee2mqtt:info 2/25/2019, 12:08:50 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":93,"linkquality":36,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:09:02 PM Received MQTT message on 'zigbee2mqtt/0x00158d0001cded1c/set' with data '{"state": "OFF"}'
  zigbee2mqtt:info 2/25/2019, 12:09:02 PM Zigbee publish to device '0x00158d0001cded1c', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2/25/2019, 12:09:02 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":93,"linkquality":36,"state":"OFF","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:09:02 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)
  zigbee2mqtt:info 2/25/2019, 12:09:02 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":93,"linkquality":34,"state":"OFF","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:09:02 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)
  zigbee2mqtt:info 2/25/2019, 12:09:02 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":93,"linkquality":34,"state":"OFF","last_seen":"2019-01-18T08:31:49.504Z"}'
  zigbee2mqtt:debug 2/25/2019, 12:09:02 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)
  zigbee2mqtt:info 2/25/2019, 12:09:02 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":93,"linkquality":34,"state":"OFF","last_seen":"2019-01-18T08:31:49.504Z"}'
```

GForce2010, 2019-02-25:
`2/25/2019, 4:08:41 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:41 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":1}'
2/25/2019, 4:08:41 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":254}'
2/25/2019, 4:08:41 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:43 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - moveToLevelWithOnOff - {"level":1,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:44 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":254}'
2/25/2019, 4:08:44 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":1}'
2/25/2019, 4:08:44 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:47 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_1/availability', payload 'online'
2/25/2019, 4:08:48 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_2/availability', payload 'online'
2/25/2019, 4:08:48 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_3/availability', payload 'online'`

GForce2010, 2019-02-25:
2/25/2019, 4:08:41 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:41 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":1}'
2/25/2019, 4:08:41 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":254}'
2/25/2019, 4:08:41 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:43 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - moveToLevelWithOnOff - {"level":1,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:44 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":254}'
2/25/2019, 4:08:44 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":1}'
2/25/2019, 4:08:44 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:47 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_1/availability', payload 'online'
2/25/2019, 4:08:48 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_2/availability', payload 'online'
2/25/2019, 4:08:48 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_3/availability', payload 'online'

GForce2010, 2019-02-25:
> 2/25/2019, 4:08:41 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:41 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":1}'
2/25/2019, 4:08:41 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":254}'
2/25/2019, 4:08:41 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:43 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - moveToLevelWithOnOff - {"level":1,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:44 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":254}'
2/25/2019, 4:08:44 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":1}'
2/25/2019, 4:08:44 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 4:08:47 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_1/availability', payload 'online'
2/25/2019, 4:08:48 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_2/availability', payload 'online'
2/25/2019, 4:08:48 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_3/availability', payload 'online'

Koenkk, 2019-02-25:
@Kryzek in your case it's expected behaviour, see explanation below:

```
# Zigbee2mqtt receives command to turn device ON
  zigbee2mqtt:debug 2/25/2019, 12:08:30 PM Received MQTT message on 'zigbee2mqtt/0x00158d0001cded1c/set' with data '{"state": "ON"}'

# Zigbee2mqtt executes this command successfully
  zigbee2mqtt:info 2/25/2019, 12:08:30 PM Zigbee publish to device '0x00158d0001cded1c', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null

# Because the command executed successfully, zigbee2mqtt assumes that the state of this device has been changed to ON, as not all devices report this, zigbee2mqtt publishes this 'assumed' state
  zigbee2mqtt:info 2/25/2019, 12:08:30 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":108,"linkquality":34,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'

# The device itself reports that is has turned on via a zigbee message
  zigbee2mqtt:debug 2/25/2019, 12:08:30 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c)

# Zigbee2mqtt publishes the above zigbee message to MQTT
  zigbee2mqtt:info 2/25/2019, 12:08:30 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"brightness":108,"linkquality":34,"state":"ON","last_seen":"2019-01-18T08:31:49.504Z"}'
```

GForce2010, 2019-02-25:
As you can see from the log below I send a command to "Her_Lamp" to set brightness to 255. I then receive a state update with it's current state then a second state report with it's new state.

```
 2/25/2019, 5:22:45 PM - debug: Ping 0x00158d00020a5a56
2/25/2019, 5:22:45 PM - debug: Successfully pinged 0x00158d00020a5a56
2/25/2019, 5:22:45 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp/availability', payload 'online'
2/25/2019, 5:22:50 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3002},{"elmType":33,"elmVal":9128},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":403},{"elmType":32,"elmVal":95}]}}' of device 'lumi.sensor_switch' (0x00158d0002c8caf6)
2/25/2019, 5:22:50 PM - info: MQTT publish: topic 'zigbee2mqtt/Butt_Her_Lamp', payload '{"battery":100,"voltage":3002,"linkquality":118}'
2/25/2019, 5:22:50 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3002},{"elmType":33,"elmVal":9128},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":403},{"elmType":32,"elmVal":95}]}}' of device 'lumi.sensor_switch' (0x00158d0002c8caf6)
2/25/2019, 5:22:54 PM - debug: Received MQTT message on 'zigbee2mqtt/Her_Lamp/set' with data '{   "brightness": 255.0 }'
2/25/2019, 5:22:54 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 5:22:54 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":1}'
2/25/2019, 5:22:55 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'RB 145' (0x00158d00020a5a56)
2/25/2019, 5:22:55 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":254}'
2/25/2019, 5:22:55 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 5:22:57 PM - debug: Received MQTT message on 'zigbee2mqtt/Her_Lamp/set' with data '{   "brightness": 1.0 }'
2/25/2019, 5:22:57 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - moveToLevelWithOnOff - {"level":1,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 5:22:57 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":254}'
2/25/2019, 5:22:57 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":1}}' of device 'RB 145' (0x00158d00020a5a56)
2/25/2019, 5:22:57 PM - info: MQTT publish: topic 'zigbee2mqtt/Her_Lamp', payload '{"state":"ON","brightness":1}'
2/25/2019, 5:22:57 PM - info: Zigbee publish to device '0x00158d00020a5a56', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
2/25/2019, 5:23:03 PM - debug: Received MQTT message on 'zigbee2mqtt/bridge/config/log_level' with data 'info'
2/25/2019, 5:23:03 PM - info: Switching log level to 'info'
2/25/2019, 5:23:03 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
2/25/2019, 5:23:08 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_1/availability', payload 'online'
2/25/2019, 5:23:08 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_2/availability', payload 'online'
2/25/2019, 5:23:09 PM - info: MQTT publish: topic 'zigbee2mqtt/Main_Study_3/availability', payload 'online'
```

Koenkk, 2019-02-26:
Fixed, please checkout the latest dev branch.

Kryzek, 2019-02-27:
> 
> 
> @Kryzek in your case it's expected behaviour, see explanation below:
> 
> 
Oh, I see. Could this self-reporting behaviour be learned (or set via configuration) per device so M2Z wouldn't need to do these publishes for the device that do this themself?

Koenkk, 2019-02-27:
@Kryzek technically this is possible, but this will increase the delay between the state acknowledge. I don't see much benefit in adding this.

GForce2010, 2019-02-27:
I'm afraid I'm all rather new to Raspberry Pi's.

I can see this instruction in the documentation but I am assuming that the update of the dev version would require a slightly different download command.

```
# Stop zigbee2mqtt and go to directory
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt

# Backup configuration
cp -R data data-backup

# Update
git checkout HEAD -- npm-shrinkwrap.json
git pull
rm -rf node_modules
npm install

# Restore configuration
cp -R data-backup/* data
rm -rf data-backup

# Start zigbee2mqtt
sudo systemctl start zigbee2mqtt 
```

Any help greatly appreciated. 

Thanks.

Koenkk, 2019-02-27:
Replace the update section with:

```bash
# Update
git checkout HEAD -- npm-shrinkwrap.json
git fetch
git checkout dev
git pull
rm -rf node_modules
npm install
```

GForce2010, 2019-02-28:
I've just updated, I am still seeing two state reports but now both state reports have the same values so it has cured the flickering on the control.

Should there still be two state reports or should there only be one?

Koenkk, 2019-02-28:
@GForce2010 two is expected, see https://github.com/Koenkk/zigbee2mqtt/issues/1155#issuecomment-467094619, thanks for verifying!

closed 2019-02-28 by Koenkk

reyhard, 2024-10-20:
Hello, 
I'm sort of getting similar issue as in this topic but with Aqara TRV E1 ( https://www.zigbee2mqtt.io/devices/SRTS-A01.html#aqara-srts-a01 ) 
```
info 2024-10-20 19:36:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/trv_03', payload '{"away_preset_temperature":"5.0","battery":100,"calibrate":null,"calibrated":true,"child_lock":false,"device_temperature":24,"external_temperature_input":null,"internal_heating_setpoint":24,"last_seen":"2024-10-20T19:36:25+02:00","linkquality":174,"local_temperature":22.7,"occupied_heating_setpoint":24.5,"power_outage_count":1,"preset":"manual","schedule":false,"schedule_settings":"mon,sun|0:00,21.0|12:00,17.0|18:00,21.0|23:59,21.0","sensor":"internal","setup":false,"system_mode":"off","update":{"installed_version":2590,"latest_version":2590,"state":"idle"},"update_available":false,"valve_alarm":false,"valve_detection":true,"voltage":3100,"window_detection":false,"window_open":false}'
info 2024-10-20 19:36:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/trv_03', payload '{"away_preset_temperature":"5.0","battery":100,"calibrate":null,"calibrated":true,"child_lock":false,"device_temperature":24,"external_temperature_input":null,"internal_heating_setpoint":24,"last_seen":"2024-10-20T19:36:25+02:00","linkquality":174,"local_temperature":22.7,"occupied_heating_setpoint":24,"power_outage_count":1,"preset":"manual","schedule":false,"schedule_settings":"mon,sun|0:00,21.0|12:00,17.0|18:00,21.0|23:59,21.0","sensor":"internal","setup":false,"system_mode":"off","update":{"installed_version":2590,"latest_version":2590,"state":"idle"},"update_available":false,"valve_alarm":false,"valve_detection":true,"voltage":3100,"window_detection":false,"window_open":false}'
```
when changing occupied_heating_setpoint from 24.5 to 24 degrees, I get two messages. Thing is also happening with other parameters (most of them I think - at least I those things which are exposed in frontend). I guess, similar to brightness, it should report 2 times same value, not 2 different ones. Similar to author of this report, I'm also using openHab and this is causing flickering issue in interface for me. I can confirm though, that zigbee lightbulbs are not showing such behaviour when switching brightness