Koenkk/zigbee2mqtt #1157: Use click and action for aqara button state in home assistant

h4nc, 2019-02-25:
@Koenkk currently only "click" is reportet to the aqara button sensor state in home assistant.
So only "single", "double" can be reportet to the state.

I use (because of the issue, only works after 1-2 seconds after triggered, you know what I mean)

```
  trigger:
    platform: mqtt
    topic: 'zigbee2mqtt/ZigbeeButton'   
  condition:
    - condition: template
      value_template: '{{ "hold" == trigger.payload_json.**action** }}' 
```

For click is looks like this

```
  trigger:
    platform: mqtt
    topic: 'zigbee2mqtt/ZigbeeButton'   
  condition:
    - condition: template
      value_template: '{{ "single" == trigger.payload_json.**click** }}' 
```
So I don't need this, but I noticed it and might be necessary for other users, that want to use the "hold" feature that is reportet in "action" not in "click".

marc-gist, 2019-02-26:
not sure this is an "issue" :) ... curious, what does the asterix's in `**click**` and `**action**` do in HA?
or is it supposed to be how i have it `value_template: "{{ value_json.click }}"` and you wanted to bold the click and action?

also note you can add:

    json_attributes: 
        - "battery"
        - "voltage"
        - "linkquality"
        - "last_seen"

h4nc, 2019-02-26:
Sry for the ** ** 
Ignore them 
I wanted to make them bold

I think json_attributes is obsolete.

It got replaced by json_attribute_topic if I remember it right.

marc-gist, 2019-02-26:
> I think json_attributes is obsolete.
> 
> It got replaced by json_attribute_topic if I remember it right.

Not in then **sensor:** section... this works for me. How else would you setup the buttons? I have them as sensors.

h4nc, 2019-02-26:
I have my devices autodiscovered. So I don't set them up manually.

I think json_attributes still work for mqtt sensors, but they are deprecated.

See here

https://www.home-assistant.io/components/sensor.mqtt/

> json_attributes_topic
> (string)(Optional)The MQTT topic subscribed to receive a JSON dictionary payload and then set as sensor attributes.
> json_attributes
> (list | string)(Optional)(Deprecated, replaced by json_attributes_topic) A list of keys to extract values from a JSON dictionary payload and then set as sensor attributes.

You will see a warning in your logs

yaniv1111, 2019-02-28:
@marc-gist , can you please show an example of automation with the sensor definition and 
json_attribute_topic?

h4nc, 2019-02-28:
I don't thing that he uses 
json_attribute_topic

What do you mean with "automation with the sensor definition and json_attribute_topic?"

Do you mean configuration instead of automation? So an example how to use json_attribute_topic in a sensor?
If so simple add the topic name like you know it from

`state_topic: "some/mqtt/topic"`

`json_attribute_topic: "some/mqtt/topic"`

in this format.

yaniv1111, 2019-02-28:
yes, configuration for the mqtt sensor and automation example

h4nc, 2019-02-28:
Search for your device here 
https://www.zigbee2mqtt.io/integration/home_assistant.html

You can find your devicename here
https://www.zigbee2mqtt.io/information/supported_devices.html

for the awara button it would be WXKG11LM or WXKG12LM

and simply add the line as I mentioned above.

vevsvevs, 2019-03-14:
Indeed, this particular aqara buttons peculiarity ruins intuitive and seamless writing of automation. IMHO, action-to-state reposting should be done on the Z2M side for this device.

marc-gist, 2019-03-14:
> Indeed, this particular aqara buttons peculiarity ruins intuitive and seamless writing of automation. IMHO, action-to-state reposting should be done on the Z2M side for this device.

Z2M could do this, but I disagree that it should... Z2M is simply a "messenger".  easy enough to automate either way.

vevsvevs, 2019-03-14:
In general, i agree. But it already has some device-specific options (occupancy_timeout and *_precision), so maybe it's a good idea to add something like a key name mapping ability for such strange-developed devices into Z2M?

tanus10, 2019-04-01:
For WXKG12LM,
`  trigger:
    - platform: mqtt
      topic: 'zigbee2mqtt/xxxxxxxxxx'
  condition:
    condition: template
    value_template: '{{ "hold" == trigger.payload_json.action }}'`
and
`  trigger:
    - platform: state
      entity_id: sensor.xxxxxxxxxx_click
      to: 'double'`

but, for WXKG01LM
`  trigger:
    - platform: state
      entity_id: sensor.yyyyyyyyyy_click
      to: 'long'`

So, we have to trigger them differently. It would be better to include in the documentation that, 'platform: state' method won't work with 'hold' and 'shake' actions for WXKG12LM

closed 2019-06-07 by stale[bot]